espressif/esp-idf #8225: How to suppress error for "static declaration follows non-static declaration" in c ?

enthunilu, 2022-01-11:
## Environment

Development Kit: ESP32-Ethernet kit_A_V1.1
Kit version
Module or chip used: ESP32
IDF version :v4.2.1
Build System: Make|CMake|idf.py
Compiler version : xtensa-esp32-elf-gcc (crosstool NCG esp-2020r3)8.4.0
Operating System: Windows
(Windows only) environment type: Command Prompt
Using an IDE?: Yes (Eclipse)

## Problem Description
I am trying to integrate Pyramid EIP on ESP32. Pyramid EIP code I have taken as it is from my earlier code, which I was able to build on IAR.
When using same code on ESP32 using esp-idf I am getting one error as,
"static declaration of table follows non-static declaration"
CODE: SELECT ALL

// In Test.c
static uint8_t table = 0;

// In Test.h
extern uint8_t table;

I know it is giving error because we have declared same variable as extern in .h file.
However due to some limitations I cannot modify code. So, I wanted to know is there a way by which I can suppress this error using any flag or something in CMakeList.txt or any other way?

Thanks for your support!!

o-marshmallow, 2022-01-12:
Hi @enthunilu ,
If you can't modify the code manually, I make the assumption that a solution where CMake will first modify the files beforehand is not valid.

The only solution I can think of is to create a header file that will "rename" your static variable thanks to a `#define`, then force the compiler to include it inside your `Test.c` thanks to `-include`. I will take `example/get-started/hello-world` as an example here.

In the `main/CMakeList.txt` file, after `idf_component_register`, I add`:
```
# Force GCC to include the file override.h at the top of hello_world_main.c 
# GCC's -include FILE is equivalent to having the line #include FILE at the top of the file
set_source_files_properties(hello_world_main.c PROPERTIES COMPILE_FLAGS "-include ../main/override.h")
```

Now, in the newly created `main/override.h`:
```
#ifndef OVERRIDE_H
#define OVERRIDE_H

/* Start by including Test.h, else the define below would also replace the name `table` in that header
#include "Test.h"
/* Create a new name for the static variable */
#define table table_static

#endif
```

Compilation should now work. This solution is dirty and not stable, as it will replace all the occurrences of `table` in `hello_world_main.c`, even if not related to the static `table`variable, so use it with caution.

Another possible solution would be to replace `table` in the `Test.h` header file, this seems less risky to me, but it depends on the content of the header file, anyway, you would only need to replace the content of `main/override.h` by:
```
#ifndef OVERRIDE_H
#define OVERRIDE_H

#define table table_global
#include "Test.h"
#undef table

#endif
```

enthunilu, 2022-01-13:
Hi @o-marshmallow,

Thanks for your quick response. Your suggestion worked for me.
Now I am able to resolve the error.

Thanks!!!

closed 2022-01-13 by enthunilu

o-marshmallow, 2022-01-13:
Thanks @enthunilu  for your feedback!