espressif/esp-idf #8226: ADF环境安装完成后编译出现问题

CYCHEN97, 2022-01-12:
@ESP-YJM[image]
ADF安装好后进行编译自动语音识别demo时，提示python版本问题，还没安装ADF之前linux中只有python3，安装好ADF后多了几个python版本
[image]

CYCHEN97, 2022-01-12:
[image]
ADF安装成功可以在VSCODE显示example
[image]
但是点这个配置后提示错误

CYCHEN97, 2022-01-12:
@ESP-YJM
麻烦帮忙看一下

ESP-YJM, 2022-01-12:
@CYCHEN97 非常抱歉，我对 ADF 和环境配置不是很熟悉，建议你去 https://github.com/espressif/esp-adf/issues 这个下面提问。我们会有负责该工程的同事帮你解答问题。

CYCHEN97, 2022-01-12:
@ESP-YJM
好的，谢谢

closed 2022-01-12 by CYCHEN97