espressif/esp-idf #8228: 用于自定义编辑esp-idf的方法 (Methods for custom editing esp-idf )

Zmmfly, 2022-01-13:
## **Is your feature request related to a problem? Please describe.**

你好，在freertos中，“vTaskGetRunTimeStats”在一段时间后将会溢出，导致结果不准确。
在freertos论坛中存在解决方案，但是需要对freertos源代码进行修改；如果我直接修改esp-idf中的代码，明显会造成工程失去移植性，除非复制一份freertos组件到工程目录下的components文件夹。
但是这样又会增加工程的体积。

Hello, in freertos, "vTaskGetRunTimeStats" will overflow after some time, resulting in inaccurate results.
There is a solution in the freertos forum, but the freertos source code needs to be modified; if I directly modify the code in esp-idf, it will obviously cause the project to lose portability, unless a copy of the freertos component is copied to the components folder in the project directory .
But this will increase the volume of the project. 

## **Describe the solution you'd like**

在我的设想中，可以增加一个构建前，在构建文件夹对esp-idf组件进行打补丁的流程。
应用工程文件夹下patches文件夹内所有的补丁后，再使用打完补丁的文件替换esp-idf源文件来进行编译。
这样，既可以保持esp-idf的完整性，又可以对其进行修改，同时又可以保持工程的简洁性。

In my imagination, a process of patching the esp-idf components in the build folder before building can be added.
After applying all the patches in the patches folder under the project folder, use the patched files to replace the esp-idf source files to compile.
In this way, the integrity of esp-idf can be maintained, and it can be modified, while maintaining the simplicity of the project. 

## **Describe alternatives you've considered**

1. 在以前的工程中，如果我需要对esp-idf组件进行修改，我会将被修改的组件复制一份到工程文件夹下的components文件夹内，再来进行修改。
但如果修改的组件比较多，工程的体积将变得巨大，并且难以管理。
2. 使用工程目录下replaces文件内的文件替换esp-idf内的源文件进行编译，这实现起来也许没有打补丁那么复杂。

===========

1. In the previous project, if I need to modify the esp-idf component, I will copy the modified component to the components folder under the project folder, and then modify it. 
But if there are many modified components, the size of the project will become huge and difficult to manage. 
2. Use the files in the replaces file in the project directory to replace the source files in esp-idf for compilation, which may not be as complicated as patching.

0xjakob, 2022-01-13:
Have you tried looking at [overriding components](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/build-system.html#cmake-components-same-name)?

Zmmfly, 2022-01-13:
> Have you tried looking at [overriding components](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/build-system.html#cmake-components-same-name)?

Yes, that's the way it is currently used, but it will make a full copy of the component in the project directory.

o-marshmallow, 2022-01-13:
Hi @Zmmfly ,
Can you show us what should be fixed exactly inside FreeRTOS component? Maybe adding a `wrapper` at link time would be enough?

closed 2022-05-19 by Zmmfly