espressif/esp-idf #8229: mesh找wifi失败

waqq1314, 2022-01-13:
[image]

如上图，左边是我这边的应用，右边是官方internal_communication例程，官方的没问题，我那边那个却提示无法找到对应的wifi ap。
但是如果我设置成另一个wifi ssid，其他什么都不变，却又能找到。
还有就是如果我这边添加设置
ESP_ERROR_CHECK(esp_mesh_set_type(1));
也能找到设置的wifi，请问这是什么原因

从现象和log看我发现了两点不一样的
1.mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x1, need_scan_router:0x0, look_for_nwk_count:1
和正常的那个need_scan:0x03不一样
2.显示mesh starts successfully后，
我这边间隔300ms左右就出现接下来的log了，而正常的那边等了4s后才显示接下来的log

两个现象都因为代码封装而无法继续深入查找原因，还请官方提示一下什么情况

shenjun7, 2022-01-14:
左边信道不匹配无法scan到AP, 如果Router信道不固定的话可以配置cfg.allow_channel_switch = true;进行全信道扫描．esp_mesh_set_type(1)设置fixed-root网络信道不匹配root可以连上，子节点无法连接，也需要匹配信道或者进行全信道扫描．

waqq1314, 2022-01-14:
> 左边信道不匹配无法scan到AP, 如果Router信道不固定的话可以配置cfg.allow_channel_switch = true;进行全信道扫描．esp_mesh_set_type(1)设置fixed-root网络信道不匹配root可以连上，子节点无法连接，也需要匹配信道或者进行全信道扫描．

非常感谢！
今天一早我挨个配置比对，也发现了这个问题，我这边改成了1，官方默认是0，我强制写成0也可以了

closed 2022-01-19 by waqq1314