espressif/esp-idf #8231: ESP32-S2 ext1_rtc_gpio_mask value incorrect in sleep_modes.c

DylanGWork, 2022-01-13:
Hi all,

I have come across an issue with the repository that requires updating to allow for the GPIO's 18-21 to be used for the ESP32-S2 interrupts in the function esp_sleep_enable_ext1_wakeup(ext_wakeup_pin_1_mask,0);

I have deleted much of the issue template as I have solved this issue, but I am submitting this issue so that the team are aware of it and can fix it in the Espressif code base.

I have the issue detailed in this repository that I used for testing and resolving the issue:
https://github.com/DylanGWork/esp32s2ulp-interrupt

- Development Kit:      ESP32-S2 mini-u
- IDF version 4.4

igrr, 2022-01-13:
Hi @DylanGWork, this was recently fixed in https://github.com/espressif/esp-idf/commit/9382b24d1dbe36d73bdba1a0d139695ca4360012. We will backport this fix to release/v4.4 and release/v4.3!

DylanGWork, 2022-01-13:
Oh excellent. I spent some time looking through the open and closed issues prior to posting. Is there a way I could have known this prior? Or did I just manage to miss the issue?

igrr, 2022-01-13:
This issue wasn't reported on Github, as far as I know. We were made aware of it by our customer support team, I suppose another customer has ran into this issue and reported it to them.

AxelLin, 2022-02-13:
> Hi @DylanGWork, this was recently fixed in [9382b24](https://github.com/espressif/esp-idf/commit/9382b24d1dbe36d73bdba1a0d139695ca4360012). We will backport this fix to release/v4.4 and release/v4.3!

@igrr 

I still don't find the fix in stable branches. 

PS.
It's not to push you, I asked this because I cannot differentiate if the backport is missed for some reason or the backport process is just slow. (Especially this is not a complex change, should not take such long time to backport)

AxelLin, 2022-03-05:
v4.3 fix: f8e45766b573cc3d13c792aeca901997c5bb875a

ginkgm, 2022-03-28:
master	 b84abb1b177b8a77329434b80cd686b27e7c5424
release/v4.4	 5f56bbd2d0b48c123fcba3ce4c25c2361ad4c78d
release/v4.3	 c166160f6491a32e81b01d1715853b3eb06446bc

closed 2022-03-28 by ginkgm