espressif/esp-idf #8232: ESP32-S2 Co-processor does not shutdown when shutdown function called

DylanGWork, 2022-01-13:
## Environment

- Development Kit:      ESP32-S2-Mini-1U
- IDF version 4.4
- 
**## Problem Description**
The function void ulp_riscv_shutdown(void) in the[ ulp_riscv_utils.c](https://github.com/espressif/esp-idf/blob/master/components/ulp/ulp_riscv/ulp_riscv_utils.c) file is not shutting down the processing.

It's comments show that it may not be programmed to shutdown, but to reset?

```
void ulp_riscv_shutdown(void)
{
    /* Setting the delay time after RISCV recv `DONE` signal, Ensure that action `RESET` can be executed in time. */
    REG_SET_FIELD(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_SHUT_2_CLK_DIS, 0x3F);

    /* suspends the ulp operation*/
    SET_PERI_REG_MASK(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_DONE);

    /* Resets the processor */
    SET_PERI_REG_MASK(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_SHUT_RESET_EN);

    while(1);
}
```
I have tried changing the line for resetting the processor to shutdown as described in the technical reference guide on page 54
https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf

RTC_CNTL_COCPU_SHUT Shut down ULP-RISC-V. (R/W)

However this results in the COCPU crashing instead.

I have attached an image which shows code, using the ulp_riscv_wakeup_main_processor function follow by the ulp_riscv_shutdown function not shutting down the ULP. This is evident by the spikes in the image being the power consumption reading of LED's turning on and off. You can see the spikes continuing through a LoRaWAN send and receive cycle in the main processor before the ULP is re-initiated before going to sleep (redundant since it never turned off).
[image]

I also describe this error on the forums here:
https://www.esp32.com/viewtopic.php?f=13&t=25475&p=90341#p90341

To use part of the issue template to summarise.

Expected outcome:
ULP Co-processor shuts down until re-initiated by the main CPU when using the ulp_riscv_shutdown in the  ulp_riscv_utils.c source file.

Actual outcome:
ULP co-processor continues to operate.

ESP-Marius, 2022-01-14:
Hi,

I think what you are seeing here is the expected behavior. `ulp_riscv_shutdown` only shuts the ULP down temporarily, it will start running again at the next wake up (decided by `ulp_set_wakeup_period`). I agree that the function name is a bit misleading though....

Could this explain what you are experiencing?

DylanGWork, 2022-01-14:
What you just explained appears to be exactly what I am experiencing.

Since my code goes to sleep every 250ms after checking some sensors, starting again at the next wake-up essentially shows no difference at all.

It would be beneficial to be able to turn off the co-processor while the main processor is operation as they share variables to change state conditions. The co-processor reading a sensor while the main processor is operating can have an impact on my state control.

DylanGWork, 2022-01-14:
cheers :)

We are doing as much analysis as we can in the co-processor to reduce power consumption.

closed 2022-01-21 by espressif-bot