espressif/esp-idf #8233: Interrupt example for ESP32-S2 Co-processor

DylanGWork, 2022-01-13:
We would like to have the ability to wake-up the ULP co-processor from an interrupt. It appears to be possible but I have been unable to produce working code.

**Describe the solution you'd like**

A situation for this code would be:
An accelerometer interrupt (or button)is activated to wake up the ULP co-processor, the co-processor evaluates analogue sensors to validate if the main CPU requires to be woken up for an alert to be transmitted.

**Describe alternatives you've considered**

We have considered moving all interrupts to the main MCU, however as we anticipated this task was achievable we did not provision our hardware with a logic gate to have all interrupts at the same logic level as each other for a function such as the esp_sleep_enable_ext1_wakeup function in sleep_modes.c

DylanGWork, 2022-01-18:
Just bumping this Feature request, is someone able to comment if this feature is possible?

ESP-Marius, 2022-01-19:
@DylanGWork The TRM seems very vague with regards to this. It mentions the timer as the only wakeup source, but further down references `RTC_CNTL_ULP_CP_GPIO_WAKEUP_ENA`. Unfortunately it doesn't really provide any information on how to actually do this.

Don't really know myself, but i'll try to chase down the answer for you today. As well as make sure we update the TRM to be more accurate and in-depth on this topic.

DylanGWork, 2022-01-19:
That would be excellent thank you.

ESP-Marius, 2022-01-19:
@DylanGWork Waking up the ULP  from a GPIO should be possible. I'll try to see if I can put together a proof of concept tomorrw.

ESP-Marius, 2022-01-20:
[0001-ulp-gpio-intr-example.patch.txt](https://github.com/espressif/esp-idf/files/7902238/0001-ulp-gpio-intr-example.patch.txt)

See the patch for a rough proof of concept example (`examples/system/ulp_riscv/gpio_intr`) where the main CPU wont be waken up from deep sleep before the GPIO goes high.

When actually implementing this feature I'll have to run it past the guys that actually knows stuff about our RTC IO/sleep power domains, but at least this example should show that it is possible.

DylanGWork, 2022-01-21:
Cheers :), great to see progress on this. I can't find the branch/location for the examples/system/ulp_riscv/gpio_intr proof of concept example?

The text file you provided helps of course, much appreciated.

ESP-Marius, 2022-01-21:
@DylanGWork there is not branch (as only our release branches are synced to github). 

But if you checkout the latest master and do `git apply 0001-ulp-gpio-intr-example.patch.txt` you should be able to see the example.

DylanGWork, 2022-02-17:
@ESP-Marius I've been working with this example a bit now and something isn't quite working and I can't figure out why.

It appears to have trouble running the timer and interrupt at the same time.

I also have added a blinking LED in the ULP instead of the wake-up, this can cause it to crash (using the same code I use in my main device which has been working for a long time now).

This has certainly worked for a proof of concept, it feels like the RTC IO /sleep power domains may need to be looked at?

Cheers,
Dylan

ESP-Marius, 2022-02-17:
@DylanGWork Working on making a proper MR for IDF now, so hopefully we should have full support for this soon. Thanks for being patient!

> It appears to have trouble running the timer and interrupt at the same time.
Just to be 100% clear: so what you want to achieve is both being able to be woken up by RTC IO as well as the timer right? Will have to check if this is even possible, at least from my quick test it didnt seem to work.

> I also have added a blinking LED in the ULP instead of the wake-up, this can cause it to crash (using the same code I use in my main device which has been working for a long time now).
Are you able to share this code? Then I can try to reproduce this crash.

> It will often wake-up several times after the first interrupt and then the interrupt will stop working (as if it's crashed or been disabled), the wake-up code is still 6 though, so I'm thinking the interrupt becomes disabled? 
I've found that the interrupt should probably be cleared when the ULP wakes up (`SET_PERI_REG_MASK(RTC_CNTL_ULP_CP_TIMER_REG, RTC_CNTL_ULP_CP_GPIO_WAKEUP_CLR);`), can you see if adding this improves your situation?

DylanGWork, 2022-02-20:
I put some example code here:
https://github.com/DylanGWork/esp32s2ulp-interrupt.git

I swear I've had it working in the past but this version is not very operational, it contains the crash as mentioned above though.

ESP-Marius, 2022-02-22:
Ran your ulp code + added a intr clear at the bottom of the ulp main (`SET_PERI_REG_MASK(RTC_CNTL_ULP_CP_TIMER_REG, RTC_CNTL_ULP_CP_GPIO_WAKEUP_CLR);` and it seems to run fine. 

Havent been able to reproduce any crashes, and the GPIO is set as expected.

DylanGWork, 2022-02-22:
I see, I have added what you suggested and in the ulp_riscv.c file I've commented out the timer setup:
SET_PERI_REG_MASK(RTC_CNTL_ULP_CP_TIMER_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN);
The code now works as purely interrupt driven.

What can we try in order to have both interrupt and timer enabled:
    /* start ULP_TIMER */
    CLEAR_PERI_REG_MASK(RTC_CNTL_ULP_CP_CTRL_REG, RTC_CNTL_ULP_CP_FORCE_START_TOP);
    SET_PERI_REG_MASK(RTC_CNTL_ULP_CP_TIMER_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN); //Timer controlled ULP
    SET_PERI_REG_MASK(RTC_CNTL_ULP_CP_TIMER_REG, RTC_CNTL_ULP_CP_GPIO_WAKEUP_ENA); //Interrupt controlled ULP

And the outcome for the code I posted would be a blinking LED every 20 seconds OR whenever the button is pressed.

Cheers for the help too btw, how to use SET_PERI_REG_MASK and CLEAR_PERI_REG_MASK can be overwhelming when first starting out, its starting to make a lot more sense now.
Cheers,
Dylan

Alvin1Zhang, 2022-03-08:
Thanks for reporting, fix is available https://github.com/espressif/esp-idf/commit/706a14884ffd2274fe31be59be74dccf04389b38, feel free to reopen.

closed 2022-03-08 by Alvin1Zhang