espressif/esp-idf #8236: WIFI Connection is lost after few minutes

Amralmasry, 2022-01-13:
## Environment

- Development Kit: 
 - Module or chip used:  [ESP32-WROOM-32| 
- IDF version  : 4.3.2
 - Operating System:     [Windows]
 - Using an IDE?: Yes Eclipse
- Power Supply:         [USB ]

## Problem Description
in station mode after connecting to my home router  by 5 ~ 10 mins , the wifi connection is lost  , I am not using any special code or something , just listen socket with 2 client connected 
 
 

### Steps to reproduce

 when using sever socket example from esp-idf

  

## Debug Logs

I get that once the connection is lost 

I (276265) wifi:state: run -> init (fa0)
I (276265) wifi:pm stop, total sleep time: 0 us / 275254296 us

W (276265) wifi:<ba-del>idx
I (276265) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (276275) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (276275) wifi:state: init -> auth (b0)
I (276285) wifi:state: auth -> assoc (0)
I (276285) wifi:state: assoc -> run (10)
I (277295) wifi:connected with Amr_Elmasry, aid = 1, channel 1, BW20, bssid = 94:a7:b7:4e:fd:1a
I (277295) wifi:security: WPA2-PSK, phy: bgn, rssi: -46
I (277305) wifi:pm start, type: 0

I (277315) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (277315) esp_netif_handlers: sta ip: 192.168.1.22, mask: 255.255.255.0, gw: 192.168.1.1
I (277315) example: got ip: 192.168.1.22

chegewara, 2022-01-14:
`wifi:pm stop, total sleep time: 0 us / 275254296 us`
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv415esp_wifi_set_ps14wifi_ps_type_t

mhdong, 2022-01-14:
hi @Amralmasry run -> init (fa0), fa0 means the STA has received the disassociation frame and reason is 4WAY_HANDSHAKE_TIMEOUT. 
Can post the complete log?
Could you provide a wireless packet capture corresponding to the unconnected wifi log?

Amralmasry, 2022-01-14:
> `wifi:pm stop, total sleep time: 0 us / 275254296 us` https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv415esp_wifi_set_ps14wifi_ps_type_t

I tried the 3 modes but no difference

Amralmasry, 2022-01-14:
> hi @Amralmasry run -> init (fa0), fa0 means the STA has received the disassociation frame and reason is 4WAY_HANDSHAKE_TIMEOUT. Can post the complete log? Could you provide a wireless packet capture corresponding to the unconnected wifi log?

D (1172) wifi:Send sta connected event
D (1172) wifi:connect status 1 -> 5
D (1172) wifi:obss scan is disabled
D (1182) wifi:start obss scan: obss scan is stopped
I (1182) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1192) esp_netif_handlers: sta ip: 192.168.1.22, mask: 255.255.255.0, gw: 192.168.1.1
I (1192) example: got ip: 192.168.1.22
I (4002) example: Socket created
I (4002) example: Socket bound, port 3333
I (4002) example: Socket listening
W (22812) wifi:<ba-add>idx:0 (ifx:0, 94:a7:b7:4e:fd:1a), tid:0, ssn:0, winSize:64
I (25272) example: connected new con 55
I (31542) example: connected new con 56
E (32542) example: Error occurred during sending: errno 104
W (32542) example: Connection closed

" from here the issue occurs"

V (468542) wifi:wpa_psk start
V (468552) wifi:wpa_psk handle succeed
D (468552) wifi:obss scan is disabled
D (468552) wifi:start obss scan: obss scan is stopped
I (468582) wifi:state: run -> init (ec0)
D (468582) wifi:recv deauth/disassoc, stop beacon/connect timer
D (468582) wifi:connect status 5 -> 4
D (468582) wifi:sta leave
I (468592) wifi:pm stop, total sleep time: 223124832 us / 466005972 us

D (468592) wifi:stop CSA timer
W (468602) wifi:<ba-del>idx
D (468602) wifi:remove 94:a7:b7:4e:fd:1a from rc list
I (468602) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
D (468612) wifi:filter: set rx policy=8
D (468612) wifi:Send disconnect event, reason=14, AP number=0
D (468622) wifi:Start wifi connect
D (468622) wifi:connect status 4 -> 0
D (468622) wifi:connect chan=0
D (468632) wifi:first chan=1
D (468632) wifi:connect status 0 -> 1
D (468632) wifi:filter: set rx policy=3
D (468642) wifi:clear scan ap list
D (468642) wifi:start scan: type=0x50f, priority=2, cb=0x4010ec34, arg=0x0, ss_state=0x1, time=466547385, index=0
addr2line -pfiaC -e e:\esp\power_sa\build\power_save.elf 0x4010ec34: [WinError 2] The system cannot find the file specified
V (468652) wifi:inter_channel_timeout: arg=0x0, ss_state=0x1
D (468652) wifi:perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
V (468662) wifi:change: chan<1,0>, dur<0,120>
V (468662) wifi:enter start op, arg=0x3ffb53ed
V (468672) wifi:scan operation: state=0x3, chan<1,0>, arg=0x3ffb53ed, status=0
V (468682) wifi:scan specific ssid=Amr_Elmasry
V (468682) wifi:start max timer
D (468692) wifi:rsn valid: gcipher=1 ucipher=3 akm=5

D (468692) wifi:profile match: ss_state=0x7
V (468692) wifi:update scan history ssid=Amr_Elmasry, channel=1
D (468702) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (468702) wifi:find first mathched ssid, scan done
V (468712) wifi:back home chan=<1,0>, current chan=<1,0>
D (468712) wifi:filter: set rx policy=4
D (468722) wifi:first chan=1
V (468722) wifi:scan_done: arg=0x0, status=0, cur_time=466625647, scan_id=128, scan state=0
V (468732) wifi:call scan_done cb, arg=0x0
D (468732) wifi:handoff_cb: status=0
V (468732) wifi:best bss has set.
D (468742) wifi:ap found, mac=94:a7:b7:4e:fd:1a
V (468742) wifi:bssid=94:a7:b7:4e:fd:1a, LR=0
D (468742) wifi:new_bss=0x3ffb6660, cur_bss=0x0, new_chan=<1,0>, cur_chan=1
D (468752) wifi:filter: set rx policy=5
I (468752) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
D (468762) wifi:connect_op: status=0, auth=5, cipher=3
D (468772) wifi:auth mode is not none
D (468772) wifi:connect_bss: auth=1, reconnect=0
I (468772) wifi:state: init -> auth (b0)
D (468782) wifi:start 1s AUTH timer
D (468782) wifi:clear scan ap list
D (468792) wifi:recv auth: seq=2, status=0
I (468792) wifi:state: auth -> assoc (0)
D (468792) wifi:restart connect 1s timer for assoc
D (468802) wifi:recv assoc: type=0x10
D (468802) wifi:filter: set rx policy=6
I (468802) wifi:state: assoc -> run (10)
D (468812) wifi:start 10s connect timer for 4 way handshake
V (468812) wifi:wpa_psk start
V (468822) wifi:wpa_psk handle succeed
V (468822) wifi:wpa_psk start
V (468832) wifi:wpa_psk handle succeed
I (468832) wifi:connected with Amr_Elmasry, aid = 1, channel 1, BW20, bssid = 94:a7:b7:4e:fd:1a
I (468832) wifi:security: WPA2-PSK, phy: bgn, rssi: -49
D (468842) wifi:remove all except 94:a7:b7:4e:fd:1a from rc list
D (468842) wifi:clear blacklist
D (468862) wifi:filter: set rx policy=7
I (468862) wifi:pm start, type: 1

D (468862) wifi:Send sta connected event
D (468862) wifi:connect status 1 -> 5
D (468862) wifi:obss scan is disabled
D (468862) wifi:start obss scan: obss scan is stopped
I (468872) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (468882) esp_netif_handlers: sta ip: 192.168.1.22, mask: 255.255.255.0, gw: 192.168.1.1
I (468882) example: got ip: 192.168.1.22
E (469542) example: Error occurred during sending: errno 113
W (469542) example: Connection closed
W (602602) wifi:<ba-add>idx:0 (ifx:0, 94:a7:b7:4e:fd:1a), tid:0, ssn:0, winSize:64

mhdong, 2022-01-14:
hi @Amralmasry  
E (32542) example: Error occurred during sending: errno 104
W (32542) example: Connection closed
is that mean application layer termenate connection?
This time we receive Deauthentication from AP with reason Message integrity code (MIC) failure.
Could you provide AP settings for us to analysis?

Amralmasry, 2022-01-14:
> hi @Amralmasry E (32542) example: Error occurred during sending: errno 104 W (32542) example: Connection closed is that mean application layer termenate connection? This time we receive [...]

 E (32542) example: Error occurred during sending: errno 104 W (32542) example: Connection closed 
 
 those are from my application , a client connected and  disconnected , nothing not normal here 
 
 the issue if waited for 5 ~ 10 
 the WIFI connection is lost and trying to reconnect once again 
 
 and this is the router settings if that what you want 
 
[image]

mhdong, 2022-01-14:
hi @Amralmasry  You can try to increase the group key update cycle by set AP setting.

Amralmasry, 2022-01-14:
the router model is ZXHN H108N V2.5
 
but it was working fine with esp8266 when I was using AT commands , this is my first time using ESP_IDF

Amralmasry, 2022-01-14:
when I increased the  group key update cycle to 900 ms instead of 600 ms (default) in my router settings 

The connection time got increased actually but it ended up disconnecting again 

(883392) wifi:wpa_psk start
V (883392) wifi:wpa_psk handle succeed
D (883392) wifi:obss scan is disabled
D (883402) wifi:start obss scan: obss scan is stopped
I (883612) wifi:state: run -> init (ec0)
D (883612) wifi:recv deauth/disassoc, stop beacon/connect timer
D (883612) wifi:connect status 5 -> 4
D (883622) wifi:sta leave
I (883622) wifi:pm stop, total sleep time: 0 us / 880976698 us

D (883622) wifi:stop CSA timer
W (883632) wifi:<ba-del>idx
D (883632) wifi:remove 94:a7:b7:4e:fd:1a from rc list
I (883632) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
D (883642) wifi:filter: set rx policy=8
D (883642) wifi:Send disconnect event, reason=14, AP number=0
D (883652) event: running post WIFI_EVENT:5 with handler 0x400dde24 and context 0x3ffbe0f0 on loop 0x3ffaf8a8

mhdong, 2022-01-14:
@Amralmasry  You can set it to 0, means never update.

Amralmasry, 2022-01-14:
> @Amralmasry You can set it to 0, means never update.

only from 100 ~ 1000 is possible 
but even though , there must be a something wrong when updating the key

AxelLin, 2022-01-14:
> @Amralmasry You can set it to 0, means never update.

@mhdong 
I don't get your point.
Is that normal that esp32 will disconnect from AP when group key update?
Then why esp8266 has no problem?

AxelLin, 2022-01-15:
@mhdong @nishanth-radja @MaxwellAlan
I think https://github.com/espressif/esp-idf/issues/7182 is the same issue.
And according to https://github.com/espressif/esp-idf/issues/7182#issuecomment-966185333, esp-idf 4.0.3 doesn't have the disconnection issue.
This looks like a regression.

mhdong, 2022-01-15:
hi @Amralmasry  @AxelLin  We have fix this issue in master, commit is 0336584f3e2be3817769ab2b766b93d0a7dfbc4c, please confirm. We will backport this merge request to v4.1 v4.2 v4.3 v4.4 later.

Amralmasry, 2022-01-15:
I tried the master but the same issue occurs 

D (215293) wifi:obss scan is disabled
D (215293) wifi:start obss scan: obss scan is stopped
I (215363) wifi:state: run -> init (ec0)
D (215363) wifi:recv deauth/disassoc, stop beacon/connect timer
D (215363) wifi:connect status 5 -> 4
D (215373) wifi:sta leave
I (215373) wifi:pm stop, total sleep time: 0 us / 199933462 us

D (215373) wifi:stop CSA timer
W (215383) wifi:<ba-del>idx
D (215383) wifi:remove 94:a7:b7:4e:fd:1a from rc list
I (215383) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
D (215393) wifi:filter: set rx policy=8
D (215393) wifi:Send disconnect event, reason=14, AP number=0

AxelLin, 2022-01-16:
@Amralmasry 
This issue only happens on some APs, could you capture wireshark log for analysis?

mhdong, 2022-01-17:
@Amralmasry  We know that there must be something wrong when updating group key. Could you enable the logs, supplicant logs and wifi logs, which can be enabled using menuconfig. By the way, please paste complete log.

kapilkedawat, 2022-01-17:
@Amralmasry besides what @mhdong mentioned, please also set logging level to debug level.

Please also provide wireless sniffer capture packets if possible.

ktonder, 2022-01-23:
We are experiencing the same issue, using a ESP32C3 on three different locations. I can also try to provide a log but since im using the arduino I need to know what changes I need to do to the tools/sdk/esp32c3/sdkconfig and the tools/sdk/esp32c3/include/config/sdkconfig.h config file?

Amralmasry, 2022-01-23:
Executing action: monitor
Running idf_monitor in directory e:\esp\power_saveaaaaaaaaaaa
Executing "C:\Users\Amr\.espressif\python_env\idf5.0_py3.9_env\Scripts\python.exe E:\esppp\New\esp-idf\esp-idf\tools/idf_monitor.py -p \\.\COM3 -b 115200 --toolchain-prefix  --target esp32 --revision 0 --print_filter  e:\esp\power_saveaaaaaaaaaaa\build\power_save.elf -m 'C:\Users\Amr\.espressif\python_env\idf5.0_py3.9_env\Scripts\python.exe' 'E:\esppp\New\esp-idf\esp-idf\tools\idf.py'"...
?[0;33m--- idf_monitor on \\.\COM3 115200 ---?[0m
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6752
load:0x40078000,len:14796
addr2line -pfiaC -e e:\esp\power_saveaaaaaaaaaaa\build\power_save.elf 0x40078000: [WinError 2] The system cannot find the file specified
load:0x40080400,len:3792
addr2line -pfiaC -e e:\esp\power_saveaaaaaaaaaaa\build\power_save.elf 0x40080400: [WinError 2] The system cannot find the file specified
entry 0x40080694
addr2line -pfiaC -e e:\esp\power_saveaaaaaaaaaaa\build\power_save.elf 0x40080694: [WinError 2] The system cannot find the file specified
I (26) boot: ESP-IDF v5.0-dev-1181-g0132079fa 2nd stage bootloader
I (26) boot: compile time 13:08:08
I (27) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1783ch ( 96316) map
I (145) esp_image: segment 1: paddr=00027864 vaddr=3ffb0000 size=03d18h ( 15640) load
I (151) esp_image: segment 2: paddr=0002b584 vaddr=40080000 size=04a94h ( 19092) load
I (160) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=7042ch (459820) map
I (326) esp_image: segment 4: paddr=000a0454 vaddr=40084a94 size=135e8h ( 79336) load
I (359) esp_image: segment 5: paddr=000b3a44 vaddr=50000000 size=00010h (    16) load
I (371) boot: Loaded app from partition at offset 0x10000
I (371) boot: Disabling RNG early entropy source...
I (383) cpu_start: Pro cpu up.
D (383) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 15 bit
D (383) efuse: In EFUSE_BLK0__DATA5_REG is used 1 bits starting with 20 bit
D (389) efuse: In EFUSE_BLK0__DATA3_REG is used 3 bits starting with 9 bit
D (396) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 2 bit
I (403) cpu_start: Starting app cpu, entry point is 0x4008141c
addr2line -pfiaC -e e:\esp\power_saveaaaaaaaaaaa\build\power_save.elf 0x4008141c: [WinError 2] The system cannot find the file specified
I (0) cpu_start: App cpu up.
D (417) clk: RTC_SLOW_CLK calibration value: 3515430
I (430) cpu_start: Pro cpu start user code
I (430) cpu_start: cpu freq: 80000000 Hz
I (430) cpu_start: Application information:
I (435) cpu_start: Project name:     power_save
I (440) cpu_start: App version:      1
I (444) cpu_start: Compile time:     Jan 23 2022 20:08:58
I (450) cpu_start: ELF file SHA256:  0b51434967b48dd2...
I (456) cpu_start: ESP-IDF:          v5.0-dev-1181-g0132079fa
D (463) memory_layout: Checking 7 reserved memory ranges:
D (468) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (475) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb7a50
D (481) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (487) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (494) memory_layout: Reserved memory range 0x40070000 - 0x40078000
addr2line -pfiaC -e e:\esp\power_saveaaaaaaaaaaa\build\power_save.elf 0x40070000: [WinError 2] The system cannot find the file specified
addr2line -pfiaC -e e:\esp\power_saveaaaaaaaaaaa\build\power_save.elf 0x40078000: [WinError 2] The system cannot find the file specified
D (500) memory_layout: Reserved memory range 0x40078000 - 0x40080000
addr2line -pfiaC -e e:\esp\power_saveaaaaaaaaaaa\build\power_save.elf 0x40078000: [WinError 2] The system cannot find the file specified
addr2line -pfiaC -e e:\esp\power_saveaaaaaaaaaaa\build\power_save.elf 0x40080000: [WinError 2] The system cannot find the file specified
D (507) memory_layout: Reserved memory range 0x40080000 - 0x4009807c
[... 407 lines cut ...]
D (531903) esp_netif_lwip: esp_netif_up_api esp_netif:0x3ffbddf4
D (531913) esp_netif_lwip: check: local, if=0x3ffbddf4 fn=0x400dc358
addr2line -pfiaC -e e:\esp\power_saveaaaaaaaaaaa\build\power_save.elf 0x400dc358: [WinError 2] The system cannot find the file specified

D (531913) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbddf4
D (531923) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (531933) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbddf4
D (531933) esp_netif_lwip: esp_netif_get_old_ip_info esp_netif:0x3ffbddf4
D (531943) esp_netif_lwip: check: remote, if=0x3ffbddf4 fn=0x400db24c
[... cut ...]
D (531993) esp_netif_handlers: esp_netif action got_ip with netif0x3ffbddf4 from event_id=0
I (532003) esp_netif_handlers: sta ip: 192.168.1.22, mask: 255.255.255.0, gw: 192.168.1.1
D (532013) event: running post IP_EVENT:0 with handler 0x400d5d60 and context 0x3ffc5174 on loop 0x3ffbcec8
addr2line -pfiaC -e e:\esp\power_saveaaaaaaaaaaa\build\power_save.elf 0x400d5d60: [WinError 2] The system cannot find the file specified
I (531983) wifi:AP's beacon interval = 512000 us, DTIM period = 1
I (532023) example: got ip: 192.168.1.22

AxelLin, 2022-01-25:
> hi @Amralmasry @AxelLin We have fix this issue in master, commit is [0336584](https://github.com/espressif/esp-idf/commit/0336584f3e2be3817769ab2b766b93d0a7dfbc4c), please confirm. We will backport this merge request to v4.1 v4.2 v4.3 v4.4 later.

@mhdong 
I'm confused. 0336584 has subject line "esp_wifi: Refactor key install code".
I don't expect such **refactor** has any bug fix. (commit log does not say anything about **bug fix**)
Does it really fix something? If yes, what was fixed?

ktonder, 2022-01-25:
I just tried to port the 0336584 patch to my code, with all the components and the wpa_supplicant. My device still disconnect each 60 minutes, like clockwork.

kapilkedawat, 2022-01-25:
Hi @Amralmasry, thanks for providing the logs, however supplicant logs were not enabled so not much to make from this.

AP has sent the deauth in the provided logs with reason code 0xc : Mic failure. 

Whenever mic failure occurs, AP and station normally starts countermeasures which renews the keys. Can you please enable the supplicant's logs and set the level as debug so that we will know what's happening here?

@ktonder Please help to provide the logs and sniffer capture to see if that's the same issue.

ktonder, 2022-01-25:
@kapilkedawat 

Thank you for having a second look at the log below. As you can see see below you see that the negotiation is completed at 15:38:18:928, but at 15:38:24:027 it sends the disconnection event. And at 15:38:24:112 I get a "Software caused connection abort"

`[15:32:39:613] ESP-ROM:esp32c3-api1-20210207
[15:32:39:664] Build:Feb  7 2021
[15:32:39:664] rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
[15:32:39:664] SPIWP:0xee
[15:32:39:664] mode:DIO, clock div:1
[15:32:39:664] load:0x3fcd6100,len:0x16bc
[15:32:39:664] load:0x403ce000,len:0x930
[15:32:39:664] load:0x403d0000,len:0x2d40
[15:32:39:664] entry 0x403ce000
[15:32:39:664] <0x1b>[0;32mI (30) boot: ESP-IDF v4.4-rc1-dirty 2nd stage bootloader<0x1b>[0m
[15:32:39:664] <0x1b>[0;32mI (30) boot: compile time 15:16:04<0x1b>[0m
[15:32:39:664] <0x1b>[0;32mI (30) boot: chip revision: 3<0x1b>[0m
[15:32:39:664] <0x1b>[0;32mI (33) boot.esp32c3: SPI Speed      : 80MHz<0x1b>[0m
[15:32:39:664] <0x1b>[0;32mI (37) boot.esp32c3: SPI Mode       : DIO<0x1b>[0m
[15:32:39:664] <0x1b>[0;32mI (42) boot.esp32c3: SPI Flash Size : 4MB<0x1b>[0m
[15:32:39:707] <0x1b>[0;32mI (47) boot: Enabling RNG early entropy source...<0x1b>[0m
[15:32:39:707] <0x1b>[0;32mI (52) boot: Partition Table:<0x1b>[0m
[15:32:39:707] <0x1b>[0;32mI (56) boot: ## Label            Usage          Type ST Offset   Length<0x1b>[0m
[15:32:39:707] <0x1b>[0;32mI (63) boot:  0 nvs              WiFi data        01 02 00009000 00005000<0x1b>[0m
[15:32:39:707] <0x1b>[0;32mI (70) boot:  1 otadata          OTA data         01 00 0000e000 00002000<0x1b>[0m
[15:32:39:707] <0x1b>[0;32mI (78) boot:  2 app0             OTA app          00 10 00010000 00300000<0x1b>[0m
[15:32:39:707] <0x1b>[0;32mI (85) boot:  3 spiffs           Unknown data     01 82 00310000 000f0000<0x1b>[0m
[15:32:39:739] <0x1b>[0;32mI (93) boot: End of partition table<0x1b>[0m
[15:32:39:739] <0x1b>[0;32mI (97) esp_image: segment 0: paddr=00010020 vaddr=3c130020 size=4c528h (312616) map<0x1b>[0m
[15:32:39:800] <0x1b>[0;32mI (154) esp_image: segment 1: paddr=0005c550 vaddr=3fc90600 size=03604h ( 13828) load<0x1b>[0m
[15:32:39:800] <0x1b>[0;32mI (157) esp_image: segment 2: paddr=0005fb5c vaddr=40380000 size=004bch (  1212) load<0x1b>[0m
[15:32:39:800] <0x1b>[0;32mI (160) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=12bae8h (1227496) map<0x1b>[0m
[15:32:39:989] <0x1b>[0;32mI (360) esp_image: segment 4: paddr=0018bb10 vaddr=403804bc size=10088h ( 65672) load<0x1b>[0m
[15:32:39:989] <0x1b>[0;32mI (379) esp_image: segment 5: paddr=0019bba0 vaddr=50000010 size=00010h (    16) load<0x1b>[0m
[15:32:40:033] <0x1b>[0;32mI (385) boot: Loaded app from partition at offset 0x10000<0x1b>[0m
[15:32:40:033] <0x1b>[0;32mI (385) boot: Disabling RNG early entropy source...<0x1b>[0m
[15:32:40:033] <0x1b>[0;32mI (400) cpu_start: Pro cpu up.<0x1b>[0m
[15:32:40:033] D (400) efuse: In EFUSE_BLK2__DATA4_REG is used 3 bits starting with 0 bit<0x1b>[0m
[15:32:40:033] D (400) efuse: In EFUSE_BLK2__DATA4_REG is used 8 bits starting with 12 bit<0x1b>[0m
[15:32:40:033] D (407) efuse: In EFUSE_BLK1__DATA3_REG is used 3 bits starting with 18 bit<0x1b>[0m
[15:32:40:076] D (414) efuse: In EFUSE_BLK1__DATA5_REG is used 5 bits starting with 5 bit<0x1b>[0m
[15:32:40:076] D (421) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 7 bit<0x1b>[0m
[15:32:40:076] D (427) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 14 bit<0x1b>[0m
[15:32:40:076] D (434) efuse: In EFUSE_BLK1__DATA4_REG is used 8 bits starting with 21 bit<0x1b>[0m
[15:32:40:076] D (442) efuse: In EFUSE_BLK1__DATA4_REG is used 3 bits starting with 29 bit<0x1b>[0m
[15:32:40:076] D (449) efuse: In EFUSE_BLK1__DATA5_REG is used 5 bits starting with 0 bit<0x1b>[0m
[15:32:40:076] D (464) clk: RTC_SLOW_CLK calibration value: 3995494<0x1b>[0m
[15:32:40:119] D (472) memprot: esp_memprot_get_pms_lock(IRAM0_SRAM)<0x1b>[0m
[15:32:40:119] D (472) memprot: esp_memprot_get_pms_lock(DRAM0_SRAM)<0x1b>[0m
[15:32:40:119] D (472) memprot: esp_memprot_get_monitor_lock(IRAM0_SRAM)<0x1b>[0m
[15:32:40:119] D (476) memprot: esp_memprot_get_monitor_lock(DRAM0_SRAM)<0x1b>[0m
[15:32:40:119] D (482) memprot: esp_memprot_set_prot(panic_handler: 1, lock: 1, split.addr: 0x00000000, mem.types: 0x00000000<0x1b>[0m
[15:32:40:119] D (492) memprot: esp_memprot_set_monitor_en(IRAM0_SRAM)<0x1b>[0m
[15:32:40:119] D (497) memprot: esp_memprot_set_monitor_en(DRAM0_SRAM)<0x1b>[0m
[15:32:40:162] D (502) memprot: esp_memprot_set_intr_matrix(IRAM0_SRAM)<0x1b>[0m
[15:32:40:162] D (508) memprot: esp_memprot_set_intr_matrix(DRAM0_SRAM)<0x1b>[0m
[15:32:40:162] D (513) memprot: Setting split line MEMPROT_IRAM0_LINE_1_SPLITLINE, addr: 0x40390600<0x1b>[0m
[15:32:40:162] D (521) memprot: Setting split line MEMPROT_IRAM0_LINE_0_SPLITLINE, addr: 0x40390600<0x1b>[0m
[15:32:40:162] D (529) memprot: Setting split line MEMPROT_IRAM0_DRAM0_SPLITLINE, addr: 0x40390600<0x1b>[0m
[15:32:40:162] D (537) memprot: Setting split line MEMPROT_DRAM0_DMA_LINE_0_SPLITLINE, addr: 0x3FC90600<0x1b>[0m
[15:32:40:205] D (545) memprot: Setting split line MEMPROT_DRAM0_DMA_LINE_1_SPLITLINE, addr: 0x3FC90600<0x1b>[0m
[15:32:40:205] D (553) memprot: esp_memprot_iram_set_pms_area(area:MEMPROT_IRAM0_PMS_AREA_0 r:1 w:0, x:1)<0x1b>[0m
[15:32:40:205] D (561) memprot: esp_memprot_iram_set_pms_area(area:MEMPROT_IRAM0_PMS_AREA_1 r:1 w:0, x:1)<0x1b>[0m
[15:32:40:205] D (570) memprot: esp_memprot_iram_set_pms_area(area:MEMPROT_IRAM0_PMS_AREA_2 r:1 w:0, x:1)<0x1b>[0m
[... cut ...]
[15:38:27:284] D (347645) wifi_init_default: Got IP wifi default handler entered<0x1b>[0m
[15:38:27:284] D (347645) esp_netif_handlers: esp_netif action got_ip with netif0x3fcb47d0 from event_id=0<0x1b>[0m
[15:38:27:310] <0x1b>[0;32mI (347655) esp_netif_handlers: sta ip: 192.168.10.171, mask: 255.255.255.0, gw: 192.168.10.1<0x1b>[0m
[15:38:27:326] .I (347745) wifi:Set ps type: 1
[15:38:27:374] 
[15:38:29:158] W (349515) wifi:<ba-add>idx:0 (ifx:0, 62:31:97:60:99:28), tid:0, ssn:7, winSize:64
[15:38:45:150] Scanning, res: 7 found device `

ktonder, 2022-01-25:
It looks like its only a problem when I use the bluetooth in coexistence, tried both with SW coexitence and without.

kapilkedawat, 2022-01-26:
Hi @ktonder 

This looks like a different issue, AP is sending G1 multiple times even after sending G2 to AP. Finally AP is sending the deauth with reason 6.

This should ideally happen with other devices as well. Can you please check with any reference devices if they also behaves in same way.

If it working with other devices, Can you please provide captures for both cases(success and failure). Please start the capture before association and make sure the EAPOL packets are not missing(M1-M4 and the G1-G2).

AxelLin, 2022-02-06:
> Hi @Amralmasry, thanks for providing the logs, however supplicant logs were not enabled so not much to make from this.
> 
> AP has sent the deauth in the provided logs with reason code 0xe : Mic failure.
> 
> Whenever mic failure occurs, AP and station normally starts countermeasures which renews the keys. **Can you please enable the supplicant's logs and set the level as debug so that we will know what's happening here? Also can you please let us know how many devices are connected to it and all are getting disconnected or not?**

@Amralmasry 
Any chance to enable supplicant logs as mentioned above?

jack0c, 2022-05-20:
Sorry, Am I missing anything? Why no one mentions that esp8266 didn't have the issue now?

AxelLin, 2022-05-30:
> Sorry, Am I missing anything? Why no one mentions that esp8266 didn't have the issue now?

https://github.com/espressif/esp-idf/issues/8236#issuecomment-1013127323
And honestly I don't know how is the status of this issue now.

AxelLin, 2022-07-26:
> hi @Amralmasry @AxelLin We have fix this issue in master, commit is [0336584](https://github.com/espressif/esp-idf/commit/0336584f3e2be3817769ab2b766b93d0a7dfbc4c), please confirm. We will backport this merge request to v4.1 v4.2 v4.3 v4.4 later.

@mhdong 
How is the backport status now?
BTW, I have to say that the commit message is so bad that it does not tell what was actually fixed.
Please don't hide the bug fix message in commit log.

AxelLin, 2022-07-26:
> Hi @ktonder
> 
> This looks like a different issue

@ktonder 
Please create a new issue.

mhdong, 2022-08-09:
@AxelLin we will not backport https://github.com/espressif/esp-idf/commit/0336584f3e2be3817769ab2b766b93d0a7dfbc4c to v4.1 and v4.2.

mhdong, 2022-08-09:
@Amralmasry I'm very sorry that I didn't reply in time. The reason why the multicast key update fails is that the router sends a beacon to tell the station by traffic indication map that there is no packet to it, and the station continue to sleep. After receiving the beacon, station will sleep again if there is no data. However, the AP thinks that the station is awake for a short period of time after receiving the beacon.  AP sends a key(group message 1 of 2) and tries 15 times. At this time, the station is sleep. 
After group key update fails, the station can detect a MIC failure. For a MIC failure on a multicast message:
1. Delete the local copy of the group key.
2. Request a new copy of the group key from the authenticator using an EAPOL message (indicates MIC failure).
Turning off power save mode may workaround this problem.
We will design a scheme compatible with this kind of router as soon as possible.

closed 2023-04-24 by Alvin1Zhang