espressif/esp-idf #8237: Random core panics: IllegalInstruction

readmodifywrite, 2022-01-13:
IDF 3.3.5
ESP32-WROVER-32
Chip revision 3
Flash at 40 MHz
Brownout detector set to 2.8V

I'm getting a lot of random core panics, all IllegalInstruction.  They occur randomly and at random PC locations.  Sometimes these locations are valid, some are not.  Here are some example core dumps (ignore the integers on the left, those are time deltas from my terminal monitor):

518893 Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
    0 Memory dump at 0x400e39f0: 3055f663 20c0c121 0528a055
    0 Core 1 register dump:
    0 PC      : 0x400e39f4  PS      : 0x00060b30  A0      : 0x800e3ad5  A1      : 0x3ffc6a80  
   21 A2      : 0x00000002  A3      : 0x3ffc7330  A4      : 0x3ffb431c  A5      : 0x00000000  
    0 A6      : 0x12345678  A7      : 0xaf165a15  A8      : 0x12345678  A9      : 0x3ffc6a10  
   21 A10     : 0x00000000  A11     : 0x000008ba  A12     : 0x000008ba  A13     : 0x3ffb365e  
    0 A14     : 0x3ffb3a78  A15     : 0x00000004  SAR     : 0x0000000a  EXCCAUSE: 0x00000000  
    0 EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
    0 ELF file SHA256: 0000000000000000
    5 Backtrace: 0x400e39f4:0x3ffc6a80 0x400e3ad2:0x3ffc6ac0 0x400e6dd0:0x3ffc6af0
    0 Rebooting...

732771 Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
    0 Core 1 register dump:
    0 PC      : 0x56f8ebe5  PS      : 0x00060930  A0      : 0x800e6dd3  A1      : 0x3ffc6ac0  
   21 A2      : 0x3ffc7340  A3      : 0x3ffb42fe  A4      : 0x00000017  A5      : 0x00000010  
    0 A6      : 0x3ffbc00c  A7      : 0x3ffbc0fc  A8      : 0x96f8ebe5  A9      : 0x3ffc6aa0  
    0 A10     : 0x00000000  A11     : 0x00000001  A12     : 0x3ffc6bc8  A13     : 0x3f4011e8  
   20 A14     : 0x0000008c  A15     : 0x2dc95c03  SAR     : 0x00000012  EXCCAUSE: 0x00000000  
    0 EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
    0 ELF file SHA256: 0000000000000000
    0 Backtrace: 0x56f8ebe5:0x3ffc6ac0 0x400e6dd0:0x3ffc6af0
    0 Rebooting...

60824 Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
    0 Memory dump at 0x400e28e0: 63a02322 c4ca8100 2280220b
    0 Core 1 register dump:
    0 PC      : 0x400e28e5  PS      : 0x00060f30  A0      : 0x800e3988  A1      : 0x3ffc6b80  
   21 A2      : 0x00000013  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f4011e8  
    0 A6      : 0x12345678  A7      : 0x05b63733  A8      : 0x800e3926  A9      : 0x3ffc6b60  
   21 A10     : 0x00000000  A11     : 0x0000008c  A12     : 0x0000008c  A13     : 0x00000000  
    0 A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x0000000a  EXCCAUSE: 0x00000000  
    0 EXCVADDR: 0x00000000  LBEG    : 0x400ec900  LEND    : 0x400ec924  LCOUNT  : 0x00000000  
    0 ELF file SHA256: 0000000000000000
    7 Backtrace: 0x400e28e5:0x3ffc6b80 0x400e3985:0x3ffc6ba0 0x400e6cc0:0x3ffc6bd0 0x4008a3d6:0x3ffc6bf0

115035 Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
    1 Memory dump at 0x400e28e0: 63a02322 c4ca8100 2280220b
    0 Core 1 register dump:
    1 PC      : 0x400e28e5  PS      : 0x00060f30  A0      : 0x800e3988  A1      : 0x3ffc6b80  
   19 A2      : 0x00000013  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f4011e8  
    1 A6      : 0x12345678  A7      : 0x08f4612f  A8      : 0x800e3926  A9      : 0x3ffc6b60  
   18 A10     : 0x00000000  A11     : 0x0000008c  A12     : 0x0000008c  A13     : 0x00000000  
    0 A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x0000000a  EXCCAUSE: 0x00000000  
    0 EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
    0 ELF file SHA256: 0000000000000000
    6 Backtrace: 0x400e28e5:0x3ffc6b80 0x400e3985:0x3ffc6ba0 0x400e6cc0:0x3ffc6bd0 0x4008a3d6:0x3ffc6bf0
    0 Rebooting...

I have a lot of these, and so far I'm only seeing this on one unit.  I'm wondering if this is just a manufacturing defect in the module?

What else can cause this?  I've used the same power supply design on other boards without issue, and the brown out detector is not triggering, so it doesn't look like a power supply issue.

I do not have the flash pins connected external to the module.

I never return from a FreeRTOS task (since not using vTaskDelete will apparently cause this) - my tasks never terminate.

atanisoft, 2022-01-14:
This could be a watchdog timeout panic reboot. Can you decode the backtraces to determine what code is being called at the time it aborted?

readmodifywrite, 2022-01-14:
It's not a watchdog, that would trigger a different exception.  The PC is different each time, and sometimes it's an invalid address.  It looks like the memory bus gets corrupted on an instruction fetch, at a seemingly random time.

So far it seems somewhat correlated to wifi usage in station mode.  Running as a soft AP doesn't seem to trigger it.

igrr, 2022-01-17:
@readmodifywrite Were all the panic messages produced with the same application .elf file? If yes, could you please share your .elf file? This would help figure out if this is a code generation issue or a runtime issue. (Since this only happens on one board, i guess it's the latter, but just to be sure...)

readmodifywrite, 2022-01-17:
@igrr Before I waste any more of your time on this one, I'm going to run a test across multiple devices to see if it is actually a systemic issue, or just an isolated hardware failure.  Right now I am strongly suspecting the latter.

readmodifywrite, 2022-01-20:
Ok, I'm pretty sure this is just a hardware fault on the module.  I replaced it and it seems to be fine.

closed 2022-01-20 by readmodifywrite