espressif/esp-idf #8239: Watchog is triggered too soon

Arthedian, 2022-01-14:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): v4.3.2
- Operating System:     macOS
- Using an IDE?: No
- Power Supply:         external 3.3V

## Problem Description

If init watchdog to be triggered in 43200 seconds, than it is triggering already after 251 seconds.

### Expected Behavior

The esp should be triggered after 43200 seconds

### Actual Behavior

it is triggering already after 251 seconds.

### Steps to reproduce

```
void testa_task(uint16_t cron_data) {
    int ret;
    esp_err_t err;
    err = esp_task_wdt_add(NULL);
    ESP_LOGD(TAG, "err when adding wdt %d", err);
    int counter = 0;
    while(1){
        counter++;
        ESP_LOGD(TAG, "counter %d", counter);
        vTaskDelay(1000 / portTICK_PERIOD_MS);

    }
}

void testa_init() {
    xTaskCreate(testa_task, "testa_task", 1024 * 8, NULL, 5, NULL);
    while (1){
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

void app_main(void) {
    esp_err_t err;
    err = esp_task_wdt_init(43200, true);
    eepromInit();
    testa_init();
}
```

ESP-Marius, 2022-01-17:
Hi,

This is probably due to an overflow during the timeout calculation. It was fixed on master (see https://github.com/espressif/esp-idf/issues/6648 for context) but seems like we missed the backport.

If you want to test the fix/solve it immediately you could apply the minor changes from b23de0f 

Thanks for letting us know about this issue, I've created a backport to 4.3 for this.

Arthedian, 2022-01-17:
If I pull latest 4.3 branch, that should be enough right?

ESP-Marius, 2022-01-17:
> If I pull latest 4.3 branch, that should be enough right?

Unfortunately not. The MR for v4.3 is made, but still needs to be merged. And after merging release/v4.3 still needs to go through automated testing before it gets synced to github. This issue should be automatically updated as soon as the fix is on github though.

AxelLin, 2022-01-17:
> If I pull latest 4.3 branch, that should be enough right?

After pulling 4.3 branch, you can checkout to a new test branch.
In your test branch, run "git cherry pick b23de0f" and verify the result.

AxelLin, 2022-01-18:
> If I pull latest 4.3 branch, that should be enough right?

It's merged to v4.3 branch. so pull latest 4.3 branch should work now.

Alvin1Zhang, 2022-01-24:
@Arthedian Thanks for reporting, fix on release/4.3 is available https://github.com/espressif/esp-idf/commit/0786d1a337ae3b190497347bab87b509297dcc0f, feel free to reopen if the issue still happens.

closed 2022-01-24 by Alvin1Zhang