espressif/esp-idf #8240: Excluding (not using) esp_sleep_enable_timer_wakeup() increases current consumption

boborjan2, 2022-01-14:
----------------------------- Delete below -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC (reproducible on custom board as well)
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32 (reproducable on ESP32-WROOM-E too)
- IDF version (run ``git describe --tags`` to find it): v4.2-1138-g7ad199d14a
- Build System:         make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB (external for custom board)

## Problem Description
Entering to light sleep without having timer as a wakeup source increases current consumption by ~500uA.

### Expected Behavior
Equal or lower current consumption . Or - if it is a chip feature - documenting the behavior.

### Actual Behavior
If we do not use timer as a wakeup (i.e. do not call esp_sleep_enable_timer_wakeup()), current consumption goes up by ~500uA.

### Steps to reproduce
Stock light sleep example, flashed on devkitc. Current consumption depends on whether gpio and/or timer is enabled as a wakeup source. Please check the foillowing table:
timer off, gpio off: 103.2mV @ 11R series resistor
timer off, gpio on: 103.6mV
timer on, gpio off: 98.4mV
timer on, gpio on: 98.8mV

### Code to reproduce this issue
https://github.com/espressif/esp-idf/blob/release/v4.2/examples/system/light_sleep/main/light_sleep_example_main.c

The issue is reproducible on a custom board based on ESP32-WROOM-E. (this is where we first observed thew phenomenon)

Thanks,
Viktor

boarchuz, 2022-01-14:
I've also observed this in a slightly different way. In a minimal program to test light sleep current, entering infinite light sleep in these two different ways had very different sleep currents:

- esp_light_sleep_start ~950uA

- enable tickless idle then vTaskDelay: ~450uA

IDF: ebf7e811b12e3c1e347340e5b9ec014e9c6319ba

esp-lis, 2022-01-17:
> ----------------------------- Delete below -----------------------------
>
> ## Environment
> * Development Kit:      ESP32-DevKitC (reproducible on custom board as well)
> * Kit version (for [...]

I'm guessing it might be the GPIO leakage that increases the current consumption by about 500 uA, I need more detailed test environment information. For example, which wake-up sources are enabled in tests of different current consumption?

esp-lis, 2022-01-17:
> I've also observed this in a slightly different way. In a minimal program to test light sleep current, entering infinite light sleep in these two different ways had very different sleep currents:
>
> * [...]

Can you describe your test environment in detail?

boborjan2, 2022-01-17:
Test in detail: take a stock DevKitC v4 board. No connection to any pins. USB 5V+ has a 11R in series on which I will measure voltage drop aka supply current.
Load stock light sleep example.

In the test, there are two calls for enabling wakeup sorces: esp_sleep_enable_timer_wakeup(), esp_sleep_enable_gpio_wakeup(). I made 4 tests:
1. esp_sleep_enable_timer_wakeup() and esp_sleep_enable_gpio_wakeup() both commented out (not called)
2. only esp_sleep_enable_timer_wakeup() is commented out (ie not called)
3. only esp_sleep_enable_gpio_wakeup() is commented out
4. both enabled (original behavior)

I measured the following voltages in the test cases:
1. timer off, gpio off: 103.2mV @ 11R series resistor
2. timer off, gpio on: 103.6mV
3. timer on, gpio off: 98.4mV
4. timer on, gpio on: 98.8mV

I can reproduce it on a custom board with custom connections on most pins. Same behavior.

I do not think it is leakage. My guess now is that if timer wakeup is enabled, somehow the flash gets turned off and that's the cause of the surplus consumption when timer wakeup is disabled.  

Viktor

boarchuz, 2022-01-17:
Hi @esp-lis,
I've repeated the measurements with a new WROOM-32D and a new WROOM-32E, with a much simpler test (current is higher, but effect is the same).

Setup is a new module, mounted on a breakout board (no other components), ESP-PROG to program/monitor, bench power supply for 3.3V through a Joulescope as ammeter. In each case I also confirmed the current with the ESP-PROG disconnected and EN tied to 3V3.

main.c:
```c
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_pm.h"
#include "esp_sleep.h"

// Uncomment to use tickless idle for lower sleep current
// #define IDF8240_USE_TICKLESS_IDLE 1

#if IDF8240_USE_TICKLESS_IDLE && !CONFIG_FREERTOS_USE_TICKLESS_IDLE
    #error "Enable PM & tickless idle"
#endif

static const char *TAG = "IDF8240";

void app_main(void)
{
    #if IDF8240_USE_TICKLESS_IDLE
        ESP_LOGI(TAG, "tickless idle");
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        const esp_pm_config_esp32_t config = {
            .max_freq_mhz = CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ,
            .min_freq_mhz = CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ,
            .light_sleep_enable = 1,
        };
        ESP_ERROR_CHECK(esp_pm_configure(&config));
    #else
        ESP_LOGI(TAG, "esp_light_sleep_start");
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        esp_light_sleep_start();
    #endif
    vTaskDelay(portMAX_DELAY);
}

```
sdkconfig.defaults
```
CONFIG_PM_ENABLE=y
CONFIG_FREERTOS_USE_TICKLESS_IDLE=y
```
IDF: v5.0-dev-1188-g3ba47f1628

Results:
* esp_light_sleep_start
  * 32D: ~1170uA
  * 32E: ~1300uA
* enable tickless idle then vTaskDelay
  * 32D: ~1730uA (+560uA)
  * 32E: ~2080uA (+700uA)

FWIW I don't think the revision is significant; I see very different light sleep current baselines for 3x32Ds from the same batch.

esp-lis, 2022-01-26:
> Hi @esp-lis, I've repeated the measurements with a new WROOM-32D and a new WROOM-32E, with a much simpler test (current is higher, but effect is the same).
>
> Setup is a new module, mounted on a [...]

I recommend migrating to release/v4.3 or newer for testing. For GPIO leakage, added PM_SLP_DISABLE_GPIO option, its location is (Top) -> Component config -> Power Management -> Disable all GPIO when chip at sleep

boarchuz, 2022-01-29:
> I recommend migrating to release/v4.3 or newer for testing. For GPIO leakage, added PM_SLP_DISABLE_GPIO option, its location is (Top) -> Component config -> Power Management -> Disable all GPIO when chip at sleep

I am seeing the same phenomenon on release/v4.4.

Enabling PM_SLP_DISABLE_GPIO further reduces sleep current in tickless idle, but not when using esp_light_sleep_start. Even if I 'manually' disable all GPIO using the following code (which reduces light sleep current in _both_ cases) there is still the issue of ~500uA higher current when using esp_light_sleep_start over tickless idle.
```
#include "driver/gpio.h"

#define GPIO_IS_INTERNAL_FLASH_PIN(gpio_num) ((gpio_num) >= 6 && (gpio_num) <= 11)

for(int i = 0; i < SOC_GPIO_PIN_COUNT; ++i) {
  if(GPIO_IS_VALID_GPIO(i) && !GPIO_IS_INTERNAL_FLASH_PIN(i)) {
    gpio_reset_pin(i);
  }
}
```

With this particular ESP32-WROOM-32E, I get 751uA using tickless idle and 1121uA using esp_light_sleep_start (+370uA).

negativekelvin, 2022-01-29:
>My guess now is that if timer wakeup is enabled, somehow the flash gets turned off and that's the cause of the surplus consumption when timer wakeup is disabled.

@boborjan2 that seems correct, vddsdio is not powered down if s_config.sleep_duration is not set
https://github.com/espressif/esp-idf/blob/bf8d4d55d0c004a5c52b7721d6ae0484565e74fa/components/esp_hw_support/sleep_modes.c#L663-L676

boarchuz, 2022-02-07:
Can confirm this for the tickless idle case too. Fixed with changes by @negativekelvin.

AxelLin, 2022-03-02:
> Can confirm this for the tickless idle case too. Fixed with changes by @negativekelvin.

@esp-lis 
Would you confirm and pick up the fix? Or someone from espressif can take a look?

AxelLin, 2022-04-06:
@Alvin1Zhang 
Can someone take a look at the fix in https://github.com/espressif/esp-idf/issues/8240#issuecomment-1024926419 ?

lennyJL, 2022-04-06:
Currently in IDF, whether the flash can be powered down depends on whether the RTC timer wakeup source is enabled or not. About the behavior of the `flash power domain` when the RTC timer wakeup source is not enabled, I will provide an MR later.

The fix in https://github.com/espressif/esp-idf/issues/8240#issuecomment-1024926419 has a little problem. It may cause the `sleep_time_adjustment` to have an incorrect value when going to sleep for the first time.

jack0c, 2022-04-24:
That is say if there are any possibilities waked up by in uncertain time(especially the time is short enough to make flash can't power up again),  we can't power down flash by VDDSDIO. Currently we just check wake up timer, which looks not enough, we need check if there are other wake up source enabled, for instance external wake up sources UART, GPIO. This fix will added later.
Whether the fix from @negativekelvin can be used in projects is also depends on the flash chip, if the flash doesn't have the issue, the fix can be used.I think we can also give this choice to the users, for instance adding a FORCE_POWER_DOWN_FLASH in Kconfig.
However, still want to check if +370uA is all from VDDSDIO. @boarchuz

negativekelvin, 2022-04-24:
>if there are any possibilities waked up by in uncertain time(especially the time is short enough to make flash can't power up again), we can't power down flash by VDDSDIO

I don't think so, code waits for flash chip to start up so only issue is if user needs a faster wake up time

lennyJL, 2022-04-24:
Add some information:
There is a **small probability**(`< 1%`) that the flash will not work properly when powered up again during power-down period.

negativekelvin, 2022-04-24:
Ok, so can we make sure that flash is powered down before entering sleep by doing it early since sleep start functions are in iram?

negativekelvin, 2022-04-28:
There are a lot of cases when sleep time is unpredictable, for example external interrupt from sensor. It may be occasionally very short time but usually long time so it is important to be able to power down flash.

boborjan2, 2022-05-11:
From 4.3 there is already an option for explicitly power down flash: CONFIG_ESP_SYSTEM_PD_FLASH. Is this issue still present there? I have been working with 4.2 only so far.
I cannot comment on the wakeup issues of the flash in case of a short sleep time but I agree that sleep times are unpredictable in general. So it is essential to make flash power down reliable in any case.

The original issue could solved however by disabling the implicit flash powerdown if timer wakeup enabled (To prevent unexpected/undocumented higher wake-up times) - I still do not understand this behavior.
It would be great to add means for explicit flash powerdown (backporting the 4.3 feature) - provided that it works 100%.

lennyJL, 2022-05-13:
We will provide users with a mechanism to power down the flash under more relaxed conditions.

AxelLin, 2022-10-13:
> Hi @boborjan2. Currently, even if the `power down flash` is configured in the menuconfig, it is not guaranteed to work 100%. As we discussed above, power down flash requires some conditions to keep safe.
> 
> We will provide users with a mechanism to power down the flash under more relaxed conditions. 

@lennyJL  Is there any update for this improvement?

lennyJL, 2023-03-10:
Hi @AxelLin @boborjan2, there is a [doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#power-down-of-flash) about the power-down of flash in `ESP-IDF Programming Guide`.

lennyJL, 2023-03-10:
@boborjan2, Can you force the flash to be powered down, and then measure whether the current is normal?

boarchuz, 2023-03-11:
@lennyJL Taking another look at this, there seems to be an issue with the SDIO power down logic.

`can_power_down_vddsdio` is only used after checking the RTC_SLEEP_PD_VDDSDIO bit here:
https://github.com/espressif/esp-idf/blob/47852846d3e89580ef4da28210b6ffc1cb5eaa9d/components/esp_hw_support/sleep_modes.c#L867
So the return value of `can_power_down_vddsdio` is always equivalent to `force_pd`:
https://github.com/espressif/esp-idf/blob/47852846d3e89580ef4da28210b6ffc1cb5eaa9d/components/esp_hw_support/sleep_modes.c#L763

This doesn't match the logic described for CONFIG_ESP_SLEEP_POWER_DOWN_FLASH in the linked doc.

For example, with CONFIG_ESP_SLEEP_POWER_DOWN_FLASH=y, this will unexpectedly power down the flash, and holding IO0 low creates a potentially unsafe power-up-and-partial-discharge loop on VDD_SDIO:
```
for(;;) {
    esp_sleep_enable_ext0_wakeup(0, 0);
    esp_light_sleep_start();
}
```

Also a minor issue is that `esp_sleep_pd_config(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_OFF)` will assert here without a preceding `esp_sleep_pd_config(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_ON)` to increment `refs`:
https://github.com/espressif/esp-idf/blob/47852846d3e89580ef4da28210b6ffc1cb5eaa9d/components/esp_hw_support/sleep_modes.c#L1459

AxelLin, 2023-04-21:
@lennyJL Any comment regarding https://github.com/espressif/esp-idf/issues/8240#issuecomment-1464822111 ?

AxelLin, 2023-05-02:
> @lennyJL Any comment regarding [#8240 (comment)](https://github.com/espressif/esp-idf/issues/8240#issuecomment-1464822111) ?

@lennyJL @jack0c 
The comments from https://github.com/espressif/esp-idf/issues/8240#issuecomment-1464822111 looks make sense to me.
What's your comment? (It could be a bug in current code)

10086loutianhao, 2023-07-20:
> @lennyJL Taking another look at this, there seems to be an issue with the SDIO power down logic.
>
> `can_power_down_vddsdio` is only used after checking the RTC_SLEEP_PD_VDDSDIO bit here: [...]

@AxelLin @boarchuz As mentioned earlier, powering down the flash during light sleep introduces risks. Therefore, if customers want to force the flash to be turned off, they must ensure that their own code allows the flash to fully power down. In the example provided above, I believe that this is something we cannot avoid at the software level. We can only encourage customers to consider the risks introduced by flash power-down when designing their applications.

10086loutianhao, 2023-07-20:
Exactly, when designing our application, we need to consider the required sleep time to determine whether to power down the flash. If the sleep time is long, it makes sense to power down the flash as it can reduce power consumption. However, if the sleep time is short, powering down the flash may actually extend our active time (due to waiting for the flash to be ready after powering up) and increase power consumption.

10086loutianhao, 2024-01-23:
[image]
note: why we power down VDD_SDIO can reduce lots of current comsumption is beacause some IO powered by VDD_SDIO,
I believe there might be a leakage current in the above IO

Sherry616, 2024-04-23:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2024-04-23 by Sherry616

powerfeatherdev, 2024-07-26:
@lennyJL 

Hi, this issue is closed, but I have a question about the more relaxed condition for powering down flash:

>We will provide users with a mechanism to power down the flash under more relaxed conditions.

In the docs, it mentions:

>Calling esp_sleep_pd_config(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_OFF) powers down flash when the RTC timer is not enabled as a wakeup source *or* the sleep time is longer than the flash power-down period.

So, is it ok for this relaxed condition to: set  GPIO as wake source for example, even if there is a chance that the GPIO wake condition occurs before the flash can be powered down completely.