espressif/esp-idf #8241: `esp_wifi_start` stops USB on ESP32-S3

tannewt, 2022-01-14:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version DevKitC: v1
- Module or chip used:  ESP32-S3-WROOM-1 D2N8R2
- IDF version (run ``git describe --tags`` to find it):
    release v4.4 commit 214d62b9ad859318520e38870c828278c2caee2a
- Build System:         CMake-ish
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32s3-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

In CircuitPython we initialize wifi dynamically when the user does `import wifi`. We initialize the USB before this happens. Now, running `esp_wifi_start` stops the USB. `tio` shows the serial connection as disconnected.

### Expected Behavior
Wifi should start and USB should stay connected.

### Actual Behavior
USB disconnects but code seems to continue to run.

### Steps to reproduce

1. https://learn.adafruit.com/building-circuitpython
2. Connect to the REPL over USB serial. Run `import wifi`.

### Code to reproduce this issue

`import wifi` in CircuitPython.

## Debug Logs

No crash.

```
I (7964) pp: pp rom version: e7ae62f
I (7964) net80211: net80211 rom version: e7ae62f
I (7974) wifi:wifi driver task: 3fce4ee8, prio:23, stack:3584, core=0
I (7974) system_api: Base MAC address is not set
I (7974) system_api: read default base MAC address from EFUSE
I (7984) wifi:wifi firmware version: 5c3d3bf
I (7984) wifi:wifi certification version: v7.0
I (7994) wifi:config NVS flash: disabled
I (7994) wifi:config nano formating: disabled
I (7994) wifi:Init data frame dynamic rx buffer num: 8
I (8004) wifi:Init management frame dynamic rx buffer num: 8
I (8004) wifi:Init management short buffer num: 32
I (8014) wifi:Init dynamic tx buffer num: 16
I (8014) wifi:Init tx cache buffer num: 16
I (8024) wifi:Init static tx FG buffer num: 2
I (8024) wifi:Init static rx buffer size: 1600
I (8024) wifi:Init static rx buffer num: 4
I (8034) wifi:Init dynamic rx buffer num: 8
I (8034) wifi_init: rx ba win: 6
I (8044) wifi_init: tcpip mbox: 32
I (8044) wifi_init: udp mbox: 6
I (8044) wifi_init: tcp mbox: 6
I (8054) wifi_init: tcp tx win: 2880
I (8054) wifi_init: tcp rx win: 2880
I (8064) wifi_init: tcp mss: 1440
I (8064) wifi_init: WiFi IRAM OP enabled
I (8064) wifi_init: WiFi RX IRAM OP enabled
I (8074) phy_init: phy_version 302,3ddfab3,Dec  3 2021,14:39:12
E (8084) phy_init: esp_phy_load_cal_data_from_nvs: NVS has not been initialized. Call nvs_flash_init before starting WiFi/BT.
W (8094) phy_init: failed to load RF calibration data (0x1101), falling back to full calibration
I (8144) wifi:mode : sta (7c:df:a1:e0:76:e0)
I (8144) wifi:enable tsf
W (8144) wifi: sta start
```

## Other items if possible

[firmware.elf.zip](https://github.com/espressif/esp-idf/files/7872684/firmware.elf.zip)
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/7872685/sdkconfig.zip)

tannewt, 2022-01-14:
CircuitPython issue is here: https://github.com/adafruit/circuitpython/issues/5662

igrr, 2022-01-14:
I think this was fixed on `master` by https://github.com/espressif/esp-idf/commit/582da32f4a7061c3717158b979c5dd8092c906e2 and we backported the commit to release/v4.4 in https://github.com/espressif/esp-idf/commit/81ce15f0e088fd28578d340287fdaf7f28879efd. However we have made a mistake while backporting. We have overlooked the fact that the dependency on a soc_caps macro `depends on USB_OTG_SUPPORTED` has just been implemented in v5.0-dev and doesn't work in release/v4.4 branch. Unfortunately, kconfiglib didn't warn about non-existent symbol in the dependency.

Could you please try replacing https://github.com/espressif/esp-idf/blob/f3e0c8bc41098f79baf2cf9f45dcbb6c1d41b3fe/components/esp_phy/Kconfig#L109 with
```
depends on IDF_TARGET_ESP32S2 || IDF_TARGET_ESP32S3 || ESP_CONSOLE_USB_SERIAL_JTAG 
```
and give it a try?

tannewt, 2022-01-14:
Thanks! That does get it past the import statement. I haven't actually tested the WiFi yet but USB continues to work.

tannewt, 2022-01-14:
WiFi works too. :+1: We have a branch under adafruit/esp-idf that we'll patch until we get it on the 4.4 branch.

Dazza0, 2022-03-04:
Backported to v4.4 in commit bf523af21d89577ec5c95335ce7a3bc1ba61886c.

closed 2022-03-04 by Dazza0