espressif/esp-idf #8246: Saving to NVS crashes

silardg, 2022-01-17:
## Environment
- Module or chip used:  ESP32-WROOM-32D
- IDF version:
    v2.12-rc3-292-g3eda077
- Build System: idf.py

I'm spamming my mqtt client with messages, and I'm getting this output when I try to save a value to the nvs.
I'm using a wrapper for the nvs.

## Debug Logs

```
abort() was called at PC 0x401c8fe2 on core 0
0x401c8fe2: __wrap__Unwind_Resume_or_Rethrow at ??:?

Backtrace:0x40091e8e:0x3ffdf920 0x400927b1:0x3ffdf940 0x400993f6:0x3ffdf960 0x401c8fe2:0x3ffdf9d0 0x401ae257:0x3ffdf9f0 0x401ae1be:0x3ffdfa10 0x401af4b5:0x3ffdfa30 0x400dc7ab:0x3ffdfa50 0x4012e8d2:0x3ffdfa80 0x4012ddd1:0x3ffdfb40 0x4012de4d:0x3ffdfb80 0x400e33ae:0x3ffdfbb0 0x400dc98f:0x3ffdfbf0 0x400dbc7b:0x3ffdfc50 0x40130faa:0x3ffdfca0 0x40131bf9:0x3ffdfcf0 0x40132804:0x3ffdfd10
0x40091e8e: panic_abort at /Users/silardgal/esp/esp-idf/components/esp_system/panic.c:379

0x400927b1: esp_system_abort at /Users/silardgal/esp/esp-idf/components/esp_system/system_api.c:112

0x400993f6: abort at /Users/silardgal/esp/esp-idf/components/newlib/abort.c:46

0x401c8fe2: __wrap__Unwind_Resume_or_Rethrow at ??:?

0x401ae257: __cxa_throw at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:90

0x401ae1be: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54

0x401af4b5: __gnu_cxx::new_allocator<char>::allocate(unsigned int, void const*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/ext/new_allocator.h:111
 (inlined by) std::allocator_traits<std::allocator<char> >::allocate(std::allocator<char>&, unsigned int) at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/bits/alloc_traits.h:436
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned int&, unsigned int) at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/bits/basic_string.tcc:153

0x400dc7ab: void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag) at /Users/silardgal/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.tcc:219

0x4012e8d2: void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct_aux<char*>(char*, char*, std::__false_type) at /Users/silardgal/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:236
 (inlined by) void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*) at /Users/silardgal/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:255
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /Users/silardgal/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:440
 (inlined by) IO::storeOutputValueuint8_t(unsigned int, unsigned char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) at /components/IO/IO.cpp:180
```

igrr, 2022-01-17:
@silardg from the backtrace, this doesn't look directly NVS related... It looks like in `IO::storeOutputValueuint8_t` (which I assume is in your code) you are creating an std::string. This fails due to lack of RAM, and the system aborts.

* Can you check that you have sufficient heap at this point?
* If you'd like to handle allocation failures gracefully in your code, you can enable C++ exception handling and catch `std::bad_alloc` exception. Alternatively, you can allocate required space with `malloc`, check for allocation failure, otherwise call "placement new" to construct the object in the allocated space. (This might not work for std::string specifically, but could be useful in other cases.)

silardg, 2022-01-17:
@igrr Thanks! Will check. 
Closing for now.

closed 2022-01-17 by silardg