espressif/esp-idf #8249: Current consumption on ESP32C3FN4 chip

vAnArhist, 2022-01-17:
## Environment

- Development Kit: none
- Module or chip used:  ``ESP32-C3FN4``
- IDF version ``v5.0-dev-1172-g6050388f51``
- Build System:  ``idf.py``
- Compiler version: **riscv32-esp-elf-gcc** ``8.4.0``
- Operating System: Windows
- (Windows only) environment type: PowerShell.
- Using an IDE?: No
- Power Supply:  external 3.3V

## Problem Description
According to this esp32c3 datasheet: [https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf](url) 
Current consumption must be ~5uA. 
I used deep_sleep example from main branch, configured wakeup from timer for 20sec (not from RTC GPIO, as in example), 
flashed it, but current consumption is nearly ~220-230uA. Why is it so high? Can I decrease that?

[image]

### Expected Behavior
Current in few uA in deep sleep
### Actual Behavior
Current - 220uA
### Steps to reproduce

1. Connect multimeter to board in current measurement mode
2. Uncheck tag *Enable wakeup from GPIO* as in screenshot
[image]
3. Compile/flash board, check measurements

### Code to reproduce this issue

https://gist.github.com/vAnArhist/fb536eaa0e89a108a224841b365113a2

## Debug Logs

```
I (48) boot: ESP-IDF v5.0-dev-1172-g6050388f51-dirty 2nd stage bootloader
I (49) boot: compile time 18:59:44
I (49) boot: chip revision: 3
I (52) boot.esp32c3: SPI Speed      : 80MHz
I (57) boot.esp32c3: SPI Mode       : DIO
I (62) boot.esp32c3: SPI Flash Size : 4MB
I (67) boot: Enabling RNG early entropy source...
I (72) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=06ed0h ( 28368) map
I (80) esp_image: segment 1: paddr=00016ef8 vaddr=3fc8b000 size=01534h (  5428) load
I (89) esp_image: segment 2: paddr=00018434 vaddr=40380000 size=07be4h ( 31716) load
I (99) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=130d0h ( 78032) map
I (105) esp_image: segment 4: paddr=000330f8 vaddr=40387be4 size=0337ch ( 13180) load
I (115) esp_image: segment 5: paddr=0003647c vaddr=50000000 size=00014h (    20) 
I (122) esp_image: segment 6: paddr=00036498 vaddr=50000018 size=00018h (    24) 
I (130) boot: Fast booting app from partition at offset 0x10000
I (137) boot: Disabling RNG early entropy source...
I (154) cpu_start: Pro cpu up.
I (162) cpu_start: Pro cpu start user code
I (162) cpu_start: cpu freq: 160000000 Hz
I (162) cpu_start: Application information:
I (165) cpu_start: Project name:     deep_sleep
I (170) cpu_start: App version:      1
I (175) cpu_start: Compile time:     Jan 17 2022 20:23:30
I (181) cpu_start: ELF file SHA256:  e07f3be3421fed05...
I (187) cpu_start: ESP-IDF:          v5.0-dev-1172-g6050388f51-dirty
I (194) heap_init: Initializing. RAM available for dynamic allocation:
I (201) heap_init: At 3FC8D3B0 len 00032C50 (203 KiB): DRAM
I (207) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (214) heap_init: At 50000030 len 00001FC0 (7 KiB): RTCRAM
I (221) spi_flash: detected chip: generic
I (225) spi_flash: flash io: dio
I (229) sleep: Configure to isolate all GPIO pins in sleep state
I (236) sleep: Enable automatic switching of GPIO sleep configuration
I (243) cpu_start: Starting scheduler.
Wake up from timer. Time spent in deep sleep: 20248ms
Enabling timer wakeup, 20s
Entering deep sleep
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x5 (DSLEEP),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x181c
load:0x403ce000,len:0x934
```
## Other items if possible

[dumps.zip](https://github.com/espressif/esp-idf/files/7884074/dumps.zip)

krzychb, 2022-01-18:
Hi @vAnArhist, could you please post a schematic and possibly a photo of your setup to measure the current?

vAnArhist, 2022-01-18:
@krzychb Hello!
Here it is. First screenshot shows controller itself, second is the step-up from battery supply. 
For now all schematic supplied from an external 3.3 step-down converter. All peripheral signals(GPIO signals) are cutted, so there cannot be any leakage on them.
Unfortunately, I cannot share photos of board and all schematic, as it is prototype of commercial product.

Measurement performed by multimeter(UNI-T UT61A) in current measurement mode. You can see setup on 2nd screenshot.

[image]
[image]

krzychb, 2022-01-19:
Thank you for posting the schematic. Our colleague measured the power consumption of a new ESP32-C3 FN4 chip on a test board with [dumps.zip](https://gist.github.com/vAnArhist/fb536eaa0e89a108a224841b365113a2) example compiled with the latest ESP-IDF from the master branch and got the value slightly below 5 µA. 

- Did you consider placing the ammeter after DC/DC BL8531 converter? Maybe the converter is faulty and takes too much current?
- Could you try another ESP32-C3 FN4 chip?

vAnArhist, 2022-01-20:
@krzychb Thanks for your time! 
We also tested deep sleep example with cut all peripheral on ESP32-C3-DevKitM-1. Current is really below 5-7 µA.
1. I tested also that example with adding some additional initialization we used in order to off transistors (used in our periph), with that API. Note that we used that api for GPIO, which are not RTC powered:
```
ESP_ERROR_CHECK(gpio_hold_en(pins_num));	
gpio_deep_sleep_hold_en();
``` 
After that, we found current increased dramatically to 19mA on cutted ESP32-C3-DevKitM-1 without any additional periph.
2. Now, we are working on physical remapping all control GPIO to RTC_GPIO, as it is unclear what state is in deep sleep.

Regarding this misunderstanding with GPIOs, I have few questions:
1. Why GPIO pins, that are connected to CPU-power domain (VDD3P3_CPU), can stay high-level in deep sleep (CPU is off in deep sleep, isn`t it) ?
2. What is the difference between RTC_GPIO and other pins?
3. What pins we must pullup externally in order to make the best performance in deep sleep? Maybe, Espressif has any reference schematic? 
4. Why internal SPI flash pins routed externally? Could we use the same pins for SPI chip if we change CS? If yes, how? For now, if I initialize SPI2 on the same pins with internal memory program just hangs.

krzychb, 2022-01-25:
> Why GPIO pins, that are connected to CPU-power domain (VDD3P3_CPU), can stay high-level in deep sleep (CPU is off in deep sleep, isn`t it) ?

CPU off does not mean the whole power domain is off, it's still on for Digital GPIOs. We will update the power "Power Scheme" figure in the datasheet to show it in a similar way as already shown in the [ESP32-S3 dataseet (see Figure 4)](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf). 

> What is the difference between RTC_GPIO and other pins? 

RTC GPIO pins are dedicated to low-power applications. Again, this is better shown in the newer datasheet for ESP32-S3 in [Figure 1: Block Diagram of ESP32­-S3](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf). We will update that one as well.

> What pins we must pullup externally in order to make the best performance in deep sleep? Maybe, Espressif has any reference schematic?

We do not have such a reference design but can offer some suggestions:

- the pad should be in a certain state (pull-up or pull-down)
- ensure that there's no voltage difference between the pad and external device, because if yes, there'll be extra current. We suggest keeping the external device in a high impedance state and make sure the pad is in a certain state (as stated in the first point)

> Why internal SPI flash pins routed externally? Could we use the same pins for SPI chip if we change CS? If yes, how? For now, if I initialize SPI2 on the same pins with internal memory program just hangs.

- the routing of SPI flash pins is because of chip package consideration (we also have ESP8685 which does not route the pins externally)
- we don't recommend using the same pins with SPI flash for other devices, this may impact the code running in the flash

vAnArhist, 2022-01-25:
@krzychb Thanks for info! Nice to hear about updating docs. 
I will close this issue, as it`s related with our board.

closed 2022-01-25 by vAnArhist