espressif/esp-idf #8251: esptool upload failure on M1 arm64 Mac

oclyke, 2022-01-18:
## Environment

- Development Kit: None - using same auto bootload circuit as ESP32 dev kits and Silicon Labs CP2102 USB-serial converter
- Kit version: Silicon Labs CP2102
- Module or chip used:  [ESP32-PICO-D4]
- IDF version: v5.0-dev-1201-g124b5f5245
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: macOS
- Using an IDE?: No
- Power Supply: USB

## Problem Description
Failure to upload compiled image using esptool.py on Mac with M1 arm64 based processor

### Expected Behavior
esptool.py can connect to ESP32 and completes upload

(expected behavior confirmed on fresh install of esp-idf at the same commit on Linux with the same hardware used on Mac)

### Actual Behavior
esptool.py fails to connect to board and outputs error message about boot mode:

```
A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x3f)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

### Steps to reproduce

```
cd ~/esp-idf/examples/get-started/hello-world
idf.py -p {PORT} flash monitor
```

### Code to reproduce this issue

Hello world example

## Debug Logs

```
Serial port /dev/cu.usbserial-110
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x3f)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

dobairoland, 2022-01-18:
This is more like a driver issue. It looks like the same as https://github.com/espressif/esptool/issues/712. Please subscribe to that issue and share any information there which might help to track down the issue. This issue cannot be solved at the ESP-IDF side.

closed 2022-01-18 by dobairoland

oclyke, 2022-01-20:
Confirming - the above issue is what I am describing and I will watch there for updates. Thanks

oclyke, 2022-11-18:
For anyone who finds this: there are some temporary fixes listed in https://github.com/espressif/esptool/issues/712