espressif/esp-idf #8252: The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.

hdpklm, 2022-01-18:
Hello

I try to build bootloader but I get this error

Windows 10
vscode 1.63.2

please help me

```
> Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 .. <

CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of 
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first  
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error in CMakeLists.txt:
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error in CMakeLists.txt:
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.20)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "C:/Users/HDP/Desktop/temp/test/bootloader/esp-idf-master/esp-idf-master/components/bootloader_support/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/HDP/Desktop/temp/test/bootloader/esp-idf-master/esp-idf-master/components/bootloader_support/build/CMakeFiles/CMakeError.log".
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 .." terminated with exit code: 1.
```

hdpklm, 2022-01-18:
than you @igrr 

I want to put (BLE, ota, lcd driver) code in bootloader. because is to big, and upload code via OTA without redownload the repetitive code.

I think is a faster way to build and download via bluetooth.

igrr, 2022-01-18:
I don't think what you seek to do is possible, not with ESP-IDF at least. The Bluetooth stack in ESP-IDF relies on FreeRTOS and several other system components. By the time you add Bluetooth and OTA to the bootloader you will end up pretty much with the same binary size as your application.

What you can try doing instead is to create a separate `updater` app and arrange your partitions like this:

* `factory` partition — for the updater app with BLE and OTA functions
* `ota0` partition — your main app, no BLE functionality (slot 0)
* `ota1` partition — your main app, no BLE functionality (slot 1)

When your main app needs to trigger an upgrade, it can call `esp_ota_set_boot_partition` function to set the boot partition to `factory` and then restart the device.

The `updater` app will boot from `factory` partition and will perform the update over BLE, writing to the unused OTA slot.
When the updater app is done, it will call `esp_ota_set_boot_partition` again to set the boot partition to the newly updated OTA slot, and restart.

Finally, the updated app will boot.

This approach reduces OTA update time (since the size of the application in the OTA slot is reduced). However this is only the case when the main application (inside the OTA slot) doesn't use Bluetooth functionality. If you use Bluetooth both for the OTA (in the "updater" app) and in the main app, this approach doesn't result in any size reduction.

hdpklm, 2022-01-18:
is a good idea, but in the case I cant use bluetooth in my app, so my idea is to find address of each function in bootloader and the register use aprameters for call them by assembly api, I think is a good idea

hdpklm, 2022-01-18:
ok thank you for your time.
But you are sure the bootloader is build with the app. bBecause wen I clean the flash "esptool.exe erase_flash", the flash are empty and I can upload my new code without burn the bootloader. so I think the boot loader is in other partition and not modify every time I upload the code.

another question. is a way to write all code in assembly for manager the global variables, bootloader, with full api of the controller ?? so I write how partition are distrubuited and get permission to read code from other partition in run time ?? 

thnak you again.

hdpklm, 2022-01-18:
I think the main bootloader is in the hardware path "C:\Users\hassan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\tools\sdk\bin"

igrr, 2022-01-18:
In ESP-IDF project (since you have opened the issue here) the bootloader is indeed built together with the app. When you run `idf.py build`, that is equivalent to running `idf.py app`, `idf.py bootloader` and `idf.py partition_table`. All three — app, bootloader, partition table — are indeed separate binaries. If you run `idf.py flash`, all 3 are flashed to the device. If you run `idf.py app-flash`, [only the app is flashed](https://github.com/espressif/esp-idf#compiling--flashing-only-the-app). 

In Arduino (referring to your 2nd comment) this is different — the bootloader comes as a binary which is built separately from the rest of the Arduino code. Currently https://github.com/espressif/esp32-arduino-lib-builder repository is used to build the bootloader and other pre-compiled libraries for Arduino.

> is a way to write all code in assembly for manager the global variables, bootloader, with full api of the controller ?? so I write how partition are distrubuited and get permission to read code from other partition in run time ??

This is quite a big a question. I doubt I can write enough to explain all that you need to know to do this. If you are really committed to doing this, ESP-IDF source code is probably going to be the best source of information. Start by reading the [application startup flow](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/startup.html) description. Then use a JTAG debugger or QEMU emulator to step through the program startup, and learn what different parts of the startup process are doing. Read the technical reference manual of the ESP32 to understand the Flash MMU, since it is necessary for XIP (direct execution of code from Flash). Also take a look at some projects which implement dynamic loading of modules for the ESP32, e.g. https://github.com/niicoooo/esp32-elfloader.

All that said, if you are working on a product there could be other (more reliable) ways to get the application down to a reasonable size:
* follow general suggestions about optimizing the binary for size: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/size.html
* use NimBLE stack instead of Bluedroid if your application requires BLE only
* consider using ESP32-C3 chip instead of ESP32 if your application requires BLE only. With ESP32-C3, the code size and RAM usage of the bluetooth stack is less than on the ESP32.

hdpklm, 2022-01-18:
ok, Thank you a lot for the information.

Have a nice day 🌹🌹❤

closed 2022-01-18 by igrr