espressif/esp-idf #8254: DOCS: Change getting started default install instructions to all instead of esp32

alexchunlin, 2022-01-18:
**Is your feature request related to a problem? Please describe.**

Having the -copy paste- install prompt suggest to only install for esp32 target sets new users up for frustration when their build fails. 

It is especially confusing because the errors you get, when building for a project you haven't installed the right packages for, are not the easily google-able kind.  

**Describe the solution you'd like**
Don't instruct users to only install.sh for esp32
Change the docs on the Get Started page to:
`cd ~/esp/esp-idf
./install.sh`

**Describe alternatives you've considered**

- I considered learning to read, but its much easier to change the docs than teach me something so complicated

**Additional context**
I think the most confusing part of this is that the `./install.sh esp32` initially looks (to me) like its installing tools for all esp32 boards (ie including the variations of esp32 like the s2 , c3, s3)  It took me a long time to narrow down that I was only installing a portion of the tools I needed (granted, because I didn't read the note when I copied the instruction line over)

dobairoland, 2022-01-18:
Thanks for the suggestion. We will think about it. However, currently I'm not convinced that we should do this because others would (and previously used to be) complaining about toolchains being installed and re-installed for chip targets they never use.

The install instructions change based on the selected documentation. So if one reads the docs for ESP32-S2 then `install.sh esp32s2` is suggested to be run.

Also there is immediately after the install instructions the note stating what to do in order to install support for more or all chip targets.

dobairoland, 2022-01-18:
Most likely the best solution for this is to improve the error message and to print a suggestion to install toolchain for the missing target.

igrr, 2022-01-18:
I think we could improve this without changing the actual installation command, by adding a sentence right between the `Linux and macOS` title and the code block. Something along these lines:

> Run the following command to install the tools for {IDF_TARGET_NAME} target only. See the note below regarding installing the the tools for all targets at once.

and update the note below accordingly.

alexchunlin, 2022-01-18:
> The install instructions change based on the selected documentation. So if one reads the docs for ESP32-S2 then `install.sh esp32s2` is suggested to be run.
>
> Also there is immediately after the [...]

yeah, I mean you're right its totally right there in the note - I think a number of things added to maximize confusion in my situation.  Maybe sharing them can help you guys decide which direction is best:
- first, I have installed esp-idf a bunch of times so I was just trying to FLY through the set up process to get things moving
- my projects are spread across all boards, and I unluckily chose esp32 as my test to see if the install had worked correctly (leading me to falsely assume the install process was correctly done for my machine)
- I was also changing up some other workflows (moving from linux boot to WSL) which further added uncertainty to my troubleshooting
- finally, as I said earlier, the errors I was getting were leading me to "goose chase" online fixes hahaha 
                    `xtensa-esp32s2-elf-gcc
 is not a full path and was not found in the PATH.` 
 
I think your proposed error message fix would be a great solution
Hopefully it can save someone else from the trouble I went through to get things working this time around! :)

dobairoland, 2022-01-19:
Yes, thanks @alexchunlin. We appreciate your feedback and we will improve this.

closed 2022-03-01 by espressif-bot