espressif/esp-idf #8257: Problem with ESP32-S3 and SD

simeondmr, 2022-01-19:
Hello,

I'm using esp32-s3 and a MicroSD card adapter, with these connections(SDMMC_HOST_FLAG_1BIT):

- CLK: GPIO_NUM_14
- CMD: GPIO_NUM_15
- d0: GPIO_NUM_2

and this is my code:

`sdmmc_card_t *card;
sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    
host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;
host.flags = SDMMC_HOST_FLAG_1BIT;

sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
sdmmc_host_init_slot(SDMMC_HOST_SLOT_1, &slot_config);

gpio_pullup_en(GPIO_NUM_14);
gpio_pulldown_dis(GPIO_NUM_14);
gpio_pullup_en(GPIO_NUM_15);
gpio_pulldown_dis(GPIO_NUM_15);
gpio_pullup_en(GPIO_NUM_2);
gpio_pulldown_dis(GPIO_NUM_2);
gpio_pullup_en(GPIO_NUM_4);
gpio_pulldown_dis(GPIO_NUM_4);
gpio_pullup_en(GPIO_NUM_12);
gpio_pulldown_dis(GPIO_NUM_12);
gpio_pullup_en(GPIO_NUM_13);
gpio_pulldown_dis(GPIO_NUM_13);

esp_vfs_fat_sdmmc_mount_config_t mount_config = {
	.format_if_mount_failed = false,
	.max_files = 5,
};

esp_err_t ret = esp_vfs_fat_sdmmc_mount("/root", &host, &slot_config, &mount_config, &card);`

The problem is that I have these errors (and the esp_vfs_fat_sdmmc_mount return always error=0x107):

E (3193) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (3193) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
Failed to initialize the card. Make sure SD card lines have pull-up resistors in place.

I have done also the eFuse, but I have also this problem.

With esp32-wroom32 and esp32-pico mini I have no problem with the same code and wire.

please could you help me?

Thanks

Simeon

igrr, 2022-01-19:
Hi @simeondmr, for ESP32-S3, SD card pins need to be configured explicitly, since SDMMC peripheral now works with arbitrary GPIOs. The docs have just been updated, please see https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/sdmmc_host.html#configuring-gpios.

You probably need something along these lines:
```c++
sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
slot_config.width = 1;
slot_config.clk = GPIO_NUM_14;
slot_config.cmd = GPIO_NUM_15;
slot_config.d0 = GPIO_NUM_2;
sdmmc_host_init_slot(SDMMC_HOST_SLOT_1, &slot_config);
```

simeondmr, 2022-01-19:
Hello @igrr, thanks a lot for your reply!

Sorry I make a mistake...I'm using the eMMC(not SD).
Added your code, using the SD, 

` esp_vfs_fat_sdmmc_mount("/root", &host, &slot_config, &mount_config, &card);`

work correctly, but for example this code fail:

`FILE *test = fopen("/root/testfile1bit.raw", "w");
`

At the moment I must use the eMMC and also with your code, the problem persist as in my first question.

Do you have any idea?

Thanks a lot!

Simeon

igrr, 2022-01-19:
If it works with SD but fails with eMMC then it could be a timing issue. Does it work with eMMC at lower frequency? I.e. if you remove `host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;` line? Do you have a logic analyzer or an oscilloscope to capture the signals between ESP32-S3 and eMMC and verify timing and signal integrity?

simeondmr, 2022-01-19:
I removed the `host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;` but not work.
Also with `host.max_freq_khz = SDMMC_FREQ_DEFAULT;` I have the same problem.

At the moment I don't have an oscilloscope...but I don't think that could be a timing issue because with `host.max_freq_khz = SDMMC_FREQ_HIGHSPEED; ` on ESP32-WROOM32 and ESP32-PICO MINI work correctly.

igrr, 2022-01-19:
ESP32-S3 has different timing of the SDMMC peripheral since the signals now go through GPIO matrix, so the fact that ESP32 works in the same conditions is not really important.

Could you describe the connections between ESP32-S3 and eMMC chip? Are both on a PCB or you are connecting them by some cables?

igrr, 2022-01-19:
You are using Waveshare SD adapter for the SD card (which works, as I understand), but what about eMMC? Are you using some eMMC-to-SD adapter? (Asking just to understand how to replicate your issue)

simeondmr, 2022-01-19:
I'm using the eMMC: **BWCTASC11P32G**

simeondmr, 2022-01-20:
Hello, do you have some news about the issue?

vmastellone, 2022-01-24:
Hello @igrr,
I'm issuing the same problem using ESP32-S3-DevKitC anda eMMC to microSD adapter.
Is there any new suggestion about this issue?
Many thanks in advance

igrr, 2022-01-24:
Hi @simeondmr @vmastellone we're ordering an eMMC-to-SD adapter to reproduce the issue, I'll keep you updated about the progress.

simeondmr, 2022-01-28:
Hello, I have an update for @igrr and @vmastellone.
I make a PCB with the same connection, but the problem is still the same.
Thanks!

lorenzopeluso, 2022-01-28:
Hello guys. 

I made the same test using PCB.

I had the same problem.

igrr, 2022-01-28:
Hi @simeondmr, @lorenzopeluso, @vmastellone. Just letting you know that I have reproduced the issue and I will try to provide some patch to you soon.

lorenzopeluso, 2022-01-28:
Hi @igrr, thanks a lot.

BR,
Lorenzo.

fito-jaeuklee, 2022-03-08:
hello~!
i wonder this problem would be fixed?

May not, this issue relate to DDR mode?

`sdmmc_host_t host = SDMMC_HOST_DEFAULT();
host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;
host.flags &= ~SDMMC_HOST_FLAG_DDR;`

TGiles1998, 2022-03-09:
Hi, Is this issue still preventing esp32s3 micropython from successfully mounting an SD card? 

Thanks

Thomas

igrr, 2022-03-09:
Hi @TGiles1998, if you have an issue with SD card, it's like not _this_ issue — this one is related to eMMC.

For eMMC, this issue is still being investigated, I'm sorry for the delay. We see that one some development boards eMMC works okay with ESP32-S3, but on other development boards (including microSD-form-factor eMMC breakout boards) this doesn't work at all or doesn't work reliably. We haven't found the root cause yet, unfortunately.

fito-jaeuklee, 2022-03-10:
Hi ~ @igrr, Thanks for the immediate response
I tested ESP-WROOM-32(ESP32-D0WDQ6) board <-> microSD-reader (1~4bit) <-> SD-Card slot <-> eMMC

if i used 1bit data line, it very work well. but use 4 bit data line, not work.
So, i refered to esp-idf-whitepaper(docs) that indicate disable eMMC 
[`host.flags &= ~SDMMC_HOST_FLAG_DDR;`]
So I set it up like that, and it worked well for the 4-bit data line.

I have other questions. I ordered ESP32-S3 board that name is [ESP32-S3-DEVKITC-1-N8].
I wonder if it has been confirmed that this board works well.

Case of ESP32-dev-board<->eMMC  not work well :
Is it a problem with the development board circuitry? Or is it a problem with the microSD and emmc breakout boards?

thanks

fito-jaeuklee, 2022-03-10:
@igrr My HW setting.
[image]

fito-jaeuklee, 2022-03-10:
[image]

fito-jaeuklee, 2022-03-10:
@igrr 
For reference, upload my HW setting!
[image]

igrr, 2022-03-10:
@fito-jaeuklee Disabling SDMMC_HOST_FLAG_DDR in host flags will disable the DDR mode for eMMC, which indeed can work around some of the signal integrity issues!

> I have other questions. I ordered ESP32-S3 board that name is [ESP32-S3-DEVKITC-1-N8]. I wonder if it has been confirmed that this board works well.

The standard devkit doesn't have a built-in eMMC, so I doubt it can work well — the only way to use it is to connect eMMC using some jumper cables, which results in rather long wires going between two boards. This may cause signal integrity issues.

Right now Espressif has one official ESP32-S3 board with 4-line SD interface: [ESP32-S3-USB-OTG](https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32s3/esp32-s3-usb-otg/index.html). It is possible to use this USB-OTG board with an eMMC-to-microSD adapter like the one in your photo. However, as of now, the same issue can be reproduced when using ESP32-S3-USB-OTG with an eMMC breakout board.

igrr, 2022-03-12:
Made some progress getting highspeed mode to work in a parallel issue: https://github.com/espressif/esp-idf/issues/8521. I hope to post a patch for eMMC (including DDR mode) soon.

vmastellone, 2022-04-13:
Hello @igrr,
do you have any good news about this issue?

Could you please indicate which eMMC development board works fine with ESP32-S3?
> We see that one some development boards eMMC works okay with ESP32-S3, but on other development boards (including microSD-form-factor eMMC breakout boards) this doesn't work at all or doesn't work reliably.

vmastellone, 2022-04-22:
Hello,
please, is there any news about this issue and my previous question?

Many thanks in advance

igrr, 2022-05-04:
Hi @vmastellone, i'm sorry for the late response. Could you please try the following patch? With this patch and DDR mode disabled (`host.flags &= ~SDMMC_HOST_FLAG_DDR;`) the driver is able to work with several types of eMMC chips.

This patch can be applied to both `master` and `release/v4.4` branches.

[0001-sdmmc-I-O-phase-adjustments.patch.txt](https://github.com/espressif/esp-idf/files/8626393/0001-sdmmc-I-O-phase-adjustments.patch.txt)

igrr, 2022-05-04:
Have updated the patch, this one also work with eMMC in 40MHz DDR mode.

Feedback is welcome!

[0001-sdmmc-I-O-phase-adjustments.patch.txt](https://github.com/espressif/esp-idf/files/8626567/0001-sdmmc-I-O-phase-adjustments.patch.txt)

oursland, 2022-05-04:
@igrr I just came across this issue and can confirm that the patch permits recognition of my eMMC chip for:

* 20 MHz, SDR, 4-bit

It also appears to function at:

* 20 MHz, DDR, 4-bit 

The 40 MHz setting did not work, as expected.

In a board I am currently designing, I am targeting eMMC 40 Mhz, DDR, 8-bit or 80 MHz, 8-bit, DDR when available.

oursland, 2022-05-04:
@igrr  I have tested the latest patch and I now get DCRC errors in both DDR and SDR modes at 40 MHz.

```
I (347) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (357) gpio: GPIO[6]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (367) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (377) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (377) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (387) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (437) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
W (437) sdmmc_req: DCRC error
E (437) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (447) diskio_sdmmc: sdmmc_read_blocks failed (265)
W (447) vfs_fat_sdmmc: failed to mount card (1)
E (457) vfs_fat_sdmmc: mount_to_vfs failed (0xffffffff).
```

igrr, 2022-05-04:
Thanks for the feedback @oursland!

Regarding the modes which can be supported, esp32-s3 can (in principle) achieve:

- 20 MHz SDR, up to 8 lines
- 40 MHz SDR, up to 8 lines
- 40 MHz SDR, up to 4 lines

Higher bandwidth modes you mention (40 MHz DDR 8-line, 80 MHz) aren't supported by the hardware. Also in 20 MHz mode, DDR is not used, the driver will only enable DDR in 40 MHz mode.

To try to understand better your setup, could you please try tweaking the following line in the patch?
```
SDMMC.clock.phase_din = 6;
```
and setting values 0, 1, or 4 there?

If you have a logic analyzer, it would also help if you measure the eMMC output delay — that is, the time between the edge of CLK and edge of one of the data pins (e.g. D0) when it is driven by the card. In the board I have, this delay is between 7 and 10ns. The driver currently sets the sampling delay to 18.5ns. If the output delay of your eMMC is less than 6ns, it could mean that the sampling point is too late for the DDR mode.

oursland, 2022-05-04:
The eMMC delay is at the limit of my scope, but appears to be around 15 ns, both with and without external pull ups.

Unfortunately, none of the phase_din settings resolved the issue.

vmastellone, 2022-05-05:
Hi @igrr,
thank you very much for the support.

I've tried with success the following configurations:
- 20 MHz SDR, 1 line
- 40 MHz SDR, 1 line
- 20 MHz SDR, 4 lines
- 40 MHz SDR, 4 lines
- 40 MHz DDR, 4 lines

Best regards

igrr, 2022-05-05:
@vmastellone Excellent, thank you for the confirmation! 

@oursland Thanks for the photo, we'll try to set up something like that. To be fair, breadboards are usually not great at signal integrity... Does your ESP32 product also use such breadboard setup, or does it use a custom PCB with ESP32 module and eMMC?

oursland, 2022-05-05:
@igrr The production system is a custom PCB with the FriendlyElec eMMC module attached via the connector.  The breadboard setup has only been used for prototyping, which worked quite nicely.

igrr, 2022-05-09:
@oursland I haven't found a way to compensate for such a large clock-to-data delay (15ns) in the driver, yet.
Because ESP32-S3 uses GPIO matrix to make SD Host pins configurable, there is a larger input delay, compared to ESP32. I think the issue is that when eMMC also has long output delay, the two delays are combined and go outside of the range we can compensate for in hardware. I've forwarded the issue to the chip designers and will let you know once they analyze it.

oursland, 2022-05-09:
@igrr 40 MHz SDR, 4-lines is sufficient for my immediate need, but will want to go to DDR prior to product launch.

I will be evaluating eMMC parts from Kingston soon.  With any luck, they'll be compatible with the ESP32-S3.

In any case, I am eager to hear the designers' analysis.

thyclouds, 2022-06-07:
Hi @vmastellone!

Could you please share if you have also used eMMC module - [Samsung KLMAG1JETD-B041](https://semiconductor.samsung.com/estorage/emmc/emmc-5-1/klmag1jetd-b041/) on the [FriendlyElec 16GB eMMC Module](https://www.friendlyelec.com/index.php?route=product/product&path=70&product_id=240).

Also can you please share what read speed can you get from these?
20 MHz SDR, 1 line
40 MHz SDR, 1 line
20 MHz SDR, 4 lines
40 MHz SDR, 4 lines
40 MHz DDR, 4 lines

I assume that if you can interface with eMMC @ 40 MHz DDR, 4 lines it doesn't make sense to interface at the lower speeds. Rights?

Thank you in advance!

thyclouds, 2022-06-15:
Hi @vmastellone, @oursland , @igrr !

Could you please share if you have been succesful with Samsung KLMAG1JETD-B041 eMMC? Are you able to overcome DCRC error? Or you ended up with a Kingson or other chip?

I assume that if you can interface with eMMC @ 40 MHz DDR 4 lines it doesn't make sense to interface at the lower speeds. Right?

Thank you in advance!

oursland, 2022-07-16:
@thyclouds I have not been able to overcome the DCRC error on either the Samsung or Kingston modules at 40 MHz DDR.  They do operate at 40 MHz SDR, which is sufficient for my current needs.

igrr, 2023-04-19:
We have now added an eMMC example for ESP32-S3: https://github.com/espressif/esp-idf/tree/master/examples/storage/emmc.

closed 2023-04-19 by igrr