espressif/esp-idf #8258: Want a better Chunked transfer support

redlightASl, 2022-01-19:
I used `esp_http_client_perform()` and the example "esp_http_client" to GET http response, it didn't have any response when **chunked transfer encoding** is under using.

So we debug for few minutes,checked GitHub issue and API references but didn't have any solution.At last,we searched the http response and found the chunked transfer encoding caused the bug.We changed the function to `esp_http_client_get_chunk_length
` and killed the bug.

`esp_http_client_get_content_length()` returns -1 when using chunked transfer and **response cannot be recv by esp_http_client_config_t.user_data**.We are expecting a new implement to mix `esp_http_client_get_chunk_length()`
and `esp_http_client_get_content_length()` so that we don't need to deal with chunked transfer encoding XD.

KaeLL, 2022-01-19:
I unconsciously assumed there was a reason to not do that, but now that you brought it up, I'm wondering if that's indeed the case and what would the reason be.

fpgamaster, 2022-01-21:
Hi @redlightASl,

I use the following code for chunked transfer of OTA images:

`int http_request(char *http_response, int *http_response_length, int range_start, int range_end, int client_id)
{
        int err = -1;
        char range[64];
        int len_received = 0;

        if (range_end > range_start) {
                sprintf(range, "bytes=%d-%d", range_start, range_end);
                esp_http_client_set_header(ota_client[client_id], "Range", range);
        }

        err = esp_http_client_open(ota_client[client_id], 0);
        if (err != ESP_OK) {
                ESP_LOGE(LOG_TAG, "Failed to open HTTP connection: %s", esp_err_to_name(err));
        } else {
                int content_length = esp_http_client_fetch_headers(ota_client[client_id]);
                if (content_length < 0) {
                        ESP_LOGE(LOG_TAG, "HTTP client fetch headers failed");
                } else {
                        len_received = esp_http_client_read_response(ota_client[client_id], http_response, *http_response_length);
                        if (len_received >= 0) {
                                ESP_LOGI(LOG_TAG, "HTTP Status = %d, content_length = %lld",
                                    esp_http_client_get_status_code(ota_client[client_id]),
                                    esp_http_client_get_content_length(ota_client[client_id]));
                        } else {
                                ESP_LOGE(LOG_TAG, "Failed to read response");
                        }
                }
        }

        esp_http_client_close(ota_client[client_id]);

        *http_response_length = len_received;

        return (err);
}
`

ESP-YJM, 2022-01-24:
@redlightASl When you call `esp_http_client_perform`, the chunked data will in `HTTP_EVENT_ON_DATA` event. You can add print for value `evt->data` and `evt->data_len` in event `HTTP_EVENT_ON_DATA`.

KaeLL, 2022-01-24:
@ESP-YJM Can't `esp_http_client_get_content_length` implicitly call `esp_http_client_is_chunked_response` and if true, return the current payload size instead of -1?

ESP-YJM, 2022-01-25:
@KaeLL I think for the API `esp_http_client_get_content_length `, it is designed to get length which http header has `Content-Length` field. If you want to get chunk type data length, you may need call `esp_http_client_get_chunk_length `. Before call it, you may need call `esp_http_client_is_chunked_response ` to judge whether it is chunked.

AxelLin, 2022-01-25:
> @ESP-YJM Can't `esp_http_client_get_content_length` implicitly call `esp_http_client_is_chunked_response` and if true, return the current payload size instead of -1?

The length returned by these 2 APIs have different meaning.
esp_http_client_get_content_length returns the length of **whole body**.
esp_http_client_get_chunk_length only return length of **current chunk**.
The API usage is different, had better don't mix them.

KaeLL, 2022-01-25:
I just thought there were other reasons other than technicality/semantics out of curiosity. Thanks for the answers.

mahavirj, 2022-02-11:
As already commented here, both APIs serve different purpose. Please find documentation pointers below:

https://github.com/espressif/esp-idf/blob/ca84d2d6c994ddf3733ebcfa83db131b6c4c5d66/components/esp_http_client/include/esp_http_client.h#L465-L475

https://github.com/espressif/esp-idf/blob/ca84d2d6c994ddf3733ebcfa83db131b6c4c5d66/components/esp_http_client/include/esp_http_client.h#L592-L603

We plan to keep them separate. I will go ahead and close this issue. If there is anything else on this topic, please feel free to re-open or let us know about it.

closed 2022-02-11 by mahavirj