espressif/esp-idf #8261: Unable to run BLE_SPP_Server example

MaazSk, 2022-01-19:
## Environment

- Development Kit:      [ESP32-DevKit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32|]
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-939-g952b4d848f
   
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
  
- Operating System:     [Linux]

- Using an IDE?: [VS Code]
- Power Supply:         [USB]

## Problem Description

I am trying to run the BLE_SPP_Server example but I am getting the following error. Simply compiled and burned the code.
```

I (963) uart: queue free spaces: 10
I (963) GATTS_SPP_DEMO: EVT 22, gatts if 3

I (963) GATTS_SPP_DEMO: event = 16

I (973) GATTS_SPP_DEMO: The number handle =b

E (973) GATTS_SPP_DEMO: GAP_EVT, event 4

I (983) GATTS_SPP_DEMO: EVT E (1003) GATTS_SPP_DEMO: GAP_EVT, event 6
```

Thank you in advance

Weijian-Espressif, 2022-03-30:
@MaazSk  I don't see any mistakes. Is there a more detailed log?

Weijian-Espressif, 2022-03-30:
I (983) GATTS_SPP_DEMO: EVT E (1003) GATTS_SPP_DEMO: GAP_EVT, event 6

This log means that ESP_GAP_BLE_ADV_START_COMPLETE_EVT is triggred, not an error

MaazSk, 2022-03-31:
@Weijian-Espressif, I am not working on it currently. I will update the logs once I am back.

closed 2024-12-17 by Alvin1Zhang