espressif/esp-idf #8264: 靠近连接开启AP模式怎么进行http通信？

CYCHEN97, 2022-01-20:
使用靠近连接例程blufi，在手机APP中设置开启AP模式，电脑连接esp32s3的wifi，APP中点击状态查看状态，一直显示SoftAP current connection：0，然后没法进行域名访问http通信；使用STA模式的时候就可以进行域名访问http通信
[image]

CYCHEN97, 2022-01-20:
@0xjakob
@esp-lis
@ESP-YJM
@negativekelvin
@alisitsyn

mhdong, 2022-01-24:
hi @CYCHEN97  在wifi init之前添加esp_netif_create_default_wifi_ap();
这样softap可为连接的sta分配ip地址，即可进行http通信。对于SoftAP current connection一直显示为0的原因在于esp_blufi_send_wifi_conn_report（）函数没有将softap连接个数更新。

CYCHEN97, 2022-02-11:
@mhdong 
有没有PC端基于Qt的蓝牙配网上位机？

mhdong, 2022-03-03:
hi @CYCHEN97  暂时没有支持基于Qt的蓝牙配网上位机。

closed 2022-03-05 by CYCHEN97