espressif/esp-idf #8266: WiFi affects ADC1 raw data

dizcza, 2022-01-20:
## Environment

- Development Kit:  TTGO-T8-ESP32
- Kit version: v1.7
- Module or chip used:  ESP32
- IDF version: v5.0-dev tag (master branch has a websocket client example broken - I haven't bothered reporting this; that's why I've stuck to v5.0-dev)
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: Ubuntu 20.04
- Using an IDE?: Only to edit the code. The program was flashed from a terminal with the `idf.py flash` command.
- Power Supply: USB

## Problem Description

When WiFi is active, in particular, an ESP32 websocket client is sending data to a server, `adc1_get_raw()` returns wrong unrealistic values. ADC values are read from an analog SDP806 differential pressure sensor. Without the Wifi running, the ADC1 values are correct - near zero. When Wifi is active, I see outliers with huge (>1k) ADC1 raw values.

The analog sensor is connected to pin 35, ADC1 channel 7.

I'm aware of ADC limitations but they concert ADC2 only - I've read this part of the documentation carefully and intentionally chosen ADC1 to avoid WiFi-related conflicts. If there are limitations to ADC1, please add them to the docs.

### Related issues

I first thought it's related to https://github.com/espressif/esp-idf/issues/7781, but Bluetooth, to my surprise, does NOT affect ADC1 readings. Only Wifi does.

This issue supplements https://github.com/espressif/esp-idf/issues/6682 though in my case the wifi does not get disconnected when I call `adc1_get_raw()` periodically.

### Expected Behavior

`adc1_get_raw()` returns small values around zero when the WiFi is running.

### Actual Behavior

`adc1_get_raw()` returns unrealistically large values when the WiFi is running.

### Code to reproduce this issue

`websocket_app_start()` is taken from the websocket example https://github.com/espressif/esp-idf/tree/master/examples/protocols/websocket.

```c
void task_send() {
     char data[512];
     
     while (1) {
            esp_websocket_client_send_bin(client, data, sizeof(data), portMAX_DELAY);
            vTaskDelay(pdMS_TO_TICKS(100));
     }
}

void app_main() {
    ...
    websocket_app_start()

    ESP_ERROR_CHECK(adc1_config_channel_atten(board->sdp.adc.channel, ADC_ATTEN_DB_11));
    adc1_config_width(ADC_WIDTH_BIT_12);
    esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, DEFAULT_VREF, &m_adc_chars);

    xTaskCreatePinnedToCore(task_send, "task_send", 4096, NULL, 0, NULL, PRO_CPU_NUM);

    ...

    // A periodic timer is set up to poll ADC1 at 1000 Hz freq
    int adc_raw = adc1_get_raw(board->sdp.adc.channel);
    int aout_v = (int) esp_adc_cal_raw_to_voltage((uint32_t) adc_raw, &m_adc_chars);
    // aout_v values are printed
```

Full code:

1. ADC1 readings: https://github.com/dizcza/esp32-sdpsensor/blob/dc6edd0b88562a05a875bffb4dbd0ddb8ff043cf/main/sdpsensor_adc.c#L66
2. A websocket client sending data in another thread: https://github.com/dizcza/esp32-sdpsensor/blob/dc6edd0b88562a05a875bffb4dbd0ddb8ff043cf/main/record_sdp.c#L243

## Other items if possible

- [sdkconfig](https://www.dropbox.com/s/x0be4a81qiichby/sdkconfig?dl=1)
- [elf file](https://www.dropbox.com/s/n2pfsnldy1ri40f/esp32-sdpsensor.elf?dl=1)

jack0c, 2022-05-20:
@dizcza Can you try commit after f3adbf995314fd95fb39472c282da502546cc9bc

dizcza, 2022-05-20:
I don't have the module with me at the moment.
You are free to close the issue and if I find it's unresolved I could reopen it.
Thanks.

Alvin1Zhang, 2022-05-23:
Thanks for reporting and updates, feel free to reopen.

closed 2022-05-23 by Alvin1Zhang

clemrdh, 2024-12-05:
I am having the exact same issue ATM

piemanau, 2025-01-14:
@dizcza Have you tested it after this change? As discussed on this issue https://github.com/esp-rs/esp-idf-hal/issues/468, I am still struggling.