espressif/esp-idf #8267: MDNS: buffer overflow when setting a TXT record with empty value

alyf80, 2022-01-20:
## Environment

- IDF version (run ``git describe --tags`` to find it):
    v4.4-rc1 (current release/v4.4 head), but the problem is also in the current master

## Problem Description

`components/mdns/mdns.c` line 705, in `_mdns_append_txt_record()`:
```
        uint8_t txt_data_len = strlen(txt->key) + txt->value_len + 1;
        tmp = (char *)malloc(txt_data_len);
        if (tmp) {
            int offset = sprintf(tmp, "%s=", txt->key);
```
This will overflow `tmp` if the TXT record has an empty value (i.e. `txt->value_len == 0`), which is allowed by the DNS-SD specification.

david-cermak, 2022-01-21:
Hi @alyf80 

Thank you for reporting this problem. Indeed, the mDNS library doesn't support zero sized values in TXT records.
This is not a critical bug, as users currently cannot register null-value record; this would fail:
```cpp
    ESP_ERROR_CHECK( mdns_service_txt_item_set_with_explicit_value_len(..., "my_key", NULL, 0) );
```
That's why this won't be fixed in the final `v4.4` release (to be included in `v4.4.1`, though).

You can find the first WIP patch below:
[mdns-Support-for-null-value-TXT-records.patch.txt](https://github.com/espressif/esp-idf/files/7912848/mdns-Support-for-null-value-TXT-records.patch.txt)

alyf80, 2022-01-21:
This, however, does _not_ fail and triggers the problem:

```c++
mdns_txt_item_t service_data[1] = {
  {"A", ""}
};

ESP_ERROR_CHECK (mdns_service_txt_set("_http", "_tcp", service_data, 1));
```

david-cermak, 2022-01-21:
Ah, thanks, haven't thought of that use-case, yes, that makes `txt->value_len == 0`, too. The TXT record would look like `key=` (compared to `key` in the null-value record case).

The above patch addresses this *empty string* case, too.

closed 2022-01-24 by espressif-bot