espressif/esp-idf #8268: is there a demo for rgb panel(which 8*3 = 24pins RGB data)

Kingwulin, 2022-01-21:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

suda-morris, 2022-01-21:
At the moment, there's only one lvgl demo in the [test folder](https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/test/test_rgb_panel.c). We will push a new example recently. But is for RGB565 LCD. 
ESP32S3 doesn't support a parallel 24 bit-width RGB LCD.

Kingwulin, 2022-01-21:
And is it possible to modify the test demo [https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/test/test_rgb_panel.c](url) to support a parallel 24 bit-width RGB LCD

Kingwulin, 2022-01-21:
> At the moment, there's only one lvgl demo in the [test folder](https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/test/test_rgb_panel.c). We will push a new example recently. But is for RGB565 LCD. ESP32S3 doesn't support a parallel 24 bit-width RGB LCD.

hello!!!  I build the code ok ,but when i run it ,it say like this:
_
[17:52:38.753]收←◆餎SP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x16dc
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f90
entry 0x403b6248
[0;32mI (24) boot: ESP-IDF v5.0-dev-1281-g09f5c1d32e 2nd stage bootloader[0m
[0;32mI (25) boot: compile time 17:40:38[0m
[0;32mI (25) boot: chip revision: 0[0m
[0;32mI (28) boot.esp32s3: Boot SPI Speed : 80MHz[0m
[0;32mI (33) boot.esp32s3: SPI Mode       : DIO[0m
[0;32mI (37) boot.esp32s3: SPI Flash Size : 2MB[0m
[0;32mI (42) boot: Enabling RNG early entropy source...[0m
[0;32mI (48) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (73) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (81) boot: End of partition table[0m
[0;32mI (85) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0d738h ( 55096) map[0m
[0;32mI (103) esp_image: segment 1: paddr=0001d760 vaddr=3fc91e10 size=028b8h ( 10424) load[0m
[0;32mI (106) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=3fafch (260860) map[0m

[17:52:38.911]收←◆[0;32mI (157) esp_image: segment 3: paddr=0005fb24 vaddr=3fc946c8 size=003ach (   940) load[0m
[0;32mI (158) esp_image: segment 4: paddr=0005fed8 vaddr=40374000 size=0de0ch ( 56844) load[0m
[0;32mI (175) esp_image: segment 5: paddr=0006dcec vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (181) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (181) boot: Disabling RNG early entropy source...[0m
[0;32mI (196) opi psram: vendor id    : 0x00 (UNKNOWN)[0m
[0;32mI (196) opi psram: dev id       : 0x03 (generation 4)[0m
[0;32mI (196) opi psram: density      : 0x07 (256 Mbit)[0m
[0;32mI (201) opi psram: good-die     : 0x01 (Pass)[0m
[0;32mI (206) opi psram: Latency      : 0x01 (Fixed)[0m
[0;32mI (211) opi psram: VCC          : 0x01 (3V)[0m
[0;32mI (216) opi psram: SRF          : 0x00 (Slow Refresh)[0m
[0;32mI (222) opi psram: BurstType    :
[17:52:39.007]收←◆ 0x01 ( Wrap)[0m
[0;32mI (227) opi psram: BurstLen     : 0x03 (1024 Byte)[0m
[0;32mI (232) opi psram: Readlatency  : 0x07 (20 cycles@Fixed)[0m
[0;32mI (238) opi psram: DriveStrength: 0x03 (1/8)[0m
[0;33mW (243) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.[0m
[0;33mW (254) MSPI Timing: tuning fail, best point is fallen back to index 1[0m
[0;32mI (260) spiram: Found 32MB SPI RAM device[0m
[0;32mI (265) spiram: Speed: 80MHz[0m
[0;32mI (269) spiram: Initialized, cache is in normal (1-core) mode.[0m
[0;31mE (275) spiram: Virtual address not enough for PSRAM![0m

abort() was called at PC 0x40376475 on core 0

Backtrac
[17:52:39.071]收←◆e:0x40375be2:0x3fceb2400x40379935:0x3fceb260 0x4037fa72:0x3fceb280 0x40376475:0x3fceb2f0 0x403753ca:0x3fceb310 0x403bb0c4:0x3fceb340 0x403bb52d:0x3fceb380 0x403b62b1:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED

ELF file SHA256: 2971b2c87cbf371e

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375888
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x16dc
load:0x403b6000,len:0xb7c
[... 26 lines cut ...]
[0;32mI (200) opi psram: dev id       : 0x03 (generation 4)[0m
[0;32mI (200) opi psram: density      : 0x07 (256 Mbit)[0m
[0;32mI (205) opi psram: good-die     : 0x01 (Pass)[0m
[0;32mI (210) opi psram: Latency      : 0x01 (Fixed)[0m
[0;32mI (215) opi psram: VCC          : 0x01 (3V)[0m
[0;32mI (220) opi psram: SRF          : 0x00 (Slow Refresh)[0m
[0;32mI (226) opi psram: BurstType    :
[17:52:39.335]收←◆ 0x01 ( Wrap)[0m
[0;32mI (231) opi psram: BurstLen     : 0x03 (1024 Byte)[0m
[0;32mI (237) opi psram: Readlatency  : 0x07 (20 cycles@Fixed)[0m
[0;32mI (243) opi psram: DriveStrength: 0x03 (1/8)[0m
[0;33mW (247) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.[0m
[0;33mW (258) MSPI Timing: tuning fail, best point is fallen back to index 1[0m
[0;32mI (265) spiram: Found 32MB SPI RAM device[0m
[0;32mI (269) spiram: Speed: 80MHz[0m
[0;32mI (273) spiram: Initialized, cache is in normal (1-core) mode.[0m
[0;31mE (279) spiram: Virtual address not enough for PSRAM![0m

abort() was called at PC 0x40376475 on core 0

Backtrac
[17:52:39.397]收←◆e:0x40375be2:0x3fceb2400x40379935:0x3fceb260 0x4037fa72:0x3fceb280 0x40376475:0x3fceb2f0 0x403753ca:0x3fceb310 0x403bb0c4:0x3fceb340 0x403bb52d:0x3fceb380 0x403b62b1:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED

_
how can i

Kingwulin, 2022-01-24:
> At the moment, there's only one lvgl demo in the [test folder](https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/test/test_rgb_panel.c). We will push a new example recently. But is for RGB565 LCD. ESP32S3 doesn't support a parallel 24 bit-width RGB LCD.

hello ，would you please give me a replay on this issues  [https://github.com/espressif/esp-idf/issues/8273](url)

suda-morris, 2022-03-07:
The RGB LCD example has been exist on the mater branch for some time. Due to the lack of enough SRAM for the frame buffer, we recommend users to enable the PSRAM (especially using Octal PSRAM) for RGB LCD application.

closed 2022-03-07 by suda-morris