espressif/esp-idf #8269: WIFI_ALL_CHANNEL_SCAN, this parameter doesn't seem to work

shun-dgut, 2022-01-21:
## Environment

- Development Kit:      [ESP32-C3-DevKitM-1]
- IDF version (run ``git describe --tags`` to find it):
   v4.3.2-198-gb9a96186fd 
- Build System:         [idf.py]
- Compiler version
   riscv32-esp-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Power Supply:         [USB]

## Problem Description

Using full channel scanning doesn't work, I built two wifi with the same SSID and password, but ESP32C3 doesn't always choose the connection with the best signal.

## Code to reproduce this issue
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = DEFAULT_SSID,
            .password = DEFAULT_PWD,
            .scan_method = WIFI_ALL_CHANNEL_SCAN, // WIFI_FAST_SCAN, // 
            .sort_method = WIFI_CONNECT_AP_BY_SIGNAL,
            .threshold.rssi = DEFAULT_RSSI,
            .threshold.authmode = DEFAULT_AUTHMODE,
        },
     };
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config));

Alvin1Zhang, 2022-01-21:
Thanks for reporting.

nishanth-radja, 2022-02-08:
@shun-dgut can we have the console logs of the DUT when it connects to both the AP's, We will be able to see the RSSI when it connects to the AP and compare the RSSI difference it sees between the AP's.

shun-dgut, 2022-02-08:
Thank you for your reply. Here is the log, rebooting the board via the rst pin appears to be random every time the bssid it connects to.
[测试ssid.log](https://github.com/espressif/esp-idf/files/8021257/ssid.log)

nishanth-radja, 2022-02-08:
@shun-dgut  Thanks for the logs.From the logs,we see that there is a 20db difference between the RSSI of the AP's.Still the Client connects randomly.On further analysis, It is seen that the AP1 with lower RSSI is at channel 1 and AP2 with higher RSSI is at channel 7. 
This issue is seen as the AP's are on different channels and we are also observing the same behavior in lab.we will get back to you on this issue.
Meanwhile can you check if this issue is seen when both the AP's are on same channel?

xueyunfei998, 2022-02-08:
hi @shun-dgut 

[2022-01-25 19:05:48.138]# RECV ASCII>
I (2550) wifi:new:<7,0>, old:<1,0>, ap:<255,255>, sta:<7,0>, prof:1
I (2550) wifi:state: init -> auth (b0)
I (2550) wifi:state: auth -> assoc (0)
I (2560) wifi:state: assoc -> init (2c0)
I (2560) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (2560) wifi:new:<1,0>, old:<7,0>, ap:<255,255>, sta:<1,0>, prof:1
I (2570) wifi:state: init -> auth (b0)
I (2580) wifi:state: auth -> assoc (0)
I (2580) wifi:state: assoc -> run (10)
I (2610) wifi:connected with tototo2.4g, aid = 1, channel 1, BW

hi @shun-dgut 

In the log you provided, channel 7, the ap with the best rssi, will be scanned every time

I (2560) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1

However, during the connection process, there was a connection error in channel 7, and then channel 1 was rescanned to connect to the ap of channel 1.

I (2560) wifi:state: assoc -> init (2c0)
I (2560) wifi:new:<1,0>, old:<7,0>, ap:<255,255>, sta:<1,0>, prof:1

The reason may be that when connecting to channel 7, the router kicked out the sta and failed to connect successfully.

AxelLin, 2022-02-08:
> However, during the connection process, there was a connection error in channel 7, and then channel 1 was rescanned to connect to the ap of channel 1.
>
> I (2560) wifi:state: assoc -> init (2c0) I [...]

But this connect error should not happen.
The log shows it sometimes can join channel 7.
So how to confirm if this connect error is esp32's issue or router's issue?
Does verbose build help to clarify the issue?

xueyunfei998, 2022-02-08:
hi @shun-dgut 

Please provide the model of the router, and I will try to reproduce this problem locally.

shun-dgut, 2022-02-08:
one is RT-AX56U and other one is OpenWrt 19.07.4 r11208-ce6496d796

AxelLin, 2022-02-08:
> one is RT-AX56U and other one is OpenWrt 19.07.4 r11208-ce6496d796

@shun-dgut  Which one is the channel 7 AP?

AxelLin, 2022-02-08:
@shun-dgut 

BTW, just for comparison.
If you move esp32 close to the channel 1 AP, will it hit connect error with channel 1 AP?
i.e. will it connect to channel 7 AP sometimes?

shun-dgut, 2022-02-08:
> channel 7 AP is RT-AX56U

shun-dgut, 2022-02-08:
I use esp32c3, I also experimented on 8266 and found that 8266 basically chooses a router with a stronger signal value.

nishanth-radja, 2022-02-08:
@shun-dgut can we get the sniffer capture on channel 7 and also  can you enable the wpa_supplicant logs on DUT.

shun-dgut, 2022-02-09:
> @shun-dgut can we get the sniffer capture on channel 7 and also can you enable the wpa_supplicant logs on DUT.

Can you tell me how to set it in the firmware?

nishanth-radja, 2022-02-10:
@shun-dgut To enable the logs, go to idf.py menuconfig->component config->supplicant->print debug message from wpa_supplicant.
once the log is enabled in the SDK config you will see that "CONFIG_WPA_DEBUG_PRINT=y" is present.

shun-dgut, 2022-02-10:
[wpa_supplicant.log](https://github.com/espressif/esp-idf/files/8038941/wpa_supplicant.log)

> @shun-dgut To enable the logs, go to idf.py menuconfig->component config->supplicant->print debug message from wpa_supplicant. once the log is enabled in the SDK config you will see that "CONFIG_WPA_DEBUG_PRINT=y" is present.

nishanth-radja, 2022-02-10:
@shun-dgut Thank you,Can we have the corresponding sniffer log on channel 7.To know the reason why the AP is rejecting the STA assoc every alternate attempt.

shun-dgut, 2022-02-10:
> @shun-dgut Thank you,Can we have the corresponding sniffer log on channel 7.To know the reason why the AP is rejecting the STA assoc every alternate attempt.

Do I need to use the packet capture card to capture network packets?

nishanth-radja, 2022-02-10:
@shun-dgut if you have a linux laptop,You can use it to sniff the channel.

ifconfig <wifi-interface> down
iwconfig <wifi-interface> mode monitor
ifconfig <wifi-interface> up
iwconfig <wifi-interface> channel 7
wireshark

shun-dgut, 2022-02-10:
[image]
[image]
[ssid.zip](https://github.com/espressif/esp-idf/files/8040467/ssid.zip)
[Uploading ssid log.txt…]()

> @shun-dgut if you have a linux laptop,You can use it to sniff the channel.
> 
> ifconfig down iwconfig mode monitor ifconfig up iwconfig channel 7 wireshark

nishanth-radja, 2022-02-10:
@shun-dgut Looks like this issue is similar to the existing issue [8192](https://github.com/espressif/esp-idf/issues/8192)
DUT initially sends the Deauth to the AP before association.This is done to clear the station entry in the assoc table of the AP.
But few AP's process this Deauth late,They send the auth response and then send a deauth again as a confirmation.
[image]

pls apply the following patch and take the sniffer capture
[patch_for_reconnect_issue.zip](https://github.com/espressif/esp-idf/files/8040604/patch_for_reconnect_issue.zip).

These files needs to be replaced in the following location.
$IDF_HOME/components/esp32/lib
clean the already existing build and rebuild again.
idf.py fullclean
idf.py build

pls apply the following patch and take the sniffer capture on channel 7.

shun-dgut, 2022-02-11:
> @shun-dgut Looks like this issue is similar to the existing issue [8192](https://github.com/espressif/esp-idf/issues/8192) DUT initially sends the Deauth to the AP before association.This is done to clear the station entry in the assoc table of the AP. But few AP's process this Deauth late,They send the auth response and then send a deauth again as a confirmation. [image]
> 
> Since you are working c3, I will get back to you with a patch for the same.

Thanks . When can you give me these patch tests

GarMingLi, 2022-02-11:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在工作中，无法亲自回复您的邮件。我将在工作结束后，尽快给您回复。

nishanth-radja, 2022-02-11:
Hi @shun-dgut on the idf 4.3, pls use the following patch
[patch_for_reconnect_issue_4.3.zip](https://github.com/espressif/esp-idf/files/8048733/patch_for_reconnect_issue_4.3.zip)
replace the files in the location "esp-idf/components/esp_wifi/lib/esp32c3" with the attached files .

clean the already existing build and rebuild again.
idf.py fullclean
idf.py build

AxelLin, 2022-02-13:
> But few AP's process this Deauth late,They send the auth response and then send a deauth again as a confirmation. 

Then how do you address this issue in your patch?

shun-dgut, 2022-02-14:
> Hi @shun-dgut on the idf 4.3, pls use the following patch [patch_for_reconnect_issue_4.3.zip](https://github.com/espressif/esp-idf/files/8048733/patch_for_reconnect_issue_4.3.zip) replace the files [...]

Tried it and it doesn't seem to work

log 

```
I (2604) wifi:connected with tototo2.4g, aid = 1, channel 6, BW
[2022-02-14 11:06:06.097]# RECV ASCII>
20, bssid = 74:44:01:02:a3:e7
I (2604) wifi:security: WPA2-PSK, phy: bgn, rssi: -64
I (2614) wifi:pm start, type: 1
I (2614) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (2644) wifi:BcnInt:102400, DTIM:2
[2022-02-14 11:06:06.809]# RECV ASCII>
[0;32mI (3344) scan: got ip:192.168.100.243[0m
[0;32mI (3344) esp_netif_handlers: sta ip: 192.168.100.243, mask: 255.255.255.0, gw: 192.168.100.1[0m
[2022-02-14 11:06:11.529]# RECV ASCII>
W (8074) wifi:<ba-add>idx:0 (ifx:0, 74:44:01:02:a3:e7), tid:0, ssn:1, winSize:64
```

nishanth-radja, 2022-02-14:
Hi @shun-dgut , can we have the capture on channel 7 with this patch?

nishanth-radja, 2022-02-14:
> > But few AP's process this Deauth late,They send the auth response and then send a deauth again as a confirmation.
> 
> Then how do you address this issue in your patch?

There is a delay between the deauth and Auth so that the deauth of the AP is processed before sending the Auth.

shun-dgut, 2022-02-16:
>

AxelLin, 2022-02-16:
@shun-dgut 
Does that mean patch_for_reconnect_issue_4.3.zip works or you have found something else to make it work?
What's your finding of this issue (since it's closed now)?

shun-dgut, 2022-02-16:
I found that if I use esp_restart() I am connecting to a wifi with a strong signal value, but using hardware RST does not result in this.

nishanth-radja, 2022-02-16:
@shun-dgut we are working on a better solution for this problem,Will get back to you ASAP

shun-dgut, 2022-02-23:
[2022-2-3.zip](https://github.com/espressif/esp-idf/files/8123177/2022-2-3.zip)
Thanks for the reply, here's the packet I grabbed.
It looks correct now and I was wondering if this fix will affect other feature

nishanth-radja, 2022-02-24:
@shun-dgut The fix was to introduce a 20ms delay after deauth,so that AP has time to process it. It will not affect any other feature.

marko-savic, 2022-07-18:
I'm able to reproduce this issue with:

- Kit: ESP32-WROOM-32UE
- Build System: [idf.py] version v4.4.1
- Operating System: [Windows]
- Power Supply: external

Currently, I don't have logs, as fw is deployed remotely with reduced logging. However, I can observe that some ESPs connects to APs which are further away, and I'm pretty sure they have lower RSSI. 

Before I make env to reproduce this, I'm wondering if this issue is known on 4.4.1?

AxelLin, 2022-07-18:
@marko-savic 

>  However, I can observe that some ESPs connects to APs which are further away, and I'm pretty sure they have lower RSSI.

Just want to make sure if you have set ".sort_method = WIFI_CONNECT_AP_BY_SIGNAL," in your wifi_config?

nishanth-radja, 2022-07-19:
@marko-savic The patch which was provided to the Customer,Is not yet merged into 4.4,Issue will be present in 4.4.1

marko-savic, 2022-07-20:
@nishanth-radja 
We have deployed 82 devices in one production hall with 3 ap in the mesh setup. 
Now it is a pretty chaotic situation. Is it possible to provide me with a patch?
I'm willing to help you with the testing if needed as it is a large-scale system working in a production environment.

nishanth-radja, 2022-07-20:
@marko-savic , Yes,I am following up with the Dev team on this.Will get back to you ASAP.

marko-savic, 2022-07-20:
@nishanth-radja the encryption is WPA2_PSK. As for PMF, I think it is from 4.4.1 explicitly set to true, I did not override the default behavior.
Additionally, I don't use NVS, but that should not matter.

The production hall is in a remote location I have no access to their network at the moment. However, I can organize sniffing, probably next week.

nishanth-radja, 2022-07-21:
@marko-savic Yes,PMF is enabled by default on the STA, I was wondering if PMF is enabled on the AP as well?Because only if PMF is enabled on both AP and STA,the PMF will be used.
Can we also get the console logs of the STA?

marko-savic, 2022-07-21:
@nishanth-radja PMF is disabled on AP. I'm not able to send you the logs, but I will make small env with two APs in the office an get you the logs.

nishanth-radja, 2022-07-22:
> @nishanth-radja I've updated to 4.4.1. In the release note of 4.4.1, there is nothing about this issue. Can you help me with the patch, please?

@marko-savic The issue is not mentioned in the release notes because the issue was on AP side.And the patch was a workaround for that particular AP.We will have to see from the logs(sniffer and console) If you are facing the same issue.

HiFiPhile, 2022-07-27:
I've the same issue.

## Environment

- Development Kit:      [esp32-c3-devkitm-1]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [esp32-c3-wroom-02]
- IDF version (run ``git describe --tags`` to find it):
    v5.0-dev-4379-g36f49f361c
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    gcc version 11.2.0 (crosstool-NG esp-2022r1-RC1)
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [VSCode]
- Power Supply:         [USB]

We have 2 APs:
- 5c:a6:e6:8a:52:86 at CH6 about -45dbm
- 5c:a6:e6:8a:53:a4 at CH1 about -80dbm.

At startup MCU connect randomly to one of them, despite `WIFI_ALL_CHANNEL_SCAN` is used.
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6110,len:0x1374
load:0x403ce000,len:0x934
load:0x403cf600,len:0x2c9c
entry 0x403ce000
I (30) boot: ESP-IDF v5.0-dev-4379-g36f49f361c 2nd stage bootloader
I (30) boot: compile time 10:20:49
I (30) boot: chip revision: V003
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 coredump         Unknown data     01 03 00010000 00010000
I (87) boot:  3 factory          factory app      00 00 00020000 00140000
I (94) boot:  4 ota_0            OTA app          00 10 00160000 00140000
I (102) boot:  5 ota_1            OTA app          00 11 002a0000 00140000
I (109) boot:  6 otadata          OTA data         01 00 003e0000 00002000
I (117) boot:  7 nvs_key          NVS keys         01 04 003e2000 00001000
I (124) boot: End of partition table
I (129) boot: Defaulting to factory image
I (133) esp_image: segment 0: paddr=00020020 vaddr=3c070020 size=15220h ( 86560) map
I (156) esp_image: segment 1: paddr=00035248 vaddr=3fc8f200 size=0279ch ( 10140) load
I (158) esp_image: segment 2: paddr=000379ec vaddr=40380000 size=0862ch ( 34348) load
I (169) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=62d3ch (404796) map
I (238) esp_image: segment 4: paddr=000a2d64 vaddr=4038862c size=06ad8h ( 27352) load
I (243) esp_image: segment 5: paddr=000a9844 vaddr=50000010 size=00004h (     4) load
I (244) esp_image: segment 6: paddr=000a9850 vaddr=50000018 size=00010h (    16) load
I (256) boot: Loaded app from partition at offset 0x20000
I (258) boot: Disabling RNG early entropy source...
I (275) cpu_start: Pro cpu up.
I (283) cpu_start: Pro cpu start user code
I (284) cpu_start: cpu freq: 80000000 Hz
I (284) cpu_start: Application information:
I (286) cpu_start: Project name:     linkify
I (291) cpu_start: App version:      0.1.0.15
I (296) cpu_start: Compile time:     Jul 27 2022 12:19:06
I (303) cpu_start: ELF file SHA256:  9836a8e08b755372...
I (308) cpu_start: ESP-IDF:          v5.0-dev-4379-g36f49f361c
I (315) heap_init: Initializing. RAM available for dynamic allocation:
[... 37 lines cut ...]
I (511) wifi_init: WiFi SLP IRAM OP enabled
I (521) phy_init: phy_version 912,d001756,Jun  2 2022,16:28:07
I (561) wifi:mode : sta (7c:df:a1:d1:d3:d4)
I (561) wifi:enable tsf
I (561) power_save: esp_wifi_set_ps().
I (561) wifi:Set ps type: 2
I (2971) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2971) wifi:state: init -> auth (b0)
I (2981) wifi:state: auth -> init (8a0)
I (2981) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2981) wifi:new:<1,0>, old:<6,0>, ap:<255,255>, sta:<1,0>, prof:1
I (2991) wifi:state: init -> auth (b0)
I (3001) wifi:state: auth -> assoc (0)
I (3081) wifi:state: assoc -> run (10)
I (3491) wifi:connected with WIFI_OGGA, aid = 3, channel 1, BW20, bssid = 5c:a6:e6:8a:53:a4
I (3491) wifi:security: WPA2-PSK, phy: bgn, rssi: -79
I (3501) wifi:pm start, type: 2
I (3501) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
I (3501) wifi:BcnInt:102400, DTIM:1
```

Here is project and wireshark capture.
[scan.zip](https://github.com/espressif/esp-idf/files/9199054/scan.zip)
[ch1.pcapng.gz](https://github.com/espressif/esp-idf/files/9199111/ch1.pcapng.gz)
[ch6.pcapng.gz](https://github.com/espressif/esp-idf/files/9199112/ch6.pcapng.gz)

nishanth-radja, 2022-07-28:
@HiFiPhile can you use this in your STA config and try it out.

.scan_method = WIFI_ALL_CHANNEL_SCAN
.sort_method = WIFI_CONNECT_AP_BY_SIGNAL,

HiFiPhile, 2022-07-28:
> @HiFiPhile can you use this in your STA config and try it out.
>
>.scan_method = WIFI_ALL_CHANNEL_SCAN
.sort_method = WIFI_CONNECT_AP_BY_SIGNAL,

I already have `WIFI_ALL_CHANNEL_SCAN` in my config. `WIFI_CONNECT_AP_BY_SIGNAL` is the default value (0) and changes nothing.

nishanth-radja, 2022-08-01:
@HiFiPhile can you pls enable the wpa_supplicant logs and mbedtls logs, From the logs provided by you it tried to asscoiate with ch6 and then got deauth .
I (2971) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2971) wifi:state: init -> auth (b0)
I (2981) wifi:state: auth -> init (8a0)

From the sniffer capture, STA is connected in ch-6 and it is not present in ch-1.
can you keep the sniffer running until the STA connects to ch1 from ch6.

HiFiPhile, 2022-08-01:
I saw AP sent deauth just after authentication request.
I searched `Reason code: Disassociated because sending STA is leaving (or has left) BSS (0x0008)` but nothing really useful, somebody says `non-aggressive load balancing` but others have this error on single AP system.
[deauth.pcapng.gz](https://github.com/espressif/esp-idf/files/9232353/deauth.pcapng.gz)

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:mode:DIO, clock div:Emode:DIO, clock div:ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6110,len:0x1374
load:0x403ce000,len:0x934
load:0x403cf600,len:0x2c9c
entry 0x403ce000
I (30) boot: ESP-IDF v5.0-dev-4379-g36f49f361c 2nd stage bootloader
I (30) boot: compile time 10:20:49
I (30) boot: chip revision: V003
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 coredump         Unknown data     01 03 00010000 00010000
I (87) boot:  3 factory          factory app      00 00 00020000 00140000
I (94) boot:  4 ota_0            OTA app          00 10 00160000 00140000
I (102) boot:  5 ota_1            OTA app          00 11 002a0000 00140000
I (109) boot:  6 otadata          OTA data         01 00 003e0000 00002000
I (117) boot:  7 nvs_key          NVS keys         01 04 003e2000 00001000
I (124) boot: End of partition table
I (129) boot: Defaulting to factory image
I (133) esp_image: segment 0: paddr=00020020 vaddr=3c070020 size=1c998h (117144) map
I (161) esp_image: segment 1: paddr=0003c9c0 vaddr=3fc8f200 size=027fch ( 10236) load
I (164) esp_image: segment 2: paddr=0003f1c4 vaddr=40380000 size=00e54h (  3668) load
I (168) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=6805ch (426076) map
I (246) esp_image: segment 4: paddr=000a8084 vaddr=40380e54 size=0e334h ( 58164) load
I (258) esp_image: segment 5: paddr=000b63c0 vaddr=50000010 size=00004h (     4) load
I (258) esp_image: segment 6: paddr=000b63cc vaddr=50000018 size=00010h (    16) load
I (268) boot: Loaded app from partition at offset 0x20000
I (270) boot: Disabling RNG early entropy source...
I (287) cpu_start: Pro cpu up.
D (287) efuse: In EFUSE_BLK2__DATA4_REG is used 2 bits starting with 0 bit
D (287) efuse: In EFUSE_BLK2__DATA4_REG is used 8 bits starting with 12 bit
D (293) efuse: In EFUSE_BLK1__DATA5_REG is used 5 bits starting with 5 bit
D (300) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 7 bit
D (307) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 14 bit
D (314) efuse: In EFUSE_BLK1__DATA4_REG is used 8 bits starting with 21 bit
D (321) efuse: In EFUSE_BLK1__DATA4_REG is used 3 bits starting with 29 bit
D (328) efuse: In EFUSE_BLK1__DATA5_REG is used 5 bits starting with 0 bit
D (343) clk: RTC_SLOW_CLK calibration value: 3756992
I (352) cpu_start: Pro cpu start user code
I (352) cpu_start: cpu freq: 80000000 Hz
I (352) cpu_start: Application information:
I (355) cpu_start: Project name:     linkify
I (360) cpu_start: App version:      0.1.0.15
I (365) cpu_start: Compile time:     Aug  1 2022 09:47:16
I (371) cpu_start: ELF file SHA256:  0b25c60645e43d64...
ELF file not found. You need to build & flash the project before running 'monitor', and the binary on the device must match the one in the build directory exactly.
I (377) cpu_start: ESP-IDF:          v5.0-dev-4379-g36f49f361c
D (383) memory_layout: Checking 4 reserved memory ranges:
D (389) memory_layout: Reserved memory range 0x3fc80000 - 0x3fc8f200
D (395) memory_layout: Reserved memory range 0x3fc8f200 - 0x3fc95be0
D (402) memory_layout: Reserved memory range 0x3fcdf060 - 0x3fce0000
D (408) memory_layout: Reserved memory range 0x50000000 - 0x50000028
[... 432 lines cut ...]
D (5058) esp_netif_lwip: if0x3fc9e208 start ip lost tmr: no need start because netif=0x3fc9e2b0 interval=120 ip=0
I (5058) wifi:BcnInt:102400, DTIM:1
D (5068) wifi:set max rate: from <rate=130, phy=3, sig=0> to <rate=144, phy=3 sig=0>
D (5078) wifi:sig_b=0, sig_g=0, sig_n=0, max_b=22, max_g=108, max_n=144
D (5088) wifi:update trc
D (5088) esp_netif_lwip: starting dhcp client
D (5088) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (5098) event: running post WIFI_EVENT:4 with handler 0x420039c0 and context 0x3fc9f2b8 on loop 0x3fc9d2a4
I (8878) power_save: In main task 1 dummy 1
I (8878) power_save: In main task 1 dummy 1
I (8898) power_save: In main task 1 dummy 1
I (8898) power_save: In main task 1 dummy 1
I (8918) power_save: In main task 1 dummy 1
I (8918) power_save: In main task 2 dummy 2
I (8938) power_save: In main task 3 dummy 3
I (8938) power_save: In main task 4 dummy 4
I (12938) power_save: In main task 1 dummy 1
I (12938) power_save: In main task 1 dummy 1
I (12958) power_save: In main task 1 dummy 1
I (12958) power_save: In main task 1 dummy 1
I (12978) power_save: In main task 1 dummy 1
I (12978) power_save: In main task 2 dummy 2
I (12998) power_save: In main task 3 dummy 3
I (12998) power_save: In main task 4 dummy 4
```

nishanth-radja, 2022-08-01:
@HiFiPhile Sta is connecting to ch 1 since it has been rejected by ch-6 AP. can you try to connect to ch6 AP by switching off ch 1 AP.To see if it connects without the presence of other AP.

marko-savic, 2022-08-11:
@nishanth-radja any update or workaround on this issue? It gives us a huge issue as there are 80+ devices fighting for 4 APs on the factory shop floor.

AxelLin, 2022-08-20:
> @marko-savic The patch which was provided for this issue,Is not yet merged into 4.4,Issue will be present in 4.4.1

@nishanth-radja 
The original issue report was against v4.3.2-198-gb9a96186fd.
Could you clarify if this is fixed in 4.3 branch or not?

nishanth-radja, 2022-08-22:
@AxelLin No,Its not merged in 4.3 branch.

AxelLin, 2022-08-30:
@nishanth-radja 

In https://github.com/espressif/esp-idf/issues/8269#issuecomment-1032315458, you mentioned that you are also observing the same behavior in lab.
But this issue is not yet fix even though you can reproduce it?

nishanth-radja, 2022-09-01:
Hi @AxelLin, we are able to see the issue in lab.Dev team is working on the same for a stable fix.

AxelLin, 2022-09-02:
> Hi @AxelLin, we are able to see the issue in lab.Dev team is working on the same for a stable fix.

Could you confirm only esp32c3 has this issue or all esp32 chips have the same issue?

AxelLin, 2022-09-02:
> > @nishanth-radja I've updated to 4.4.1. In the release note of 4.4.1, there is nothing about this issue. Can you help me with the patch, please?
> 
> @marko-savic The issue is not mentioned in the release notes because the issue was on AP side.

If the issue is on AP side, why 8266 works fine? https://github.com/espressif/esp-idf/issues/8269#issuecomment-1032381462

vik-gokhale, 2022-09-06:
Hi @marko-savic 
please allow me to suggest a few changes in the network profile that is setup in the application.
'.threshold.rssi = -77' is more like a generic range that seems to be covering multiple APs in the office facility setup.
Can you please use a better range that you want the device to connect to?
For instance - if the nearest AP to the device is at -40dBm, it is appropriate to use '.threshold.rssi = -50'. 
Please let us know if this solves the problem?

AxelLin, 2022-09-06:
> Hi @marko-savic please allow me to suggest a few changes in the network profile that is setup in the application. '.threshold.rssi = -77' is more like a generic range that seems to be covering [...]

@vik-gokhale 
This sounds a strange suggestion to me.
So the user needs to know the nearest AP's RSSI, then set proper .threshold.rssi to make it work?

I don't understand why you think rssi threshold setting is related to choosing the best signal AP.
AFAICT, the WIFI_CONNECT_AP_BY_SIGNAL should work with .threshold.rssi = DEFAULT_RSSI

vik-gokhale, 2022-09-06:
@AxelLin 
Upon using WIFI_CONNECT_AP_BY_SIGNAL, ESP device does try connecting to the nearest AP.
In case the AP does rejects or does not accept the initial connection for whatever reason (including a bug on the AP), RSSI threshold can be made use of to choose the next best alternate.

User does not need to know the exact RSSI of the nearest AP, but the Threshold is more like "try connecting to an RSSI better than the set threshold, if available".

AxelLin, 2022-09-06:
> @AxelLin Upon using WIFI_CONNECT_AP_BY_SIGNAL, ESP device does try connecting to the nearest AP. In case the AP does rejects or does not accept the initial connection for whatever reason (including a bug on the AP, WiFi protocol limitations etc), RSSI threshold can be made use of to choose the next best alternate.

@vik-gokhale 
You just set threshold and hope the issue will be filtered out by threshold. That won't work.
Did you ever check why 8266 works fine? https://github.com/espressif/esp-idf/issues/8269#issuecomment-1032381462

[28 more replies]

closed 2024-02-27 by Alvin1Zhang