espressif/esp-idf #8272: I2C initialisiation err

Wtpaul, 2022-01-21:
I believe there is an error on line 647 of components/driver/i2c.c

I2C_CHECK(i2c_get_clk_src(i2c_conf) != I2C_SCLK_MAX, I2C_CLK_FLAG_ERR_STR, ESP_ERR_INVALID_ARG);

This check will flag an error if the i2c bus speed is not set to Max
I believe this should be 

I2C_CHECK(i2c_get_clk_src(i2c_conf) <= I2C_SCLK_MAX, I2C_CLK_FLAG_ERR_STR, ESP_ERR_INVALID_ARG);

o-marshmallow, 2022-01-24:
Hi @Wtpaul ,

I guess you are talking about IDF v4.3, as `master` branch doesn't have macro anymore.
In theory, I would say yes, `i2c_get_clk_src(i2c_conf)` must not be greater than  or equal to `I2C_SCLK_MAX`, but in practice this will never happen. Here is the implementation of `i2c_get_clk_src`:
```
static i2c_sclk_t i2c_get_clk_src(const i2c_config_t *i2c_conf)
{
    for (i2c_sclk_t clk = I2C_SCLK_DEFAULT + 1; clk < I2C_SCLK_MAX; clk++) {
        /* Something, with a return inside */
    }
    return I2C_SCLK_MAX;     // flag invalid;
}
```
We can never return a value greater than `I2C_SCLK_MAX`. Plus, that function is static, we are guaranteed this implementation will remain in this file, and thus always correlated with `i2c_get_clk_src(i2c_conf) != I2C_SCLK_MAX` test

Wtpaul, 2022-01-24:
Hi @o-marshmallow 
Thanks for your reply
Appologies, yes it is IDF v4.3

The problem is.. 
when i try to config my i2c bus as a master at 100kHz (not the max speed)
the function i2c_param_config() exits the function at line 647 without running the rest of the configuaration.

The '!=I2C_SCLK_MAX' means that the master I2C bus can only be configured to the max speed.
Changing this line to '>I2C_SCLK_MAX' ( '>=' would also be a mistake) allows the i2c config to be completed up to and including the max bus speed.

I hope this makes sense.  
Thanks

o-marshmallow, 2022-01-25:
Hi @Wtpaul ,

I just tried the `i2c_tools` example on the latest commit of `release/v4.3` on an ESP32-WROVER-B module, after setting frequency to 100KHz. It works fine, the effective frequency I get is around 92KHz.

Which module are you using and which IDF commit in particular? Do you have a minimal example that reproduces the error?
Here is a simple config you can try adding in your project:
```
static esp_err_t i2c_master_driver_initialize(void)
{
    i2c_config_t conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = YOUR_I2C_SDA_PIN,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_io_num = YOUR_I2C_SCL_PIN,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = 100000,
    };
    return i2c_param_config(i2c_port, &conf);
}
```

`I2C_CHECK` macro will make sure that the condition we pass to it is *true*, if it is not, then an error occurs. Thus, `I2C_CHECK(i2c_get_clk_src(i2c_conf) != I2C_SCLK_MAX)` makes sure that the condition is *true*, so, the config is *not* `I2C_SCLK_MAX`.

If you change this line to `I2C_CHECK(i2c_get_clk_src(i2c_conf) <= I2C_SCLK_MAX)`, it means that you allow the config to be `I2C_SCLK_MAX`, which is not a valid value, here are the valid values:
```
I2C_SCLK_DEFAULT = 0,    /*!< I2C source clock not selected*/
I2C_SCLK_APB,            /*!< I2C source clock from APB, 80M*/
I2C_SCLK_XTAL,           /*!< I2C source clock from XTAL, 40M */
I2C_SCLK_RTC,            /*!< I2C source clock from 8M RTC, 8M */
I2C_SCLK_REF_TICK,       /*!< I2C source clock from REF_TICK, 1M */
```

Wtpaul, 2022-01-25:
OK
I was wrong..
Appologies for wasting your time. 

These flags are clock source and not clock frequency.
My issue was fixed by adding conf.clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL;

    ```
conf.mode				= I2C_MODE_MASTER;
    conf.sda_io_num			= 26;
    conf.scl_io_num			= 25;
    conf.sda_pullup_en		= GPIO_PULLUP_ENABLE;
    conf.scl_pullup_en		= GPIO_PULLUP_ENABLE;
    conf.master.clk_speed	= 100000;
    conf.clk_flags			= I2C_SCLK_SRC_FLAG_FOR_NOMAL;
```

o-marshmallow, 2022-01-26:
@Wtpaul Great, thanks for providing your solution, this will help others!
Closing this issue.

closed 2022-01-26 by o-marshmallow