espressif/esp-idf #8273: the demo esp-idf/examples/peripherals/lcd/rgb_panel/  runing fail !

Kingwulin, 2022-01-22:
I build the code sucess!
but when i run it on esp32s3
there is some promble

`ELF file SHA256: 1cad12071299277c

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375888
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x16f8
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f90
entry 0x403b6248
I (29) boot: ESP-IDF v5.0-dev-1293-ge090b6b031 2nd stage bootloader
I (29) boot: compile time 09:32:15
I (29) boot: chip revision: 0
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0d740h ( 55104) map
I (108) esp_image: segment 1: paddr=0001d768 vaddr=3fc91e10 size=028b0h ( 10416) load
I (110) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=3fafch (260860) map
I (162) esp_image: segment 3: paddr=0005fb24 vaddr=3fc946c0 size=003c4h (   964) load
I (162) esp_image: segment 4: paddr=0005fef0 vaddr=40374000 size=0de0ch ( 56844) load
I (180) esp_image: segment 5: paddr=0006dd04 vaddr=50000000 size=00010h (    16) load
I (186) boot: Loaded app from partition at offset 0x10000
I (186) boot: Disabling RNG early entropy source...
I (200) opi psram: vendor id    : 0x00 (UNKNOWN)
I (200) opi psram: dev id       : 0x03 (generation 4)
I (200) opi psram: density      : 0x07 (256 Mbit)
I (205) opi psram: good-die     : 0x01 (Pass)
I (210) opi psram: Latency      : 0x01 (Fixed)
I (215) opi psram: VCC          : 0x01 (3V)
I (220) opi psram: SRF          : 0x00 (Slow Refresh)
I (226) opi psram: BurstType    : 0x01 ( Wrap)
I (231) opi psram: BurstLen     : 0x03 (1024 Byte)
I (237) opi psram: Readlatency  : 0x07 (20 cycles@Fixed)
I (243) opi psram: DriveStrength: 0x03 (1/8)
W (247) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
W (258) MSPI Timing: tuning fail, best point is fallen back to index 1
I (265) spiram: Found 32MB SPI RAM device
I (269) spiram: Speed: 80MHz
I (273) spiram: Initialized, cache is in normal (1-core) mode.
E (279) spiram: Virtual address not enough for PSRAM!

abort() was called at PC 0x40376475 on core 0

Backtrace:0x40375be2:0x3fceb2400x40379935:0x3fceb260 0x4037fa72:0x3fceb280 0x40376475:0x3fceb2f0 0x403753ca:0x3fceb310 0x403bb0c4:0x3fceb340 0x403bb52d:0x3fceb380 0x403b62b1:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED

`

how can i solve it,thanks!

suda-morris, 2022-01-24:
What's the PSRAM model you're using? Are you sure your ESP32-S3 is embedded with an octal PSRAM? We recommend using octal PSRAM for RGB LCD to achieve higher throughput.

Kingwulin, 2022-01-24:
> What's the PSRAM model you're using? Are you sure your ESP32-S3 is embedded with an octal PSRAM? We recommend using octal PSRAM for RGB LCD to achieve higher throughput.

I am using a board name  "ESP32-S3-DevKitM-1" which sale here [https://m.tb.cn/h.fPKDSkZ?sm=8066c8?tk=fdjH2c6move](url)
I don't kown what PSRAM use for 
can I configure the param and make it work well
or shall i need to add some other handware

Kingwulin, 2022-01-24:
hello which one I have to select if want to make the demo running ok
ESP32-S3-DevKitC-1
1、N8 
2、N8R2 
3、N8R8  
4、N32R8V
[image]

suda-morris, 2022-01-24:
Have you checked the datasheet? https://www.espressif.com/en/support/documents/technical-documents?keys=&field_type_tid%5B%5D=840
The S3-DevKitM User Guide says the module used is **ESP32-S3-MINI-1**, and seeing this [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3-mini-1_mini-1u_datasheet_en.pdf), the PSRAM size is zero, which means you board doesn't have a PSRAM.

Kingwulin, 2022-01-24:
Hello 
I want to sure that the S3-DevKitC have a PSRAM
And which type I should select below:
1、N8
2、N8R2
3、N8R8
4、N32R8V

suda-morris, 2022-01-25:
N8R8 should be OK in this case.

BarakChamo, 2022-01-25:
Having the exact same issue with the same example running on an ESP32-S3-DevKitC-1 v1.0 E2N8R2 on master (5.0)

Initially the correct size SPI RAM is detected:
```
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 2MB
```

However the spiram lib then incorrectly detects 32MB and fails
```
W (247) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
W (258) MSPI Timing: tuning fail, best point is fallen back to index 1
I (265) spiram: Found 32MB SPI RAM device
I (269) spiram: Speed: 80MHz
I (273) spiram: Initialized, cache is in normal (1-core) mode.
E (279) spiram: Virtual address not enough for PSRAM!
```

BarakChamo, 2022-01-25:
Just to add to that, no issues running the i80 example with SPIRAM enabled in the config using the same board.

suda-morris, 2022-01-26:
Hi @BarakChamo 
The RGB panel example has a default configuration that enables **Octal PSRAM**, see [here](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/rgb_panel/sdkconfig.defaults.esp32s3) And your board probably is with a **Quad PSRAM**.

BarakChamo, 2022-01-26:
@suda-morris thanks for your reply.

I actually caught that and tried switching to Quad in the config but had the same issue.
Is there anything in the code that relies on Octal SPI RAM or is it safe to assume that this should still be working with the C1 devkit?

Kingwulin, 2022-01-26:
> Having the exact same issue with the same example running on an ESP32-S3-DevKitC-1 v1.0 E2N8R2 on master (5.0)
> 
> Initially the correct size SPI RAM is detected:
> 
> ```
> I (32) boot.esp32s3: Boot SPI Speed : 80MHz
> I (37) boot.esp32s3: SPI Mode       : DIO
> I (42) boot.esp32s3: SPI Flash Size : 2MB
> ```
> 
> However the spiram lib then incorrectly detects 32MB and fails
> 
> ```
> W (247) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
> W (258) MSPI Timing: tuning fail, best point is fallen back to index 1
> I (265) spiram: Found 32MB SPI RAM device
> I (269) spiram: Speed: 80MHz
> I (273) spiram: Initialized, cache is in normal (1-core) mode.
> E (279) spiram: Virtual address not enough for PSRAM!
> ```
> 
> update:
> 
> This is the actual abort log
> 
> ```
> abort() was called at PC 0x40376475 on core 0
> 0x40376475: esp_spiram_init_cache at C:/Users/Barak/esp/esp-idf/components/esp_hw_support/port/esp32s3/spiram.c:69 (discriminator 3)
> 
> 
> 
> Backtrace:0x40375be2:0x3fceb2400x40379935:0x3fceb260 0x4037fa72:0x3fceb280 0x40376475:0x3fceb2f0 0x403753ca:0x3fceb310 0x403bb0c4:0x3fceb340 0x403bb52d:0x3fceb380 0x403b62b1:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED
> 0x40375be2: panic_abort at C:/Users/Barak/esp/esp-idf/components/esp_system/panic.c:403
> 
> 0x40379935: esp_system_abort at C:/Users/Barak/esp/esp-idf/components/esp_system/esp_system.c:130
> 
> 0x4037fa72: abort at C:/Users/Barak/esp/esp-idf/components/newlib/abort.c:46
> 
> 0x40376475: esp_spiram_init_cache at C:/Users/Barak/esp/esp-idf/components/esp_hw_support/port/esp32s3/spiram.c:69 (discriminator 3)
> 
> 0x403753ca: call_start_cpu0 at C:/Users/Barak/esp/esp-idf/components/esp_system/port/cpu_start.c:410
> ```

@BarakChamo Hi,would please add log [here](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/port/esp32s3/opiram_psram.c#:~:text=s_print_psram_info(%26mode_reg)%3B,0x7%20%3F%20PSRAM_SIZE_32MB%20%3A%200%3B) and to see what s_psram_size and mode_reg are
Thanks

Kingwulin, 2022-01-26:
> Having the exact same issue with the same example running on an ESP32-S3-DevKitC-1 v1.0 E2N8R2 on master (5.0)
>
> Initially the correct size SPI RAM is detected:
>
> ```
> I (32) boot.esp32s3: Boot SPI [...]

@BarakChamo Hi,would please add log [here](https://github.com/espressif/esp-idf/blob/41761fc762e859a76148ede014e296b4093eefb0/components/esp_hw_support/port/esp32s3/opiram_psram.c#L328) and to see what s_psram_size and mode_reg are
Thanks

Kingwulin, 2022-01-26:
> Having the exact same issue with the same example running on an ESP32-S3-DevKitC-1 v1.0 E2N8R2 on master (5.0)
>
> Initially the correct size SPI RAM is detected:
>
> ```
> I (32) boot.esp32s3: Boot SPI [...]

@BarakChamo Hi,would you please show me a log here what the val of mode_reg.mr2.density and s_psram_size are [here](https://github.com/espressif/esp-idf/blob/41761fc762e859a76148ede014e296b4093eefb0/components/esp_hw_support/port/esp32s3/opiram_psram.c#L328)
Thank you

Kingwulin, 2022-01-27:
> @suda-morris thanks for your reply.
> 
> I actually caught that and tried switching to Quad in the config but had the same issue. Is there anything in the code that relies on Octal SPI RAM or is it safe to assume that this should still be working with the C1 devkit?

I had the question same as your, did you solve it

Kingwulin, 2022-01-27:
> @suda-morris thanks for your reply.
> 
> I actually caught that and tried switching to Quad in the config but had the same issue. Is there anything in the code that relies on Octal SPI RAM or is it safe to assume that this should still be working with the C1 devkit?

@BarakChamo @suda-morris  I had the question same as it, how can i deal with it

Kingwulin, 2022-01-27:
**it running well when i change to ESP32-S3-DevKitC-1  N8R8**
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x16f8
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f90
entry 0x403b6248
I (24) boot: ESP-IDF v5.0-dev-1350-g41761fc762 2nd stage bootloader
I (25) boot: compile time 17:12:39
I (25) boot: chip revision: 0
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 2MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0d740h ( 55104) map
I (103) esp_image: segment 1: paddr=0001d768 vaddr=3fc91e10 size=028b0h ( 10416) load
I (106) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=3fafch (260860) map
I (157) esp_image: segment 3: paddr=0005fb24 vaddr=3fc946c0 size=003c4h (   964) load
I (158) esp_image: segment 4: paddr=0005fef0 vaddr=40374000 size=0de0ch ( 56844) load
I (175) esp_image: segment 5: paddr=0006dd04 vaddr=50000000 size=00010h (    16) load
I (181) boot: Loaded app from partition at offset 0x10000
I (181) boot: Disabling RNG early entropy source...
I (196) opi psram: vendor id    : 0x0d (AP)
I (196) opi psram: dev id       : 0x02 (generation 3)
I (196) opi psram: density      : 0x03 (64 Mbit)
I (200) opi psram: good-die     : 0x01 (Pass)
I (205) opi psram: Latency      : 0x01 (Fixed)
I (210) opi psram: VCC          : 0x01 (3V)
I (215) opi psram: SRF          : 0x01 (Fast Refresh)
I (221) opi psram: BurstType    : 0x01 (Hybrid Wrap)
I (227) opi psram: BurstLen     : 0x01 (32 Byte)
I (232) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (238) opi psram: DriveStrength: 0x00 (1/1)
W (243) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (254) spiram: Found 8MB SPI RAM device
I (258) spiram: Speed: 80MHz
I (261) spiram: Initialized, cache is in normal (1-core) mode.
I (268) cpu_start: Pro cpu up.
I (271) cpu_start: Starting app cpu, entry point is 0x40375244
I (0) cpu_start: App cpu up.
I (698) spiram: SPI SRAM memory test OK
I (707) cpu_start: Pro cpu start user code
I (707) cpu_start: cpu freq: 160000000 Hz
I (707) cpu_start: Application information:
I (710) cpu_start: Project name:     rgb_panel
I (715) cpu_start: App version:      v5.0-dev-1350-g41761fc762
I (721) cpu_start: Compile time:     Jan 26 2022 17:11:58
I (727) cpu_start: ELF file SHA256:  72f032e498a0383a...
I (733) cpu_start: ESP-IDF:          v5.0-dev-1350-g41761fc762
I (740) heap_init: Initializing. RAM available for dynamic allocation:
I (747) heap_init: At 3FC958D0 len 0004A730 (297 KiB): D/IRAM
I (754) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (760) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (766) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (773) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (781) spi_flash: detected chip: generic
I (785) spi_flash: flash io: dio
W (789) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (803) sleep: Configure to isolate all GPIO pins in sleep state
I (809) sleep: Enable automatic switching of GPIO sleep configuration
I (817) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (827) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (837) example: Turn off LCD backlight
I (847) gpio: GPIO[39]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (857) example: Install RGB panel driver
I (867) example: Turn on LCD backlight
I (867) example: Initialize LVGL library
I (867) example: Register display driver to LVGL
I (877) example: Install LVGL tick timer
I (877) example: Display LVGL Scatter Chart

```

suda-morris, 2022-01-27:
@BarakChamo There are some Kconfig combinations to make the Flash and PSRAM work at a specific frequency. [See this doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html)

Kingwulin, 2022-01-27:
> @BarakChamo There are some Kconfig combinations to make the Flash and PSRAM work at a specific frequency. [See this doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html)

I am using the panel as the pdf below,but I can't make my panel work or light
I am not sure VLED+/- pins in the panel should to connect which pin in esp32 board
And I just using the RGB data at low bit (R0-R4,G0-G5,B0-B4)
[2021923171633414.pdf](https://github.com/espressif/esp-idf/files/7948772/2021923171633414.pdf)

Kingwulin, 2022-01-27:
> @BarakChamo There are some Kconfig combinations to make the Flash and PSRAM work at a specific frequency. [See this doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html)

Holle ,I can make 480*272 running well
for  800*480  I modify code 
```
// The pixel number in horizontal and vertical
#define EXAMPLE_LCD_H_RES              800//480
#define EXAMPLE_LCD_V_RES              480//272
```
but i can't running well,I have not  idea to deal with it,below is the datasheet of panel(800*480) 
need your hand ,Thanks
[CC0432I40R-01D (单IC）.pdf](https://github.com/espressif/esp-idf/files/7949782/CC0432I40R-01D.IC.pdf)

hpwit, 2022-02-23:
Hello I have the same issue Think my module is E2N8

` (186) boot: Disabling RNG early entropy source...
I (201) opi psram: vendor id    : 0x00 (UNKNOWN)
I (201) opi psram: dev id       : 0x03 (generation 4)
I (201) opi psram: density      : 0x07 (256 Mbit)
I (206) opi psram: good-die     : 0x01 (Pass)
I (211) opi psram: Latency      : 0x01 (Fixed)
I (216) opi psram: VCC          : 0x01 (3V)
I (221) opi psram: SRF          : 0x00 (Slow Refresh)
I (226) opi psram: BurstType    : 0x01 ( Wrap)
I (232) opi psram: BurstLen     : 0x03 (1024 Byte)
I (237) opi psram: Readlatency  : 0x07 (20 cycles@Fixed)
I (243) opi psram: DriveStrength: 0x03 (1/8)
W (248) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
W (259) MSPI Timing: tuning fail, best point is fallen back to index 1
I (265) spiram: Found 32MB SPI RAM device
I (270) spiram: Speed: 80MHz
I (274) spiram: Initialized, cache is in normal (1-core) mode.
E (280) spiram: Virtual address not enough for PSRAM!`
do you have tips or should I buy new boards

krzychb, 2022-02-24:
> Hello I have the same issue Think my module is E2N8

Hi @hpwit, could you please post the complete part number? Searching on the internet I could not find the matching ESP32-S3 module (i.e. with "E2N8" in the name).

hpwit, 2022-02-24:
hello @krzychb 
 here is the board ESP32-S3-DEVKITC-1-N8
I have ordered ESP32-S3-DEVKITC-1-N8R8 as it seems to work with that chip
thank you

Alvin1Zhang, 2022-05-16:
Thanks for reporting and sharing the updates, sorry for slow turnaround, fix is available at https://github.com/espressif/esp-idf/commit/ca1e2af1d6518245ecdc8265e122f577def7698d, feel free to reopen.

closed 2022-05-16 by Alvin1Zhang

prabhu012, 2022-06-23:
hello I'm also getting same error

W (750) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
W (760) MSPI Timing: tuning fail, best point is fallen back to index 1
I (767) spiram: Found 256MBit SPI RAM device
I (772) spiram: SPI RAM mode: sram 80m
I (776) spiram: PSRAM initialized, cache is in normal (1-core) mode.

what i do how to fixed it

ESP32DE, 2023-05-18:
..just come in for bookmark and been updated