espressif/esp-idf #8274: esp_partition_get_sha256()

HondaRulez, 2022-01-23:
## Environment

own board.
- Module or chip used:  [ESP32-WROOM-3|]
- IDF version (run ``git describe --tags`` to find it):
    release 4.3.1, but seems all are affected.
- Build System:         [platformio]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    toolchain-xtensa-esp32 8.4.0+2021r1
- Operating System:     [Linux|]
- Using an IDE?: [Yes platformio]

## Problem Description
esp_partition_get_sha256() function returns ESP_OK on tampered app image (type: app/factory)
the API says the hash checked before return.

### Expected Behavior
returns with ESP_ERR_IMAGE_INVALID

### Actual Behavior
returns ESP_OK

### Steps to reproduce

modify a random byte in the firmware.bin
upload to a nonboot partition (bootlader will reject the boot on that due the  invalid hash)

start the  test app on a bootable partition...

### Code to reproduce this issue

part = esp_partition_find_first(ESP_PARTITION_TYPE_APP, ESP_PARTITION_SUBTYPE_ANY, "factory")

err = esp_partition_get_sha256(part, sha);

 if(err != ESP_OK) {
        ESP_LOGE(dtag,"Hash failed");
  }

boarchuz, 2022-01-23:
More details:
The descriptions for `esp_partition_get_sha256` and `bootloader_common_get_sha256_of_partition` include:

> The hash is verified before returning, if app content is invalid then the function returns ESP_ERR_IMAGE_INVALID.

However, for app partitions with a hash appended, the metadata is read, then the hash is simply copied and ESP_OK returned:
https://github.com/espressif/esp-idf/blob/faf0f61cdba67fa5f7e680f2f5b0966c8d69ce9a/components/bootloader_support/src/bootloader_common.c#L150-L162

In `esp_image_get_metadata`, do_verify = false and sha_handle = NULL such that the full image is not actually hashed or even read, only the header and hash are read in order to populate the metadata:
https://github.com/espressif/esp-idf/blob/faf0f61cdba67fa5f7e680f2f5b0966c8d69ce9a/components/bootloader_support/src/esp_image_format.c#L283-L301

igrr, 2022-01-23:
Thanks for reporting! Looks like an issue in the function documentation, it never actually verified the hash.

That said, adding another function which would verify the hash does make sense.

HondaRulez, 2022-01-23:
yes, I already checked the IDF source, so I j wrote a small function to check the image integrity.

esp_err_t check_app_sha(esp_partition_t *part)
{
    uint8_t hash[ESP_PARTITION_HASH_LEN];

    esp_err_t err = ESP_OK;
    esp_image_metadata_t md;

    const esp_partition_pos_t ppos = {
        .offset = part->address,
        .size = part->size,
    };

    do {
        if((err = esp_image_get_metadata(&ppos, &md)) != ESP_OK)
            break;
        if(! md.image.hash_appended) {
            err = ESP_ERR_IMAGE_INVALID;
            break;
        }
        if((err = bootloader_sha256_flash_contents(part->address, (md.image_len-ESP_PARTITION_HASH_LEN), hash)) != ESP_OK)
            break;

        if(memcmp(md.image_digest, hash, ESP_PARTITION_HASH_LEN) != 0) {
            err = ESP_ERR_IMAGE_INVALID;
            break;
        }
    } while (0);

//    ESP_LOG_BUFFER_HEX(dtag, md.image_digest,32);
//    ESP_LOG_BUFFER_HEX(dtag, hash,32);

    return(err);
}

KonstantinKondrashov, 2022-01-24:
Hi @HondaRulez!
Yes, it is a bug that was added after ca481e1 . Before these changes we called `esp_image_verify()` instead of  `esp_image_get_metadata()` but due to the wrong comment, we decided to skip checking the hash.

I added a check for this case in `bootloader_common_get_sha256_of_partition()`. We will create bakports for 4.4 and 4.3.

```diff
diff --git a/components/bootloader_support/src/bootloader_common.c b/components/bootloader_support/src/bootloader_common.c
index e4594f2..709504d 100644
--- a/components/bootloader_support/src/bootloader_common.c
+++ b/components/bootloader_support/src/bootloader_common.c
@@ -158,6 +158,12 @@ esp_err_t bootloader_common_get_sha256_of_partition (uint32_t address, uint32_t
         }
         if (data.image.hash_appended) {
             memcpy(out_sha_256, data.image_digest, ESP_PARTITION_HASH_LEN);
+            uint8_t calc_sha256[ESP_PARTITION_HASH_LEN];
+            // The hash is verified before returning, if app content is invalid then the function returns ESP_ERR_IMAGE_INVALID.
+            esp_err_t error = bootloader_sha256_flash_contents(address, data.image_len - ESP_PARTITION_HASH_LEN, calc_sha256);
+            if (error || memcmp(data.image_digest, calc_sha256, ESP_PARTITION_HASH_LEN) != 0) {
+                return ESP_ERR_IMAGE_INVALID;
+            }
             return ESP_OK;
         }
         // If image doesn't have a appended hash then hash calculates for entire image.
```

closed 2022-01-25 by espressif-bot