espressif/esp-idf #8275: TWAI on Pin 19/20 on ESP32S3

franz-ms-muc, 2022-01-23:
i use a TWAI on Pin 19/20 on ESP32S3.
ESP-IDF v4.4-rc1
chip revision: 0
CONFIG_EXAMPLE_TX_GPIO_NUM=19
CONFIG_EXAMPLE_RX_GPIO_NUM=20

when try to send CAN MEssage i get: 

E (1153) TWAI Master: hw_CanSendMsg failed, 259 ESP_ERR_INVALID_STATE

on ohter Pins works fine. 
e.g. 
CONFIG_EXAMPLE_TX_GPIO_NUM=4
CONFIG_EXAMPLE_RX_GPIO_NUM=5

no Problem. 

i know USB is on Pin 19/20. 
how to disable ?

franz-ms-muc, 2022-01-23:
ESP-IDF v5.0-dev-1093-gdf1f9ec26e same.

franz-ms-muc, 2022-01-24:
if test twai_self_test
with 
CONFIG_EXAMPLE_TX_GPIO_NUM=19
CONFIG_EXAMPLE_RX_GPIO_NUM=20

the Error picture is a bit different, but also fail: 

```
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375640
0x40375640: esp_restart_noos at /home/franz/esp-idf-v4.4-rc1/components/esp_system/port/soc/esp32s3/system_internal.c:143 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x164c
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f58
entry 0x403b6248
I (29) boot: ESP-IDF v4.4-rc1 2nd stage bootloader
I (29) boot: compile time 12:07:29
I (29) boot: chip revision: 0
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (40) boot.esp32s3: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=07d38h ( 32056) map
I (102) esp_image: segment 1: paddr=00017d60 vaddr=3fc90810 size=02674h (  9844) load
I (107) esp_image: segment 2: paddr=0001a3dc vaddr=40374000 size=05c3ch ( 23612) load
I (119) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=18cc4h (101572) map
I (140) esp_image: segment 4: paddr=00038cec vaddr=40379c3c size=06bcch ( 27596) load
I (146) esp_image: segment 5: paddr=0003f8c0 vaddr=50000000 size=00010h (    16) load
I (152) boot: Loaded app from partition at offset 0x10000
I (152) boot: Disabling RNG early entropy source...
I (169) cpu_start: Pro cpu up.
I (169) cpu_start: Starting app cpu, entry point is 0x403750bc
0x403750bc: call_start_cpu1 at /home/franz/esp-idf-v4.4-rc1/components/esp_system/port/cpu_start.c:156

I (147) cpu_start: App cpu up.
I (183) cpu_start: Pro cpu start user code
I (183) cpu_start: cpu freq: 160000000
I (183) cpu_start: Application information:
I (186) cpu_start: Project name:     twai_self_test_example
I (192) cpu_start: App version:      adb92ab-dirty
I (197) cpu_start: Compile time:     Jan 24 2022 12:07:45
I (204) cpu_start: ELF file SHA256:  d2aad83298687ecb...
I (210) cpu_start: ESP-IDF:          v4.4-rc1
I (215) heap_init: Initializing. RAM available for dynamic allocation:
I (222) heap_init: At 3FC93890 len 0004C770 (305 KiB): D/IRAM
I (228) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (235) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (241) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (248) spi_flash: detected chip: generic
I (252) spi_flash: flash io: dio
W (256) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (270) sleep: Configure to isolate all GPIO pins in sleep state
I (276) sleep: Enable automatic switching of GPIO sleep configuration
I (283) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (304) TWAI Self Test: Driver installed
I (304) TWAI Self Test: Driver started
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40378e08
0x40378e08: _esp_error_check_failed at /home/franz/esp-idf-v4.4-rc1/components/esp_system/esp_err.c:42

file: "../main/twai_self_test_example_main.c" line 66
func: twai_transmit_task
expression: twai_transmit(&tx_msg, portMAX_DELAY)

abort() was called at PC 0x40378e0b on core 0
0x40378e0b: _esp_error_check_failed at /home/franz/esp-idf-v4.4-rc1/components/esp_system/esp_err.c:43

Backtrace:0x4037598a:0x3fce38e00x40378e15:0x3fce3900 0x4037ec32:0x3fce3920 0x40378e0b:0x3fce3990 0x42005589:0x3fce39b0 0x4037ba5d:0x3fce39f0 
0x4037598a: panic_abort at /home/franz/esp-idf-v4.4-rc1/components/esp_system/panic.c:402

0x40378e15: esp_system_abort at /home/franz/esp-idf-v4.4-rc1/components/esp_system/esp_system.c:121

0x4037ec32: abort at /home/franz/esp-idf-v4.4-rc1/components/newlib/abort.c:46

0x40378e0b: _esp_error_check_failed at /home/franz/esp-idf-v4.4-rc1/components/esp_system/esp_err.c:43

0x42005589: twai_transmit_task at /home/franz/git/krone/isobus_sample/twai_self_test/build/../main/twai_self_test_example_main.c:66 (discriminator 1)

0x4037ba5d: vPortTaskWrapper at /home/franz/esp-idf-v4.4-rc1/components/freertos/port/xtensa/port.c:131

ELF file SHA256: d2aad83298687ecb

Rebooting...

```

Dazza0, 2022-01-28:
@franz-ms-muc Try adding the code snippet below at the start of `app_main`. Pins 19 and 20 are by default connected to the internal USB PHY,  So we need to set a couple their IO MUX registers to route them to the GPIO matrix instead of their dedicated functions (i.e., the USB PHY).

```c
#include "soc/gpio_periph.h"
#include "hal/gpio_hal.h"

...

    gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[TX_GPIO_NUM], PIN_FUNC_GPIO);
    gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[RX_GPIO_NUM], PIN_FUNC_GPIO);
```

franz-ms-muc, 2022-02-22:
cool. seems to work. ! THANKS !

closed 2022-02-22 by franz-ms-muc

franz-ms-muc, 2022-02-25:
Confirm again: working !