espressif/esp-idf #8276: esp function with shared stack

hdpklm, 2022-01-24:
Hello

I try to build this code, but got this esrror. any idea ???

https://docs.espressif.com/projects/esp-idf/en/v4.4-beta1/esp32/api-reference/system/esp_function_with_shared_stack.html

```
../main/blink_example_main.c: In function 'call_share':
../main/blink_example_main.c:85:5: error: unknown type name 'SemaphoreHandle_t'; did you mean 'xSemaphoreHandle'?
     SemaphoreHandle_t printf_lock = xSemaphoreCreateMutex();
     ^~~~~~~~~~~~~~~~~
     xSemaphoreHandle
../main/blink_example_main.c:85:37: error: implicit declaration of function 'xSemaphoreCreateMutex'; did you mean 'xSemaphoreHandle'? [-Werror=implicit-function-declaration]
     SemaphoreHandle_t printf_lock = xSemaphoreCreateMutex();
                                     ^~~~~~~~~~~~~~~~~~~~~
                                     xSemaphoreHandle
../main/blink_example_main.c:85:23: warning: unused variable 'printf_lock' [-Wunused-variable]
     SemaphoreHandle_t printf_lock = xSemaphoreCreateMutex();
                       ^~~~~~~~~~~
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.
```

ESP-Marius, 2022-01-24:
`SemaphoreHandle_t` is a Freertos structure, you probably forgot to include the relevant headers.

Try adding:

```
#include "freertos/FreeRTOS.h"
#include "freertos/semphr.h"
```

In general our documentation code snippets are just meant to illustrate a simple usage of e.g. a function, not a fully compilable example, but I agree that in this case we made it even more confusing by wrapping it in an`app_main`.

hdpklm, 2022-01-24:
thank you @ESP-Marius its build and work 🌹🌹🌹😍😍.

and need to include this file 
`#include "esp_expression_with_stack.h"`

ESP-Marius, 2022-01-24:
That's good to hear! 

Marking this issue as resolved then.

closed 2022-01-24 by ESP-Marius