espressif/esp-idf #8277: Failure running cmake:

Nnicholasnjue, 2022-01-24:
Hello, I've been stuck with setting up esp-idf on eclipse for the past one week now. Right now am getting this error:

cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\Users\ENGINEER\esp\esp-idf\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32 C:\Users\ENGINEER\eclipse-workspace\hello_world
Failure running cmake: 

when trying to compile the hello_world example on eclipse.
Mysteriously with the esp_idf cmd and power shell, am able to compile and upload to esp32.

Coffeeye, 2022-01-27:
Happened to me on reinstal with ESP-IDF Windows Installer after [Configuring CMake Toolchain](https://github.com/espressif/idf-eclipse-plugin#configuring-cmake-toolchain)

kolipakakondal, 2022-06-07:
Hello,

This could happen when PATH is not configured correctly. Eclipse plugin provides a way to configure all the required tools and paths, to do this:
1. Navigate to the `Espressif` Menu
2. ESP-IDF Tools Manager > Install Tools

This will check and install the required tools(If it's already installed, it won't install again!) and configure the paths to eclipse CDT build environment.

After running the "Install Tools" action from the eclipse, you could verify the configured paths in the preferences section.
`Preferences > C/C++ >  Build >  Environment`

harsha8370, 2024-08-09:
Hi @kolipakakondal 

I am getting this type of error
can you help me to resolve this issue?
[image]

kolipakakondal, 2024-08-09:
Hi @harsha8370 Please follow the instructions from [here](https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-to-configure-esp-adf-in-espressif-ide) to build an esp-adf example in the espressif-ide 3.0

I have updated the docs after verifying with the latest [espressif-ide 3.0](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.0.0) with [esp-adf-v2.6](https://github.com/espressif/esp-adf/releases/tag/v2.6)

[image]

harsha8370, 2024-08-09:
@kolipakakondal 
i am getting this type of an error 
can u pls tell me proper steps
[image]

harsha8370, 2024-08-19:
Hi @kolipakakondal 
I am getting cmake file  error and I configured all the things properly

please give me solutions to resolve this issue ASAP 
[image]