espressif/esp-idf #8278: TCP_POLL_Read Select error 113, error = software caused connection abortion, fd = 54

MadsHHLund, 2022-01-24:
TCP_POLL_Read Select error 113, error = software caused connection abortion, fd = 54

I experience difference whether my ESP32 devices are running cable or wifi.

I have a number of ESP32 devices that communicate via MQTT.

My devices use a Wiz610io circuit.

If the devices are running wifi, I have a pretty stable connection.

If the devices are connected with wired ethernet, I get once or twice a day, the following error ..
**"TCP_POLL_Read Select error 113, error = software caused connection abortion, fd = 54"**

MQTT then reports: MQTT_EVENT_DISCONNECTED

I have some code which after 5 min restarts the boxes and then it all runs again.

Heap: 150000
ESP-IDF v.4.3.1

Is there anyone who can give me a hint as to why I am experiencing these differences on cablet and wifi connections?
What should I look for?

Thanks in advance.

MadsHHLund, 2022-01-28:
**I believe I solved the issue**.

I implemented two updates in esp_eth_mac_w6100.c

      //ulTaskNotifyTake(pdTRUE, portMAX_DELAY);

       // check if the task receives any notification
        if (ulTaskNotifyTake(pdTRUE, pdMS_TO_TICKS(1000)) == 0 &&    // if no notification ...
            gpio_get_level(emac->int_gpio_num) != 0) {               // ...and no interrupt asserted
            continue;                                                // -> just continue to check again
        }

and 
    uint16_t int_level = __builtin_bswap16(0xFFFF);
     MAC_CHECK(w6100_write(emac, W6100_INTPTMR, &int_level, sizeof(int_level)) == ESP_OK, "write INTLEVEL failed", err, ESP_FAIL);

closed 2022-01-28 by MadsHHLund

tansirui1998, 2023-04-12:
I am not very familiar with this modification method, could you please explain the modification in detail? Thanks in advance