espressif/esp-idf #8279: Implement ftruncate function

igrr, 2022-01-24:
**Is your feature request related to a problem? Please describe.**

`truncate` function was requested in https://github.com/espressif/esp-idf/issues/1980 and implemented in https://github.com/espressif/esp-idf/commit/c77f84e788819a5070a9ea63b9a73fb0aea6df15, however the related `ftruncate` function was not implemented.

**Describe the solution you'd like**

* Implement `ftruncate` function in VFS
* Implement `ftruncate` in FAT VFS

**Describe alternatives you've considered**

None

**Additional context**

Requested in a comment to the closed issue: https://github.com/espressif/esp-idf/issues/1980#issuecomment-990092159
Issues that would be solved by implementing this:

* https://github.com/joltwallet/esp_littlefs/issues/57
* https://github.com/kuba--/zip/issues/231

Rajssss, 2022-02-16:
@igrr any update on this?

MabezDev, 2022-03-03:
FYI the Rust standard library relies on `ftruncate` being available, first reported here: https://github.com/esp-rs/rust/issues/100.

closed 2022-03-09 by espressif-bot