espressif/esp-idf #8283: I couldn't find the file name is "esp_rom_tjpgd.h", Pls help me

Alvinwang64, 2022-01-25:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

AxelLin, 2022-01-25:
https://github.com/espressif/esp-idf/blob/master/components/esp_rom/include/esp_rom_tjpgd.h

Alvinwang64, 2022-01-25:
I found it!! Thanks!

王兴文
联系电话：18600890991

 
https://github.com/espressif/esp-idf/blob/master/components/esp_rom/include/esp_rom_tjpgd.h

Alvin1Zhang, 2022-01-25:
@AxelLin Many thanks to the notes. @Alvinwang64 Thanks for reporting, feel free to reopen.

closed 2022-01-25 by Alvin1Zhang