espressif/esp-idf #8285: mqtt/ssl_ds/configure_ds.py errors out in production mode

bschick, 2022-01-25:
## Environment

- Module or chip used:  ESP32-C3-MINI-1-N4
- ESP-IDF v4.4-rc1
- Build System:        idf.py
- Operating System:    macOS
- Using an IDE?: No
- Power Supply:    USB

## Problem Description

Using esp-idf/examples/protocols/mqtt/ssl_ds/configure_ds.py to setup a private key for digital signatures fails when the` --production` option is used. The problem is that the fuse is read and write protected when the BLOCK_KEYN is burned and then the configure_ds.py script tries to read the fuse to confirm the hmac key was correctly written. This is the exception:

```
Traceback (most recent call last):
  File "../esp-idf/examples/protocols/mqtt/ssl_ds/configure_ds.py", line 399, in <module>
    main()
  File "../esp-idf/examples/protocols/mqtt/ssl_ds/configure_ds.py", line 385, in main
    hmac_key_read = configure_efuse_key_block(args, idf_target)
  File "../esp-idf/examples/protocols/mqtt/ssl_ds/configure_ds.py", line 275, in configure_efuse_key_block
    hmac_key_read = bytes.fromhex(hmac_key_read)
ValueError: non-hexadecimal number found in fromhex() arg at position 0
```
Fortunately, the script has not yet cleaned up the hmac_key_file so the NVS partition can be created manually and uploaded. Unrelated, but while editing the file it would also be nice if non-production mode also disabled write protect as well as read protect. 

### Expected Behavior

In production mode, configure_ds.py should not try to readback the BLOCK_KEY value to confirm it was written correctly. Note that this patch also disables write protect on the fuse when in non-production mode as well. Feel free to ignore that:

```
diff --git a/examples/protocols/mqtt/ssl_ds/configure_ds.py b/examples/protocols/mqtt/ssl_ds/configure_ds.py
old mode 100644
new mode 100755
index 44f2c9e64b..265501cbb2
--- a/examples/protocols/mqtt/ssl_ds/configure_ds.py
+++ b/examples/protocols/mqtt/ssl_ds/configure_ds.py
@@ -165,7 +165,7 @@ def efuse_summary(args, idf_target):
 #       The function makes use of the "espefuse.py" script to burn the HMAC key on the efuse.
 def efuse_burn_key(args, idf_target):
     # In case of a development (default) usecase we disable the read protection.
-    key_block_status = '--no-read-protect'
+    key_block_status = '--no-read-protect --no-write-protect'
 
     if args.production is True:
         # Whitespace character will have no additional effect on the command and
@@ -263,18 +263,23 @@ def configure_efuse_key_block(args, idf_target):
             key_file.write(new_hmac_key)
         # Burn efuse key
         efuse_burn_key(args, idf_target)
-        # Read fresh summary of the efuse to read the key value from efuse.
-        # If the key read from efuse matches with the key generated
-        # on host then burn_key operation was successfull
-        new_efuse_summary_json = get_efuse_summary_json(args, idf_target)
-        hmac_key_read = new_efuse_summary_json[key_blk]['value']
-        hmac_key_read = bytes.fromhex(hmac_key_read)
-        if new_hmac_key == hmac_key_read:
-            print('Key was successfully written to the efuse (KEY BLOCK %1d)' % (args.efuse_key_id))
+        if args.production is False:
+            # Read fresh summary of the efuse to read the key value from efuse.
+            # If the key read from efuse matches with the key generated
+            # on host then burn_key operation was successfull
+            new_efuse_summary_json = get_efuse_summary_json(args, idf_target)
+            hmac_key_read = new_efuse_summary_json[key_blk]['value']
+            print(hmac_key_read)
+            hmac_key_read = bytes.fromhex(hmac_key_read)
+            if new_hmac_key == hmac_key_read:
+                print('Key was successfully written to the efuse (KEY BLOCK %1d)' % (args.efuse_key_id))
+            else:
+                print('ERROR: Failed to burn the hmac key to efuse (KEY BLOCK %1d),'
+                      '\nPlease execute the script again using a different key id' % (args.efuse_key_id))
+                return None
         else:
-            print('ERROR: Failed to burn the hmac key to efuse (KEY BLOCK %1d),'
-                  '\nPlease execute the script again using a different key id' % (args.efuse_key_id))
-            return None
+            # In production mode, KEY BLOCK is read protected and cannot be validated
+            hmac_key_read = new_hmac_key
     else:
         # If the efuse key block is redable, then read the key from efuse block and use it for encrypting the RSA private key parameters.
         # If the efuse key block is not redable or it has key purpose set to a different
```

AdityaHPatwardhan, 2022-01-25:
@bschick Thanks for the issue and your patch as well. Seems the appropriate behaviour of the verifying mechanism was not updated for the production mode.

For non-production mode I think it would be helpful for the users to see the contents key block in the `summary` of the efuse. I think that allows some flexibility in the development mode. Making them Read protect would restrict the usage.

bschick, 2022-01-27:
@AdityaHPatwardhan agreed on your 2nd point. For non-production mode, my patch keeps the existing --no-read-protect and adds --no-write-protect, but that's pointless. Either way, the important part is removing the exception and process exit in production mode.

closed 2022-02-01 by espressif-bot