espressif/esp-idf #8286: Use ATECC608A with https-server

vats-egoo, 2022-01-25:
We would like to use the Microchip's ATECC608A-TNGTLS Secure Element to securely store critical device identification and authentication information. Our project uses a web server similar to the "https_server" from the IDF examples, but the web server seems to **not** use the ATECC608A. Our goal is to use the pre-provisioned with keys and certificates ATECC608A device.

We use the esp-cryptoauthlib and have followed the instructions, enabled ATECC608A in menuconfig, etc.

After looking in the esp-tls component source code, we have noticed that the HTTPS server component never uses the Secure Element for the web server, only for client, like it is here:

https://github.com/espressif/esp-idf/blob/5bad27d0d504e98e5206a7535a6143c9a914fbc6/components/esp-tls/esp_tls_mbedtls.c#L510

Our solution was to try to copy locally the esp-tls component and add this functionality in the set_server_config() function too, so we can use the secure context from the ATECC608A. We have copied this in set_server_config():
```c
#ifdef CONFIG_ESP_TLS_USE_SECURE_ELEMENT
        ret = esp_set_atecc608a_pki_context(tls, (esp_tls_cfg_t *)cfg);
        if (ret != ESP_OK) {
            return ret;
        }
#else
        ESP_LOGE(TAG, "Please enable secure element support for ESP-TLS in menuconfig");
        return ESP_FAIL;
#endif /* CONFIG_ESP_TLS_USE_SECURE_ELEMENT */
```
This seems to have worked, we can verify that we use the ATECC608A certificate and there is a speedup in the TLS handshake.

We are wondering if there is any reason not to have this functionality enabled by default. Is our solution correct, and could you suggest a better way to enable the ATECC608A when using the https-server component?

AdityaHPatwardhan, 2022-02-12:
Hi @vats-egoo Thanks for the issue.
There was no reason as such to not add the support in https_server. Maybe it got left out because it was never requested.
I see that you have used the code from the client part. I can surely raise the MR to make it a part of the esp-idf.

vats-egoo, 2022-02-17:
Hi, @AdityaHPatwardhan thank you for the reply.
It could be a useful addition to the https_server component. I would appreciate it if the feature is added as a part of the esp-idf.
We are using the ATECC608A and it is a very neat piece of hardware and would like to help if you need it in order to facilitate the process of integrating the new features.
Thanks.

AdityaHPatwardhan, 2022-03-21:
Hi @vats-egoo Sorry for the delayed reply, I have made this patch, Can you please test this on your side ?
[https_server_secure_elememt.patch.zip](https://github.com/espressif/esp-idf/files/8314288/https_server_secure_elememt.patch.zip)

AdityaHPatwardhan, 2022-03-22:
@vats-egoo Thanks for checking, I will raise a MR to merge it in the IDF.
Right thanks for pointing that out.
Actually In case of trust custom we will need the servercert to be provided externally, But I think that can be handled inside the esp-tls component when we have enabled secure element.
I will skip that check!

closed 2022-03-31 by espressif-bot