espressif/esp-idf #8289: Can not switch off bluetooth nimble logs

goofy2k, 2022-01-25:
I use ESP-IDF 4.2.

When I run a NimBLE  BLEClient, I get the logs shown below.  How can I switch these off? I looked in the menuconfig bluetooth/nimble section, but there are no items to control these logs: 

GATT procedure initiated: write no rsp; att_handle=16 len=19
ble_att_clt_tx_write_cmd(): 0x54:0x69:0x6d:0x65:0x20:0x73:0x69:0x6e:0x63:0x65:0x20:0x62:0x6f:0x6f:0x74:0x3a:0x20:0x32:0x35txed att command: write cmd; conn=0 handle=0x0010
host tx hci data; handle=0 length=26
ble_hs_hci_acl_tx(): 0x00 0x00 0x1a 0x00 0x16 0x00 0x04 0x00 0x52 0x10 0x00 0x54 0x69 0x6d 0x65 0x20 0x73 0x69 0x6e 0x63 0x65 0x20 0x62 0x6f 0x6f 0x74 0x3a 0x20 0x32 0x35
Number of Completed Packets: num_handles=1
handle:0 pkts:1

rahult-github, 2022-01-26:
Hi @goofy2k 

Can you please try below steps:

1. go to blecent example
2. Run "rm -rf sdkconfig sdkconfig.old build"
3. idf.py set-target esp32 
4. Compile , flash and run monitor.

The prints you have mentioned are visible when log level is set to debug.  

Please confirm that Component Config -> Log Output -> Default Log   points to "Info" .

If above steps still show the issue, request you to please share your sdkconfig .

goofy2k, 2022-01-26:
Thanks Rahult! This example does not produce the output. So it is OK.

I inspected my application.  It still contained a repetitive task copied from the example that I started from. This task contained some printf's for information. I removed those and now everything is fine. 

Please feel free to close the issue.

Fred

closed 2022-01-27 by rahult-github