espressif/esp-idf #8290: Can't read from I2C sensor with clock rate of 400kHz on core versions 2.0.1 and above

caiser01, 2022-01-26:
### Board

ESP32 Dev Module, ESP32 Wrover Module

### Device Description

ESP32 DevKit wired to AT42QT1070 I2C capacitive touch sensor chip

### Hardware Configuration

SDA and SCL are on default pins 21 and 22. Pull-up resistor values are 10k.

### Version

v2.0.2

### IDE Name

Arduino IDE 1.8.15

### Operating System

macOS, Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Problem:
I've been using the attached sketch to interface ESP32 modules to the AT42QT1070 capacitive touch sensor. Going back to 1.0.x versions of the Arduino ESP32 core, this sketch has always worked correctly. It continued to work for version 2.0.0 of the core.

Beginning with version 2.0.1, I can no longer read from the AT42QT1070 with the I2C clock speed set to 400kHz as I always have. The only way to get the sketch to read the AT42QT1070 with version 2.0.1 or 2.0.2 is to lower the I2C clock speed to 200kHz. If I revert the core version to 2.0.0, everything behaves as expected and the ESP32 can read the AT42QT1070 at 400kHz as it did before.

Expected behavior:
Using the attached sketch, the ESP32 should be able to read from the AT42QT1070 with a 400kHz I2C clock.

[ESP32_touch_panel_tester.zip](https://github.com/espressif/arduino-esp32/files/7916441/ESP32_touch_panel_tester.zip)

Sample logic analyzer captures reading the firmware version register of the AT42QT1070 which should always report 0x15:

Core version 2.0.0, I2C clock at 400kHz:
[image]
Core version 2.0.2, I2C clock at 200kHz:
[image]
Core version 2.0.2, I2C clock at 400kHz:
[image]

### Sketch

```cpp
Complete sketch and library for the AT42QT1070 attached in the description.

I2C configuration in void setup():
Wire.begin();
Wire.setClock(400000);

Register read function in AT42QT1070.cpp:
unsigned char AT42QT1070::readRegister(unsigned char regAddr){
  Wire.beginTransmission(AT42QT1070_ADDRESS);
  Wire.write(regAddr);
  Wire.endTransmission(false);
  
  Wire.requestFrom(AT42QT1070_ADDRESS, 1);
  unsigned char byteRead = Wire.read();
  return byteRead;
}
```

### Debug Message

```plain
Output with core version 2.0.2 and I2C clock at 200kHz showing correct values for Chip ID and firmware version:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0
[ ⸮⸮mum⸮⸮⸮⸮2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
ESP-IDF version is: v4.4-beta1-189-ga79dc75f0a
AT42QT1070 Tester
Chip ID is 0x2E
Chip firmware version is 0x15
This program continuously polls the *CHANGE pin
to determine whether a key was pressed.

Output with core version 2.0.2 and I2C clock at 400kHz showing incorrect values for Chip ID and firmware version:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0
[ ⸮⸮mum⸮⸮⸮⸮2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
ESP-IDF version is: v4.4-beta1-189-ga79dc75f0a
AT42QT1070 Tester
Chip ID is 0xFF
Chip firmware version is 0xFF
This program continuously polls the *CHANGE pin
to determine whether a key was pressed.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Alvin1Zhang, 2022-01-27:
Thanks for reporting.

caiser01, 2022-01-27:
You're welcome!

caiser01, 2022-04-06:
I've just tested this again with the same sample program attached above but with the new Arduino ESP32 core v2.0.3-RC1 and I'm still unable to communicate with the AT42QT1070 chip at clock rates above 200kHz. Switching back to Arduino ESP32 core v2.0.0 allows the sensor to communicate correctly at up to 400kHz.

The IDF version reported by my sample program when core v2.0.3-RC1 is installed is v4.4-367-gc29343eb94

The AT42QT1070 is the only chip I've tried so far that exhibits this problem. I will begin checking through other I2C chips I have to see if any of those are also affected. Maybe another chip is easier for other people to source and use in debugging this issue.

caiser01, 2022-07-11:
The problem I was having went away after upgrading to the latest Arduino ESP32 core v2.0.4 (with ESP-IDF v4.4.1-472-gc9140caf8c).

Thanks to those who resolved this bug, either intentionally or unintentionally!

closed 2022-07-11 by caiser01