espressif/esp-idf #8295: Unaligned access to PS-RAM fails with LoadStoreAlignment

pavlohamov, 2022-01-26:
## Environment

- Development Kit:      ESP32-S2-Saola-1
- Module or chip used: esp32s2
- IDF version  v5.0-dev-1350-g41761fc762   sha: 41761fc762
- Build System:   idf.py
- Compiler version xtensa-esp32s2-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:    Linux
- Using an IDE?:  NO
- Power Supply:         USB

## Problem Description

My device uses external 2 megs of PS-RAM for buffers allocation. It is mandatory to have unaligned access to those buffers.
On previous  IDF that i've used (sha: 9f303290d8) it was working fine. Now it fails with LoadStoreAlignment exception

### Expected Behavior
R/W operation succeeded 

### Actual Behavior
R/W operation fails 

### Steps to reproduce

1. Take sample from https://github.com/espressif/esp-idf/pull/8294
2. Build it and flash it into esp32s2-Saola
3. observe LoadStoreAlignment

## Debug Logs

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6268,len:0x684
load:0x4004c000,len:0x828
load:0x40050000,len:0x2b94
entry 0x4004c174
I (77) cache: Instruction cache 	: size 16KB, 4Ways, cache line size 32Byte
I (78) cache: Data cache 		: size 8KB, 4Ways, cache line size 32Byte
I (81) spiram: Found 16MBit SPI RAM device
I (86) spiram: SPI RAM mode: sram 80m
I (90) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (316) cpu_start: Pro cpu up.
I (324) cpu_start: Pro cpu start user code
I (324) cpu_start: cpu freq: 240000000 Hz
I (324) cpu_start: Application information:
I (327) cpu_start: Project name:     hello_world
I (332) cpu_start: App version:      v5.0-dev-1351-g2a721882ff-dirty
I (339) cpu_start: Compile time:     Jan 26 2022 17:08:04
I (345) cpu_start: ELF file SHA256:  275df55ff59758ea...
I (351) cpu_start: ESP-IDF:          v5.0-dev-1351-g2a721882ff-dirty
I (358) heap_init: Initializing. RAM available for dynamic allocation:
I (366) heap_init: At 3FFC9F00 len 00032100 (200 KiB): DRAM
I (372) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (378) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (522) spi_flash: detected chip: winbond
I (523) spi_flash: flash io: dio
W (523) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
W (533) cpu_start: eFuse virtual mode is enabled. If Secure boot or Flash encryption is enabled then it does not provide any security. FOR TESTING ONLY!
I (549) esp_core_dump_uart: Init core dump to UART
I (553) cpu_start: Starting scheduler on PRO CPU.

I (559) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
0x3ffd5bc0: ok 
Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008979c  PS      : 0x00060730  A0      : 0x800897fa  A1      : 0x3ffcd170  
0x4008979c: memRegTest at esp-idf/examples/spiramTest/main/main.c:50

A2      : 0x3f500980  A3      : 0x00100000  A4      : 0x0000000c  A5      : 0x3ffd5b94  
A6      : 0x00000006  A7      : 0x00000000  A8      : 0x3f500981  A9      : 0x00000001  
A10     : 0x000000d0  A11     : 0x3ffcd2e0  A12     : 0x00000001  A13     : 0xa39d1636  
A14     : 0x3ffd5b64  A15     : 0x000007ff  SAR     : 0x00000001  EXCCAUSE: 0x00000009  
EXCVADDR: 0x3f500981  LBEG    : 0x00000001  LEND    : 0xa39d1636  LCOUNT  : 0x40029451  
0x40029451: _xt_user_exc at esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:633

Backtrace:0x40089799:0x3ffcd1700x400897f7:0x3ffcd1a0 0x4008982e:0x3ffcd1d0 0x40099fb7:0x3ffcd200 0x4002fe5a:0x3ffcd230 
0x40089799: memRegTest at esp-idf/examples/spiramTest/main/main.c:48
0x400897f7: memTest at esp-idf/examples/spiramTest/main/main.c:66
0x4008982e: app_main at esp-idf/examples/spiramTest/main/main.c:74
0x40099fb7: main_task at esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:127
0x4002fe5a: vPortTaskWrapper at esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

## Find out that malloc on older IDF resulted in address 0x3fd80980, but new produces 0x3f500980

Other branch in main.c file (covered with #if 01) was used to find out exact address where unaligned access fails (in range from 0x3FF8_0000 to 0x3F50_0000). Result is 0x3F800000. Addresses lower than 0x3f800000 will fail for unaligned access. 

```
0x3f800400: 
ok
0x3f800000: 
ok
0x3f7ffc00: 
Guru Meditation Error: Core  0 panic'ed (Memory protection fault). 
Write operation at address 0x3f41f000 not permitted (INVALID_MEM_TYPE)

Core  0 register dump:
PC      : 0x4008975a  PS      : 0x00060534  A0      : 0x80089800  A1      : 0x3ffcd1b0  
0x4008975a: memRegTest at esp-idf/examples/spiramTest/main/main.c:26 (discriminator 3)

A2      : 0x3f7ffc00  A3      : 0x00000080  A4      : 0x0bd1321d  A5      : 0x3ffd5ba0  
A6      : 0x40087260  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x000000ad  
0x40087260: timer_task at esp-idf/components/esp_timer/src/esp_timer.c:384

A10     : 0x3f7ffc00  A11     : 0x3ffcd2f8  A12     : 0x0000000c  A13     : 0x3ffd5bac  
A14     : 0x00000006  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x0000000c  LEND    : 0x3ffd5bac  LCOUNT  : 0x400271d8  
0x400271d8: xt_highint4 at esp-idf/components/esp_system/port/soc/esp32s2/highint_hdl.S:58

Backtrace:0x40089757:0x3ffcd1b00x400897fd:0x3ffcd1e0 0x4009a0ab:0x3ffcd210 0x4002fe5a:0x3ffcd240 
0x40089757: memRegTest at esp-idf/examples/spiramTest/main/main.c:25 (discriminator 3)
0x400897fd: app_main at esp-idf/examples/spiramTest/main/main.c:81
0x4009a0ab: main_task at esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:127
0x4002fe5a: vPortTaskWrapper at esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```
BTW some open questions:
1) why access 0x3f7ffc00 ends up in MPU fault? 
2) why accessing to [0x3FF80000 - meg; 0x3FF80000] gives valid result? 
3) Seems like PS ram is mapped twice into high and low memories

Icarus113, 2022-01-28:
Hi, 0x3f7ffc00 is on a special bus where unaligned access is not supported (between 0x3f500000~0x3f800000) on esp32s2.
Whereas 0x3f800000 to 0x3ff80000 (also for PSRAM), as well as those addresses in Internal RAM do support the unaligned access (unaligned handled by hardware option, hardware will automatically deal with this).
    
Could you use IDF v4.4 version? We'll handle this before v5.0.

pavlohamov, 2022-01-31:
@Icarus113 Thank you for explanation. 4.4 works fine for my purpose. 
Will use it till v5 release, as you suggested

shadowpho, 2022-02-05:
I am seeing the same error, but I am on 4.4 strangely.  Will try to narrow it down.

> 
>  [0;32mI (2827) gpio: GPIO[44]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1
> | Pulldown: 0| Intr:0 [0m
>                                                     
> Guru Meditation Error: Core  0 panic'ed (Memory protection fault). 
>            
> Write operation at address 0x3f41f000 not permitted (INVALID_MEM_TYPE)
> 
>       
>                                                                                 
> Core  0 register dump:
>                                                         
> PC      : 0x40039217  PS      : 0x00060934  A0      : 0x8003a05b  A1      : 0x3f
> fd2f50  
>                                                                       
> A2      : 0x7ffa26e0  A3      : 0x40039cf0  A4      : 0x3ffd2fe0  A5      : 0x3f
> fd2e18  
>                                                                       
> A6      : 0x3ffcf8cc  A7      : 0xfffffffc  A8      : 0x00000000  A9      : 0x3f
> fd2f20  
>                                                                       
> A10     : 0x3ffd2e18  A11     : 0xe2afaa03  A12     : 0x00000001  A13     : 0xe2
> afaa03  
>                                                                       
> A14     : 0x3ffcf8cc  A15     : 0x00000020  SAR     : 0x0000000f  EXCCAUSE: 0x00
> 000007  
>                                                                       
> EXCVADDR: 0x00000000  LBEG    : 0x00000001  LEND    : 0xe2afaa03  LCOUNT  : 0x40
> 0296a0  
> 
>                                                                      
>                                                                                 
> Backtrace:0x40039214:0x3ffd2f500x4003a058:0x3ffd2f80 0x4003a388:0x3ffd2fb0 0x400
> 84142:0x3ffd2fe0 0x4008419c:0x3ffd3020 0x4002ec52:0x3ffd3060 0x4002e88b:0x3ffd30
> 90 0x400c0159:0x3ffd30d0 0x400c1f9c:0x3ffd3100 0x400c3713:0x3ffd3130 0x400c3a9e:
> 0x3ffd31c0 0x400c3df0:0x3ffd3210 0x400c1a48:0x3ffd3280 0x400c231a:0x3ffd32f0 0x4
> 00c23e1:0x3ffd3330 0x400c056f:0x3ffd3360 0x400c05a5:0x3ffd33a0 0x400871de:0x3ffd
> 33d0 0x400872d8:0x3ffd3400 0x4014112f:0x3ffd3480 
> 
> 
>                                                          
>                                                                                 
> ELF file SHA256: d3aa5f0ef1015143
> 
>                                                                 
> CPU halted.
>                             
> 
>

igrr, 2022-02-05:
@shadowpho could you please open a new issue and add the information mentioned in the issue template? Although the error reported also mentions "Memory protection fault", it doesn't look like the same issue as @pavlohamov has encountered.

shadowpho, 2022-02-06:
Hmm you are right the backtrace is different, even though it tries to access the same location

pavlohamov, 2022-08-15:
@Icarus113 is there any news for this issue? So far I took approach from esp32 code. Enabling ESP32_IRAM_AS_8BIT_ACCESSIBLE_MEMORY for esp32s2 and adding 
`+#define SOC_CACHE_APP_LOW       SOC_EXTRAM_DATA_LOW
+#define SOC_CACHE_APP_HIGH      SOC_EXTRAM_DATA_HIGH`
to the components/soc/esp32s2/include/soc/soc.h. 
Could such solution be merged into mainline? As you are saying that unaligned access is handled by hw for addresses > 0x3f800000 it seems to me that HW handling vs SW handling won't add much overhead. At least for my project (majorly using the external PS-RAM) I se not significant performance drop

Icarus113, 2022-08-16:
@pavlohamov  Hi, I'm currently working on this, should be ready in these days, but still needs  some time to clean up the code.

closed 2022-09-02 by espressif-bot