espressif/esp-idf #8296: TCP socket connect time out option

SimonPVS, 2022-01-26:
Is there an option to set the TCP socket connection time-out time?
Because when connecting to a not existing device, it takes long for the 'connect' function to return..

I've seen the option 'SO_CONTIMEO', but is seems like this is not implemented in the lwip stack used by esp-idf.

Another solution for me would be an option to close/stop a connecting socket, which will make the 'connect' function return.

Thanks!

igrr, 2022-01-26:
@SimonPVS could you see if this example helps? https://github.com/espressif/esp-idf/blob/41761fc762e859a76148ede014e296b4093eefb0/examples/protocols/sockets/non_blocking/main/non_blocking_socket_example.c#L136-L151

SimonPVS, 2022-01-28:
Hi! 
Thanks, this works for me. 
But in case someone has a solution using blocking sockets, they are still welcome to share :)

atanisoft, 2022-01-28:
@SimonPVS The approach used in arduino-esp32 is similar to the above except it uses LwIP API for connect.
https://github.com/espressif/arduino-esp32/blob/6a7bcabd6b7a33f074f93ed60e5cc4378d350b81/libraries/WiFi/src/WiFiClient.cpp#L213-L246

For blocking `connect()` API there is no good way to timeout the connect early other than multiple tasks that I'm aware of.

karawin, 2022-01-29:
setsockopt (sockfd, SOL_SOCKET, SO_RCVTIMEO, (char *)&timeout, sizeof(timeout);

david-cermak, 2022-04-11:
@SimonPVS The only way to configure a timeout on TCP connection is to switch the socket to non-blocking mode and wait (`select()` or `poll()`) until it's writable.

Please note, that POSIX doesn't define TCP connection timeout options for blocking sockets. While this is be implemented on some platforms (i.e.  on linux `SO_SNDTIMEO` could be used), it's not supported in lwip.

More information could be found for example in [Unit Network Programming](https://books.google.com/books?id=ptSC4LpwGA0C&pg=PA381#v=onepage&q&f=false) book, section `14.2. Socket Timeouts`:

```
... All three techniques work with IO operations (read, write...) ....
...select can be used to place a timeout on *connect* only in a non-blocking mode
... and the two socket options do not work with *connect*
```

closed 2022-05-06 by Alvin1Zhang