espressif/esp-idf #8297: BLE Provisioning in esp-idf v4.4 on Android 12 causes extra pairing prompt

joshgalicic, 2022-01-26:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.4
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0

- Operating System:     macOS
- Using an IDE?: No
- Power Supply:        external 5V

## Problem Description

BLE protocom provisioning on Android now triggers a prompt on the Android device asking the user to pair. On esp-idf version 4.3.2 the esp module can pair without prompting the user on the Android device. This happens with the provisioning example Android app on Android 12 on Pixel 6 pro.

### Expected Behavior

The provisioning app should list discovered devices, the user should select the device then be able to provision the device.

### Actual Behavior

The provisioning app lists discovered devices, upon selecting a device, a system notification is shown asking the user if they want to pair with the device. If the user ignores this notification or declines, the esp device is not provisionable.

### Steps to reproduce

1. Flash the BLE provisioning example to the esp module
2. Attempt to provision the device using the demo Android app
3. Select the scanned device and see the new system notification that stops immediate pairing

[image]

joshgalicic, 2022-01-27:
Similarly on iOS, the example provisioning app prompts to pair twice, once when the scanned device is selected, and again after the PoP token is submitted.

GeoffArmstrong, 2022-02-09:
I'm seeing this issue on both iOS and Android as well.  Works just fine on 4.2 and 4.3, but with 4.4 I'm seeing these prompts, and the root cause appears to be on the ESP32 SoC side since there is no change on the mobile app side.

laukik-hase, 2022-02-11:
Hello, @joshgalicic and @GeoffArmstrong!

Thank you for the issue report.
 
Could you try out this patch - [wifi_prov_ble_pair.zip](https://github.com/espressif/esp-idf/files/8048778/wifi_prov_ble_pair.zip)? It adds a config option `WIFI_PROV_BLE_SECURE_PAIRING` which can be toggled at `(Top) → Component config → Wi-Fi Provisioning Manager`. Disabling it should fix the issue.

GeoffArmstrong, 2022-02-12:
I applied the patch and disabled that option, and it fixed the issue. Thank
you!

rahult-github, 2022-02-23:
Hi @GeoffArmstrong  , @joshgalicic ,

 We have made a generic fix for this issue. Can you kindly confirm if issue is resolved with attached patch ? We can make it part of official release. 

[0001-Merge-branch-bugfix-wifi_prov_mgr_conn_issue-into-ma.patch.txt](https://github.com/espressif/esp-idf/files/8124454/0001-Merge-branch-bugfix-wifi_prov_mgr_conn_issue-into-ma.patch.txt)

Thanks,
 Rahul

GeoffArmstrong, 2022-02-24:
Hi @rahult-github 

I removed the previous patch that you sent and then applied this new patch however the problem described in this issue has  unfortunately reappeared.

rahult-github, 2022-02-25:
Hi @GeoffArmstrong ,

 Thanks for trying the patch. Can you please confirm that when the patch was applied, you ensured that previous pairing / bonding information on the remote phone ( android phone ) was removed ? 

Basically, it should be fresh pairing and then any instance of trying to pair / unpair should work. 

Thanks,
 Rahul

GeoffArmstrong, 2022-02-25:
Hey @rahult-github 

I have a question about this - should users be seeing a iOS/Android "pair" message at all?  Firmware built with IDF 4.3.x our app is able to communicate with the device without getting a pairing message. Going forward, is the user going to need to answer a pairing prompt from the mobile OS?

I'll test again with your most recent patch and actually accept the "pairing" request.  I just ignored the pairing message because I figured users shouldn't be getting it in the first place. Our specific goal of using the Bluetooth interface for our IoT device is to just initialize a WiFi connection, and then never use Bluetooth again. So for our needs increased security on a Bluetooth connection isn't as important as a simple setup experience since it's only used for a brief moment.

ndevev, 2022-02-28:
Hi @rahult-github 
I can confirm that the [first patch](https://github.com/espressif/esp-idf/files/8048778/wifi_prov_ble_pair.zip) is working, but the [second](https://github.com/espressif/esp-idf/files/8124454/0001-Merge-branch-bugfix-wifi_prov_mgr_conn_issue-into-ma.patch.txt) is not.

ndevev, 2022-03-02:
Hi @rahult-github,

Is it expected that the first one will be included in release/v4.4?

rahult-github, 2022-03-02:
Hi @ndevev , @GeoffArmstrong  
 
 We will need some more debugging, before we can simply disable some piece of code. 

 As pointed out by @GeoffArmstrong  in above comment: 

`Works just fine on 4.2 and 4.3, but with 4.4 I'm seeing these prompts,` 
 
However, the first patch disables the code which is present even on v4.3 . 

`https://github.com/espressif/esp-idf/blob/release/v4.3/components/protocomm/src/transports/protocomm_nimble.c#L498`

This issue may be specific to android version. We tried with multiple phones , but were of version Android 10 and 11 and are not able to reproduce the issue at our end.

May i request you to kindly share with us debug enabled logs ( with / without the patch ) so we can analyse and give a proper fix, rather than just disabling a segment of code that may cause regression.

Thanks,
 Rahul

ndevev, 2022-03-03:
Hi @rahult-github!

It's getting weirder. I tried again to generate the debug logs you asked, but now the whole provisioning doesn' t work. After the BLE connection, the service discovery is OK, then the Android app tries to send the prov-session command with the client pub key, but it is failed because of timeout. And no pairing request popup is appear. And there is nothing interesting in the debug log too.
The same thing happens with and without the patch and with the iPhone app.
I don’t understand why I experienced anything else last time. Maybe I didn't do fullclean.

```
D (00:00:32.260) protocomm_nimble: Inside BLE connect w/ conn_id - 0
I (00:00:33.139) protocomm_nimble: mtu update event; conn_handle=0 cid=4 mtu=256

D (00:00:37.457) protocomm_nimble: disconnect; reason=531
D (00:00:37.457) protocomm_nimble: Inside disconnect w/ session - 0
I (00:00:37.462) NimBLE: GAP procedure initiated: advertise;
I (00:00:37.465) NimBLE: disc_mode=2
I (00:00:37.469) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (00:00:37.480) NimBLE:
```

But v4.3.2-350-g040ae4ac72 still works well (and without any pairing popup).

rahult-github, 2022-03-03:
Hi @ndevev ,

 This looks completely different from the original issue. 
 Can you please enable debug logs by  modifying log level via menuconfig like below, and help capture one log ?

Component Config  --> Log Output --> Default Log Verbosity --> Debug 
Component Config --> Bluetooth --> NimBLE Options --> NimBLE Host log verbosity --> Debug Logs
 

Thanks,
 Rahul

ndevev, 2022-03-03:
It turned out that the provisioning was not successful due to turning the log verbosity level to debug.
So I cannot create successful provisioning connection if the debug log verbosity is debug.

GeoffArmstrong, 2022-03-10:
Hey @rahult-github 

> Thanks for trying the patch. Can you please confirm that when the patch was applied, you ensured that previous pairing / bonding information on the remote phone ( android phone ) was removed ?

I had some time today, so I tested again and this time used a phone that had previously never communicated with that ESP32 before via BLE just to be sure that there were no lingering problems, and the the prompting problem didn't exist with the [first patch](https://github.com/espressif/esp-idf/files/8048778/wifi_prov_ble_pair.zip) but it did with [the second](https://github.com/espressif/esp-idf/files/8124454/0001-Merge-branch-bugfix-wifi_prov_mgr_conn_issue-into-ma.patch.txt).

rahult-github, 2022-03-16:
Hi @GeoffArmstrong ,
  Link encryption is by default now enabled for characteristic read / write during provisioning. If link is not encrypted, it will go for link encrpytion which needs LTK. If LTK is not present, it will proceed for pairing to generate one. This should be the reason pop-ups are observed now. 

Can you please confirm if attached patch results in no pop ups during provisioning at your end. This patch should be applied in $IDF_PATH 

[Double_Pairing_Prompt.txt](https://github.com/espressif/esp-idf/files/8261863/Double_Pairing_Prompt.txt)

Thanks,
 Rahul

GeoffArmstrong, 2022-03-16:
Hi @rahult-github ,

Thanks, I reverted the previous patches and applied the [latest patch](https://github.com/espressif/esp-idf/files/8261863/Double_Pairing_Prompt.txt) and the pairing worked without any extra Bluetooth pairing prompts.

I've only done a few minutes of testing, but it's looking good so far and I'll follow up here if I find any more issues when doing more comprehensive regression testing.

Alvin1Zhang, 2022-05-06:
Thanks for reporting and sorry for slow turnaround, fixes are available

* master branch: https://github.com/espressif/esp-idf/commit/c4eb0c3584620d78e5e700b3e54cd6e25eff2fa2
* release:4.4: https://github.com/espressif/esp-idf/commit/1861ca57001a25ff621b975a6b003204d941d8d5
* release/4.3: https://github.com/espressif/esp-idf/commit/6a92fb2d76cd7bc1bc12233baee0ef69c0b08b33

Feel free to reopen if the issue still happens. Thanks.

closed 2022-05-06 by Alvin1Zhang