espressif/esp-idf #8298: To support 802.11r in STA mode

AxelLin, 2022-01-27:
The same request was in https://github.com/espressif/esp-idf/issues/3671
802.11k,v are supported but there is no update for 802.11r so far.
Add this issue for tacking 802.11r status.

Actually, I had asked the 802.11r status several times, summary as below:
21 Jun 2019, it says "we may prioritise 11r though" https://github.com/espressif/esp-idf/issues/3671#issuecomment-504357006
18 Dec 2020, it says "11r support will be added in future." https://github.com/espressif/esp-idf/issues/3671#issuecomment-747897503
25 Jan 2021, it says "We are targeting 2nd quarter of 2021." https://github.com/espressif/esp-idf/issues/3671#issuecomment-766896837
3 Aug 2021, it says "Our intention of providing the tentative timeline is to give visibility to everyone based on the current inputs." https://github.com/espressif/esp-idf/pull/7305#issuecomment-891679391

So I'm very confused about the 802.11r status, it's 2022 now, anyone can update the status? e.g. Will esp-idf-5.0 support 11r?

sagb2015, 2022-05-06:
This is merged internally. Should be available on github soon.

AxelLin, 2022-05-17:
Closed by 805b5c9

closed 2022-05-17 by AxelLin