espressif/esp-idf #8299: socket close problem

chegewara, 2022-01-27:
Hi,
im not sure it is a real issue or just i am having problem to use api.
Description:
- i am making c++ wrapper around TCP/UDP socket server api
- basically it is working, i can open multiple connections from client(s) and all seems to work
- when i close all client connections i can close TCP socket and recreate it as either TCP or UDP on the same port
- problem is when i close all active connections from esp-idf; all connections are closed properly, but i cant close socket
`shutdown()` and `close()` return 0, which means no error, but socket is not closed (fd is not released) and when i re-create socket, then fd is next consecutive value instead of first free and bind is returning `errno 112` which means `EADDRINUSE`
- i cant test with esp-idf example, because it is not ready to use multiple connections

- board: esp32
- esp-idf: 309c1113f4943e8c0eade3fb2c87d6e78ff34ac3

Thanks

chegewara, 2022-01-27:
Ok, ive found something. The address is released and can be reused after 120 seconds since disconnecting all tcp clients by esp32.
Looks like it is not an issue, just like i suspected, but please confirm it.

Thanks

ESP-YJM, 2022-01-27:
@chegewara This is not a problem. For the socket close, we will enter `TIME-WAIT` state when the peer not send `FIN` to us. This state may keep about 2 minutes, which is like you saw. This matches the TCP protocol. If you want to bind this address, you can use `SO_RESUEADDR` socket option. And if you want to close socket immediately, you can use `SO_LINGER` socket option to send RST to peer.

chegewara, 2022-01-27:
Thanks, i will try
Yes, i am aware i can use `SO_REUSEADDR`, but it is not helpful in case i want to change from TCP to UDP server type.

It is resolved/answered so i will close it now, thanks

closed 2022-01-27 by chegewara