espressif/esp-idf #8300: PPPOS MQTT Core Dump

Sjurinho, 2022-01-27:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.2.2
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System: Linux
- Using an IDE?: VSCode with the ESP-IDF extension
- Power Supply: USB

## Problem Description
The pppos_client example in the esp_modem example crashes and issues a core dump while connected to mqtt. It happens when the mqtt client receives a message.

//Detailed problem description goes here.
I am running an ESP32 connected to a Monarch Sequans GSM, communicating through UART. Then after a while it suddenly issues a core dump when trying to handle a received message (normally on an incoming ping response). I have attached a photo of the core dump issue, and it seems like it occurs within the **pppos.c** file.

### Code to reproduce this issue
The pppos_client_main.c example from the (esp_modem)[https://components.espressif.com/component/espressif/esp_modem] component (v.0.1.13). The only modification I have done to the code is removing the xEventGroupSet within the mqtt_handler to avoid disconnecting from the server.

## Debug Logs
[image]

Alvin1Zhang, 2022-01-27:
@Sjurinho Thanks for reporting, checked you might create a post on our forum https://esp32.com/viewtopic.php?f=13&p=91148#p91148.

nilesh-dryad, 2022-01-27:
I also faced this problem sometimes with my test setup. It happens very rarely but it happens. I am not able to reproduce it when it happens. Crash is exactly at the same file same line as you mentioned. I couldn't find the reason.

nilesh-dryad, 2022-01-28:
Hello I am adding coredump logs for more information. Thought it might help. This is the log from ESP32S2. I am also using external PSRAM in my application.

```
nileshkumar@Nilesh:~/Desktop/core_dump/BG10$ espcoredump.py  info_corefile -t raw -c ./203 border_sylva_1b.elf 
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffdae6c, name: 'tiT', GDB name: 'process 1073589868'

================== CURRENT THREAD REGISTERS ===================
exccause       0x1c (LoadProhibitedCause)
excvaddr       0xa
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
epc7           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x4008b457
pc             0x400d0150	0x400d0150 <pppos_input+108>
lbeg           0x0	0
lend           0x0	0
lcount         0x0	0
sar            0xe	14
ps             0x60e20	396832
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x800d0300	-2146630912
a1             0x3ffdadb0	1073589680
a2             0x3ffe1204	1073615364
a3             0xf0b8	61624
a4             0x0	0
a5             0x3ffe11a0	1073615264
a6             0x34	52
a7             0x3fff51dd	1073697245
a8             0x60	96
a9             0x3ffdad60	1073589600
a10            0x3ffdf7bc	1073608636
a11            0x3ffdae10	1073589776
a12            0x3ffdf7bc	1073608636
a13            0x3ffeaec0	1073655488
a14            0x1	1
a15            0x58c	1420

==================== CURRENT THREAD STACK =====================
#0  0x400d0150 in pppos_input (ppp=0x3ffe1204, s=0x3fff51dd <error: Cannot access memory at address 0x3fff51dd>, l=52) at /home/dani/dryad/gd-daniel/gw-mesh/devel/gw-mesh/esp-idf/components/lwip/lwip/src/netif/ppp/pppos.c:544
#1  0x400d0300 in pppos_input_sys (p=0x3fff51c4, inp=<optimized out>) at /home/dani/dryad/gd-daniel/gw-mesh/devel/gw-mesh/esp-idf/components/lwip/lwip/src/netif/ppp/pppos.c:454
#2  0x400bbea4 in tcpip_thread_handle_msg (msg=0x3ffee7ec) at /home/dani/dryad/gd-daniel/gw-mesh/devel/gw-mesh/esp-idf/components/lwip/lwip/src/api/tcpip.c:180
#3  0x400bbf23 in tcpip_thread (arg=0x0) at /home/dani/dryad/gd-daniel/gw-mesh/devel/gw-mesh/esp-idf/components/lwip/lwip/src/api/tcpip.c:154
#4  0x40031d4c in vPortTaskWrapper (pxCode=0x400bbef0 <tcpip_thread>, pvParameters=0x0) at /home/dani/dryad/gd-daniel/gw-mesh/devel/gw-mesh/esp-idf/components/freertos/port/xtensa/port.c:168
[... 183 lines cut ...]
.coredump.tasks.data 0x3ffd91b4 0x154 RW 
.coredump.tasks.data 0x3ffd90b0 0xfc RW 
.coredump.tasks.data 0x3ffde254 0x154 RW 
.coredump.tasks.data 0x3ffea850 0x120 RW 
.coredump.tasks.data 0x3ff9fd3c 0x154 RW 
.coredump.tasks.data 0x3ffd4b00 0x118 RW 
.coredump.tasks.data 0x3ffdde80 0x154 RW 
.coredump.tasks.data 0x3ffe8010 0xf8 RW 
.coredump.tasks.data 0x3ffddfd8 0x154 RW 
.coredump.tasks.data 0x3ffe8b10 0xf4 RW 
.coredump.tasks.data 0x3ffde504 0x154 RW 
.coredump.tasks.data 0x3fff1410 0x5d4 RW 
.coredump.tasks.data 0x3ffdcb9c 0x154 RW 
.coredump.tasks.data 0x3ffe6c50 0x2c0 RW 
.coredump.tasks.data 0x3ffddd28 0x154 RW 
.coredump.tasks.data 0x3ffddc20 0x100 RW 
.coredump.tasks.data 0x3ff9fa40 0x154 RW 
.coredump.tasks.data 0x3ff9f910 0x128 RW 
.coredump.tasks.data 0x3ffdbac8 0x154 RW 
.coredump.tasks.data 0x3ffdb970 0x150 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

david-cermak, 2022-01-28:
@Sjurinho @nilesh-dryad Thank you for this detailed report and +1 for sharing the core-dump!

Could you please try to cherry-pick changes in [this PR](https://github.com/espressif/esp-lwip/pull/24)?
Closed that some time ago, as I wasn't able to reproduce it, neither to understand how could the `in_tail` become `nullptr` even if the packet got corrupted.
If it helps I would very much appreciate if you could share the packet capture (esp. the hexdump of the packet that caused this)

nilesh-dryad, 2022-01-28:
@david-cermak Thank you for suggesting cherry pick. As I said it is very hard to reproduce it and I already had the coredump from previous crash. It just happens sometimes but I will put hexdump so when it happens I can share the hexdump.

Sjurinho, 2022-02-01:
@david-cermak thank you for the suggestion. I haven't set up a new test yet, but I will get a new test up and running soon. I will share the logs if/when I run into the issues again

Sjurinho, 2022-02-10:
A small update: 
I encountered the issue again today, and it seems like the solution you suggested worked @david-cermak
I've found out that the issue arises every time the modem sends a second 'rx break'. Then afterwards, when trying to respond to any incoming message it raises a core panic. Not sure exactly why though. Attached you can find an image of the pattern I am talking about
[image]

Sjurinho, 2022-02-10:
> A small update: I encountered the issue again today, and it seems like the solution you suggested worked @david-cermak I've found out that the issue arises every time the modem sends a second 'rx [...]

The screenshot is from before merging in the cherry-pick solution you suggested @david-cermak. After implementing, this is what happens
[image]

closed 2022-03-09 by espressif-bot