espressif/esp-idf #8301: ESP32 example (release v4.4)  “spiffsgen” crash, when change optimization level to "debug without optimization(-O0)"

gaupen1186, 2022-01-27:
esp32_devKitC_V4
windows 10
ESP-IDF Release v4.4

create example <spiffsgen>, then change optimization level to "debug without optimization(-O0)", download and run, it will crash.
Change optimization level to default "debug (-O0)", it will be OK again.

crash info:

I (0) cpu_start: App cpu up.
I (262) cpu_start: Pro cpu start user code
I (262) cpu_start: cpu freq: 160000000
I (262) cpu_start: Application information:
I (267) cpu_start: Project name:     spiffsgen
I (272) cpu_start: App version:      1
I (276) cpu_start: Compile time:     Jan 27 2022 17:36:11
I (282) cpu_start: ELF file SHA256:  90a5d6d3273faa50...
I (288) cpu_start: ESP-IDF:          v4.4-41-g55b1756d5f
I (295) heap_init: Initializing. RAM available for dynamic allocation:
I (302) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (308) heap_init: At 3FFB3288 len 0002CD78 (179 KiB): DRAM
I (314) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (320) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (327) heap_init: At 400921A0 len 0000DE60 (55 KiB): IRAM
I (335) spi_flash: detected chip: generic
I (338) spi_flash: flash io: dio
W (341) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (356) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (366) example: Initializing SPIFFS
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

Core  0 register dump:
PC      : 0x400f142c  PS      : 0x00060b34  A0      : 0x800847f0  A1      : 0x3ffb6560
0x400f142c: xPortSetInterruptMaskFromISR at D:/ESP32/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:562

A2      : 0x00000001  A3      : 0x15122500  A4      : 0x00060b20  A5      : 0x00000000
A6      : 0x0000cdcd  A7      : 0x3ffb6560  A8      : 0x80083897  A9      : 0x3ffb6520
A10     : 0x00000000  A11     : 0x15122500  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x0000cdcd  A15     : 0x3ffb6520  SAR     : 0x0000001b  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace:0x400f1429:0x3ffb65600x400847ed:0x3ffb65a0 0x40084d34:0x3ffb65d0 0x400849a8:0x3ffb6670 0x400e2432:0x3ffb66d0 0x400e2220:0x3ffb67b0 0x400e226a:0x3ffb67e0 0x400e23a1:0x3ffb6820 0x400d7810:0x3ffb6860 0x400d7d32:0x3ffb68c0 0x400d750d:0x3ffb69a0 0x400f1d22:0x3ffb69f0
0x400f1429: get_count_bits_in_reg at D:/ESP32/esp-idf/components/efuse/src/esp_efuse_utility.c:331

0x400847ed: Cache_Flush at D:/ESP32/esp-idf/components/esp_rom/include/esp32/rom/cache.h:141

0x40084d34: spi_flash_mmap_pages at D:/ESP32/esp-idf/components/spi_flash/flash_mmap.c:279

0x400849a8: spi_flash_mmap at D:/ESP32/esp-idf/components/spi_flash/flash_mmap.c:156

0x400e2432: load_partitions at D:/ESP32/esp-idf/components/spi_flash/partition.c:184

0x400e2220: ensure_partitions_loaded at D:/ESP32/esp-idf/components/spi_flash/partition.c:74

0x400e226a: esp_partition_find at D:/ESP32/esp-idf/components/spi_flash/partition.c:87

0x400e23a1: esp_partition_find_first at D:/ESP32/esp-idf/components/spi_flash/partition.c:140

0x400d7810: esp_spiffs_init at D:/ESP32/esp-idf/components/spiffs/esp_spiffs.c:170

0x400d7d32: esp_vfs_spiffs_register at D:/ESP32/esp-idf/components/spiffs/esp_spiffs.c:381

0x400d750d: app_main at D:\ESP32\eclipse-workspace\spiffsgen\build/../main/spiffsgen_example_main.c:97

0x400f1d22: main_task at D:/ESP32/esp-idf/components/freertos/port/port_common.c:129

ESP-Marius, 2022-01-27:
Hi, 

Thanks for reporting this. Seems like what's happening is that `xPortSetInterruptMaskFromISR` is not getting inlined at `-O0`, and therefor during `Cache_Flush()`->`esp_ipc_isr_stall_other_cpu()` we end up accessing a function in flash while the cache is disabled.

[image]

We'll fix it!

closed 2022-02-23 by espressif-bot

ginkgm, 2022-04-17:
Target Branch | Merged by commit | Supported since IDF ver. | Last   N.A. IDF ver.
-- | -- | -- | --
master | 06d8a5bf33fbc6d94bbcea75b3eb4623ff130275 | N.A. | v5.0-dev
release/v4.4 | c9ce438466c1bdffc1e683f34f12296bb9c77f61 | v4.4.1 | N.A.
release/v4.3 | cd598bd7c00b01e9385b199e3729f4d6cf22bf7d | N.A. | v4.3.2