espressif/esp-idf #8302: Problem SecureBootV2 and FlashEncryption

saidsnanou1, 2022-01-27:
## Environment

- Development Kit:   None
- Kit version : v3
- Module or chip used:  ESP32-WROVER-IE
- IDF version (run ``git describe --tags`` to find it): v4.3.1-480-g233dc30fb1
- Build System:    idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:   Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: no
- Power Supply:    USB

## Problem Description
First, I enabled Flash Encryption (development mode) without secure boot V2 and it works fine. Then I disabled Flash Encryption and enabled secure boot V2 and it works fine. 
But when I tested both together I got this error. 
```
C:\ESP-IoT\esp-idf\work\OTA_github_s\hello_OTA>idf.py encrypted-flash monitor
Executing action: encrypted-flash
Serial port COM6
Connecting.....
Detecting chip type... ESP32
Running ninja in directory c:\esp-iot\esp-idf\work\ota_github_s\hello_ota\build
Executing "ninja encrypted-flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D C:\ESP-IoT\esp-idf\components\espt...P-IoT/esp-idf/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32 -p COM6 -b 460800 --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB --encrypt 0xd000 partition_table/partition-table.bin 0x12000 ota_data_initial.bin 0x20000 hello_world.bin
esptool.py v3.2-dev
Serial port COM6
Connecting....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 44:17:93:64:79:4c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x0000d000 to 0x0000dfff...
Flash will be erased from 0x00012000 to 0x00013fff...
Flash will be erased from 0x00020000 to 0x00050fff...

WARNING: - compress and encrypt options are mutually exclusive
Will flash partition_table/partition-table.bin uncompressed
Writing at 0x0000d000... (100 %)
Wrote 16384 bytes at 0x0000d000 in 0.4 seconds (316.9 kbit/s)...

WARNING: - compress and encrypt options are mutually exclusive
Will flash ota_data_initial.bin uncompressed
Writing at 0x00012000... (100 %)
Wrote 16384 bytes at 0x00012000 in 0.5 seconds (261.9 kbit/s)...

WARNING: - compress and encrypt options are mutually exclusive
Will flash hello_world.bin uncompressed
Writing at 0x00020000... (7 %)
Writing at 0x00024000... (15 %)
Writing at 0x00028000... (23 %)
Writing at 0x0002c000... (30 %)
Writing at 0x00030000... (38 %)
Writing at 0x00034000... (46 %)
Writing at 0x00038000... (53 %)
Writing at 0x0003c000... (61 %)
Writing at 0x00040000... (69 %)
Writing at 0x00044000... (76 %)
Writing at 0x00048000... (84 %)
Writing at 0x0004c000... (92 %)
Writing at 0x00050000... (100 %)
[... 94 lines cut ...]
BLOCK2 (BLOCK2):                         Secure boot key
   = a2 36 c8 66 24 71 e4 64 2e a1 16 a9 9d 96 1d 81 e1 2f 96 61 ce 56 f9 ac cb 1d 21 e1 13 df cb 88 R/-
BLOCK3 (BLOCK3):                         Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
```

2.  idf.py menuconfig

[image]

[image]

2. idf.py build
3. esptool.py --chip esp32 --port COM6 --baud 115200 --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x1000 ./build/bootloader/bootloader.bin
4. idf.py encrypted-flash monitor

## Other items if possible

- [ ] sdkconfig file 
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/7949473/sdkconfig.zip)

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

negativekelvin, 2022-01-27:
You have to use encrypt flag when flashing bootloader

saidsnanou1, 2022-01-28:
thank you very much, it works well.

mahavirj, 2022-01-28:
@saidsnanou1 

Glad to see that your problem is solved per suggestion from @negativekelvin!

To elaborate: Odd number of set bits in `FLASH_CRYPT_CNT` field signifies that flash encryption is enabled on device. And hence one must flash encrypted bootloader and other artifacts. Please refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#relevant-efuses for more information on this.

I will go ahead and close the issue.

closed 2022-01-28 by mahavirj