espressif/esp-idf #8304: nvs_set_blob leaks

hkbora, 2022-01-27:
Following up on this issue (https://github.com/espressif/esp-idf/issues/1642) I found a scenario where nvs_set_blob produces a memory leak in the range of 368 to 1480 bytes.

### Code to reproduce this issue

#include <iostream>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <nvs.h>
#include <sstream>
#include "esp_system.h"
#include <esp_log.h>
#include <nvs_flash.h>
#include <string.h>

#define MAIN_TAG "main"
#define BUF_SIZE 10000
#define LOOP_COUNT 10

std::string makeFileName(uint32_t index)
{
    std::ostringstream outStr;
    outStr << "testname_" << index;
    return outStr.str();
}

extern "C" void app_main()
{
    uint32_t start_free_heap = esp_get_free_heap_size();

    ESP_LOGI(MAIN_TAG, "START, free heap size: %d", start_free_heap);

    nvs_flash_init();
    
    esp_err_t error = 0;
    int i = 0;
    uint32_t length;
    nvs_handle nvsHandle;

    nvs_open("test2", NVS_READWRITE, &nvsHandle);
    ESP_LOGI(MAIN_TAG, "size: %d", BUF_SIZE);

    char * buffer = (char *)malloc(BUF_SIZE);
    memset(buffer, 0xFF, BUF_SIZE);

    char * buffer2 = (char *)malloc(BUF_SIZE);
    memset(buffer2, 0x00, BUF_SIZE);

    // Set blobb
    error = nvs_set_blob(nvsHandle, makeFileName(i).c_str(), buffer, BUF_SIZE);
    nvs_commit(nvsHandle);
    ESP_LOGI(MAIN_TAG, "set blob @%d, error = %d, error_name = %s, heap = %d", i, error, esp_err_to_name(error), esp_get_free_heap_size());
    usleep(10);

    // Get blob to prove its really there
    length = BUF_SIZE + 1;
    error = nvs_get_blob(nvsHandle, makeFileName(i).c_str(), buffer2, &length);
    nvs_commit(nvsHandle);
    ESP_LOGI(MAIN_TAG, "get blob @%d, error = %d, error_name = %s, heap = %d", i, error, esp_err_to_name(error), esp_get_free_heap_size());
    usleep(10);

    nvs_close(nvsHandle);

    free(buffer);
    free(buffer2);
    
    usleep(10);
    
    ESP_LOGI(MAIN_TAG, "START, free heap size: %d", start_free_heap);
    ESP_LOGI(MAIN_TAG, "STOP, free heap size: %d", esp_get_free_heap_size());
    ESP_LOGI(MAIN_TAG, "START - STOP, memory diff: %d", start_free_heap - esp_get_free_heap_size());
[... 244 lines cut ...]
I (308) cpu_start: cpu freq: 160000000
I (308) cpu_start: Application information:
I (313) cpu_start: Project name:     nvs-rw-value
I (318) cpu_start: App version:      1
I (322) cpu_start: Compile time:     Jan 26 2022 10:46:22
I (329) cpu_start: ELF file SHA256:  4162bd82d743830e...
I (334) cpu_start: ESP-IDF:          v4.3.2
I (340) heap_init: Initializing. RAM available for dynamic allocation:
I (347) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (353) heap_init: At 3FFB4930 len 0002B6D0 (173 KiB): DRAM
I (359) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (365) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (372) heap_init: At 4008BF68 len 00014098 (80 KiB): IRAM
I (379) spi_flash: detected chip: generic
I (383) spi_flash: flash io: dio
W (387) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (403) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (411) main: START, free heap size: 286400
I (431) main: size: 10000
I (471) main: set blob @0, error = 0, error_name = ESP_OK, heap = 264880
I (511) main: get blob @0, error = 0, error_name = ESP_OK, heap = 264880
I (511) main: START, free heap size: 286400
I (521) main: STOP, free heap size: 284920
I (521) main: START - STOP, memory diff: 1480

0xjakob, 2022-01-29:
@hkbora Thanks, we'll take a look at this issue!

Please keep in mind that there are holidays in Shanghai due to Chinese New Year, so the work on the issue might be slightly delayed.

hkbora, 2022-02-10:
Hello,

is there any update in this case?

Regards Hannes

benschmoo, 2022-02-21:
Hi,

i recently got the same issue. Are there any updates yet?

Regards, Bene

0xjakob, 2022-02-22:
Hi @hkbora Sorry for the really long response time!

Are you sure the memory leak comes from the NVS set/get blob functions? I moved the memory tracking functions directly around the `nvs_set/get_blob()` functions and got only 80 bytes "leak". Then I removed the calls to `makeFileName()` which allocate a string out of the area and now there is no leak at all.

Furthermore, the `nvs_flash_init()` call is in between the memory heap size measurement without any call to `nvs_flash_deinit()`. `nvs_flash_init()` will allocate heap memory for internal data structures. This is expected.

Btw., I would also not recommend to use the (non-trivial) return value of a function like this: `nvs_get_blob(nvsHandle, makeFileName(i).c_str(), buffer2, &length);`. I'm not sure if the compiler is able to always detect that some part of the return value of `makeFileName()` (in this case the internal buffer via `.c_str()`) will be used later.

0xjakob, 2022-02-22:
@benschmoo Could you please describe the issue in more detail and post some code here? How did you measure the memory leak?

0xjakob, 2022-03-07:
@hkbora @benschmoo Are you still able to reproduce memory leaks when setting NVS blobs or can we close this issue?

0xjakob, 2022-03-16:
@hkbora We'll close this issue. If you find that the problem still persists even when measuring only around `nvs_set_blob()`, feel free to re-open the issue.

closed 2022-03-16 by 0xjakob