espressif/esp-idf #8307: mdns crash due to not check empty proto, service

chegewara, 2022-01-27:
Hi, im not sure if its due to my badly initialized mdns, but i believe it should be "fixed" anyway, sometimes we can receive requests with only instance, and having service and proto empty:
```
instance: test
instance: test
instance: test
instance: test
GET / HTTP/1.1
User-Agent: Dalvik/2.1.0 (Linux; U; Android 11; SM-A725F Build/RP1A.200720.012)
Host: 192.168.0.113:50000
Connection: Keep-Alive
Accept-Encoding: gzip

instance: softap-8AE4
service: _telnet
proto: _tcp
```

This is causing crash here:
https://github.com/espressif/esp-idf/blob/master/components/mdns/mdns.c#L287-L288

I dont know if it is good solution, as i am fairly new to mdns, but this works to me:
```
    return (_str_null_or_empty(service) || !strcasecmp(srv->service, service)) && 
            (_str_null_or_empty(instance) || _mdns_instance_name_match(srv->instance, instance)) &&
            (_str_null_or_empty(proto) || !strcasecmp(srv->proto, proto)) && 
            (_str_null_or_empty(hostname) || !strcasecmp(srv->hostname, hostname));
```
Long story short, i am trying to add mdns to product with TCP/UDP server.
Testing with `Service browser` app on android.
esp-idf 309c1113f4943e8c0eade3fb2c87d6e78ff34ac3

Thanks

chegewara, 2022-01-27:
Another bug this time, is that mdns allows to set port to 0 (at least my synology NAS server is advertising with port 0 on `_device-info._tcp`),  but this check will return with `ESP_ERR_INVALID_ARG` 
https://github.com/espressif/esp-idf/blob/master/components/mdns/mdns.c#L5168

chegewara, 2022-01-31:
mdns also crash here due to `service == NULL`
https://github.com/espressif/esp-idf/blob/master/components/mdns/mdns.c#L1608

Simply check solves it
`if(!service) return;` - `continue` will stay in endless loop

chegewara, 2022-01-31:
There is few more crashes. The easy way to recreate it is:
- run 2 esp32 with the same mdns code
- when 1 esp32 is fully operating, then restart another
- i will try to test with mdns example and confirm

```
    const char* name, const char* ip, uint16_t port, int type // TCP or UDP;
    ESP_RETURN_ON_ERROR(mdns_init(), TAG, "failed to init mdns");
    ESP_RETURN_ON_ERROR(mdns_hostname_set(name), TAG, "failed to set hostname");
    ESP_LOGI(TAG, "mdns hostname set to: [%s]", name);
    //set default mDNS instance name
    ESP_ERROR_CHECK_WITHOUT_ABORT(mdns_instance_name_set(name));

    mdns_ip_addr_t addr4;
    esp_netif_str_to_ip4(ip, &addr4.addr.u_addr.ip4);
    addr4.addr.type = ESP_IPADDR_TYPE_V4;
    addr4.next = NULL;
    ESP_ERROR_CHECK_WITHOUT_ABORT(mdns_delegate_hostname_add(name, &addr4));
       
    if(type) ESP_ERROR_CHECK_WITHOUT_ABORT( mdns_service_add(NULL, "_telnet", "_udp", port, NULL, 0) );
    else ESP_ERROR_CHECK_WITHOUT_ABORT( mdns_service_add(NULL, "_telnet", "_tcp", port, NULL, 0) );
```

david-cermak, 2022-02-01:
Hi @chegewara 

Thanks very much for sharing the steps to reproduce those issues! I've just checked that they are fixed by the recently published bugfix related to default instances: https://github.com/espressif/esp-idf/commit/f46dffca627 (that's not included in the version you reference above)

Could you please update your master and test again? The proposed fixes you shared make sense in general, but we have to make sure that if the `service` or `proto` becomes `NULL` we don't have an issue with fqdn parsing, or mark the name `invalid`.

As for allowing the `por` to be zero, I don't think we should support that, it's an invalid port number and although `avahi` allows for it, `dns-sd` doesn't.

chegewara, 2022-02-01:
Thanks for checking it. I will update to master today and check it again.
Im not saying it is very important to support port 0, just saying that my synology NAS server, which is based on linux is advertising `_device-info` with port 0, and i thought it is normally supported, but of course i am noob in that matter and i will accept that answer.

chegewara, 2022-02-01:
Just in case, this is android app i am using:
https://play.google.com/store/apps/details?id=com.druk.servicebrowser&hl=en_US&gl=US

esp32 does not crash until i want to read complete information advertised by mdns service.

david-cermak, 2022-02-01:
Oh, thanks for the additional pointers! I'm able to reproduce the issue with `ServiceBrowser`! will fix.

david-cermak, 2022-02-02:
@chegewara Found one more crash and one mem-leak when testing this scenario. Here's the preliminary patch, it's very WIP, but should at least help you to continue with testing.
[mdns_WIP_fix_null_deref.txt](https://github.com/espressif/esp-idf/files/7984673/mdns_WIP_fix_null_deref.txt)

chegewara, 2022-02-02:
Thanks, i will try to test it.

chegewara, 2022-02-06:
Hi @david-cermak 
sorry for late reply, but i am sick.
After quick test i can tell the patch seems to work, and no longer crash.

Thanks

david-cermak, 2022-02-07:
Hi @chegewara 

Thanks for the feedback! hope you get well soon, I've found some more (minor) issues (plus, have realized the leak-fix above is incorrect).
The correct fix to `master` is coming soon.

closed 2022-02-25 by espressif-bot

me-no-dev, 2022-05-09:
@david-cermak any clues when this will be backported to 4.4?