espressif/esp-idf #8308: Access memory failed when malloc & free function repeatedly

HeloWong, 2022-01-28:
# Access memory failed when malloc & free function repeatedly

I got an ESP32-DevKit module with 4MB PSRAM, and I got a memory problem when I try to use malloc & free function to create a memory pool

## Problem

when using the malloc & free function many times (about 10 times, depending on the amount of alloc memory block), the watchdog will be triggered after running 2~3 cycles, if try to access memory.
 

## Code and Report
here is my code and its trace:

```cpp
#include <stdio.h>
#include <stdlib.h>
#include <vector>

#include "esp_system.h"
#include "esp_timer.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

constexpr int N = 1024 * 4000;

void fill_array(uint8_t *array, size_t len, int threshold) {
    if (array == NULL) return;
    unsigned int seed = esp_timer_get_time();
    srand(seed);
    for (size_t i = 0; i<len; i++) {
        *(array + i) = ((uint8_t)rand()) / (uint8_t)(RAND_MAX) * threshold;
    }
    return;
}

uint32_t sum_8b(uint8_t *array, size_t len) {
    if (array == NULL) return 0xffffffff;
    uint32_t ret = 0;
    for (size_t i = 0; i<len; i++) {
        ret += *(array + i);
    }
    return ret;
}

extern "C" void app_main(void)
{
    for (int i=0; i<10; i++)
    {
        uint8_t *pool = (uint8_t *)malloc(N);
        if (pool == NULL) {
            printf("cannot alloc memory!\n");
            break;
        }
        fill_array(pool, N, 25);
        int ret = sum_8b(pool, N);
        free(pool);
    }
    return;
}
``` 

and it reported:

```
E (16255) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (16255) task_wdt:  - IDLE (CPU 0)
E (16255) task_wdt: Tasks currently running:
E (16255) task_wdt: CPU 0: main
E (16255) task_wdt: CPU 1: IDLE
E (16255) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400D96A8:0x3FFB0820 0x40084A35:0x3FFB0840 0x400E7ED1:0x3FFB6280 0x400D7DBE:0x3FFB62A0 0x400E8AAA:0x3FFB62E0 0x4008BA05:0x3FFB6300
```

Dazza0, 2022-01-29:
@HeloWong Is the memory access actually failing, or is your loop just triggering the watchdog timer? Based on your log, I don't see any memory access/allocation/free errors.

Try adding a small delay at the end of each loop iteration to prevent the task watchdog from triggering. More details about the task watchdog [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html#task-watchdog-timer).
```c
extern "C" void app_main(void)
{
    for (int i=0; i<10; i++)
    {
        uint8_t *pool = (uint8_t *)malloc(N);
        if (pool == NULL) {
            printf("cannot alloc memory!\n");
            break;
        }
        fill_array(pool, N, 25);
        int ret = sum_8b(pool, N);
        free(pool);
        vTaskDelay(pdMS_TO_TICKS(50));
    }
    return;
}
```

Dazza0, 2022-02-14:
@HeloWong This issue looks like a task watchdog timeout issue. Please try adding delay as suggested above. Closing this issue, please feel free to reopen.

closed 2022-02-14 by Dazza0