espressif/esp-idf #8310: ESP_INTR_FLAG_IRAM not working with i2c (i2c_driver_install) on v4.4 tag (works on v4.3.2 tag)

greenaddress, 2022-01-28:
## Environment

- Development kit: custom board esp32 eco3
- IDF version: v4.4 tag
- Build System:       idf.py
- Compiler version : one from v4.4
- Operating System:     linux
- Using an IDE?: no
- Power Supply:  USB
- Compiled for size (Os)
- bootloader: from v4.4
- secure boot and flash encryption off

## Problem Description

calling i2c_driver_install with ESP_INTR_FLAG_IRAM doesn't work in v4.4 (but worked fine in v4.3.2)

### Expected Behavior

i2c_master_cmd_begin always returns  ESP_OK

### Actual Behavior

i2c_master_cmd_begin sometimes returns  ESP_ERR_INVALID_ARG

### Steps to reproduce

call 

i2c_driver_install  (w/ ESP_INTR_FLAG_IRAM set)
i2c_master_cmd_begin

-> ESP_ERR_INVALID_ARG

We are still investigating exactly when it happens, it seems some i2c calls succeed first and that it always fails in the same place (however it doesn't fail without the flag or with the flag and v4.3.2).

The device we are talking to via i2c is an axp192.

greenaddress, 2022-01-28:
this may be an instance of https://github.com/espressif/esp-idf/issues/8173

greenaddress, 2022-01-28:
code seems to work with SPIRAM off.

ginkgm, 2022-01-29:
Hi @greenaddress ,
ESP_INTR_FLAG_IRAM  means that you want your ISR to run when the cache is disabled. See https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/intr_alloc.html#iram-safe-interrupt-handlers, https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash_concurrency.html#iram-safe-interrupt-handlers.

In this case, all the code and data used by this ISR are supposed to be in internal memories, otherwise when the ISR use them when cache is disabled, ESP32 will panic. See this issue: https://github.com/espressif/esp-idf/issues/8173

We added a check in recent versions, to fix this issue. But you need to make sure the data buffer used by I2C is placed in the DRAM by calling heap_caps_malloc with [MALLOC_CAP_INTERNAL flag](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html#c.MALLOC_CAP_INTERNAL)

JamieDriver, 2022-02-02:
The issue, as documented clearly in #8173 is that as we are using a single byte value, the check `esp_ptr_internal(cmd_link->cmd.data)` always fails because `cmd.data` is not a pointer at all, but a byte value.  

The fix we require (I have cherry-picked this commit and tested locally) is as given on #8173, 
8d1fa78378482ac32865b3616574a8cf80ae8bd9, which currently appears to be in `master` only, afaics.

greenaddress, 2022-02-02:
hi @ginkgm thanks for your reply - I confirm it is as @JamieDriver said

ginkgm, 2022-02-25:
Thanks @JamieDriver @greenaddress 

I think the fix is now ready, see: https://github.com/espressif/esp-idf/issues/8173

closed 2022-02-25 by ginkgm